linux-can/can-utils #359: Cannot dump any data

loxK, 2022-04-27:
I own a [waveshare CAN HAT](https://thepihut.com/products/rs485-can-hat-for-raspberry-pi) on a raspberry zero. I need to read data broadcasted by a BMS ([CAN doc here](https://www.zeva.com.au/Products/datasheets/BMS16v2_CAN_Protocol.pdf)). The manufacturer told me "All data gets sent out by default so you don’t need to request anything, just receive and interpret."

I connected the "H" and the "L" of the HAT respectively to the "H" and the "L" of the BMS. I have the manufacturer's external monitor connected to the same CAN ports on the BMS and working properly, thus, showing CAN BUS data without issue.

I installed can-utils  v 2020.11.0-1  arm64 from repository.

```bash
# uname -a
Linux rasp0 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```

```bash
# lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

```bash
# dmesg 
[    9.703968] CAN device driver interface
[    9.793985] mcp251x spi0.0 can0: MCP2515 successfully initialized.
[ 1027.644470] can: controller area network core
[ 1027.644606] NET: Registered PF_CAN protocol family
[ 1027.658904] can: raw protocol
[ 1070.343869] IPv6: ADDRCONF(NETDEV_CHANGE): can0: link becomes ready
```

```bash
# ip link set can0 up type can bitrate 250000
# ip -det link show can0
4: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can <LOOPBACK> state ERROR-ACTIVE restart-ms 0 
	  bitrate 250000 sample-point 0.833 
	  tq 333 prop-seg 4 phase-seg1 5 phase-seg2 2 sjw 1
	  mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 6000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
```

I then try to dump, but I get no output at all:

```
# candump any -x
```

If I do `cangen can0` while `candump` is running, I can see some output.

What am I doing wrong ? The raspberry and the CAN HAT are brand new.

loxK, 2022-04-27:
It looks like the 120 Ohms is here looking at the [HAT schematics](https://www.waveshare.com/w/upload/1/1d/RS485_CAN_HAT_Schematic.pdf):

[image]

I think I have the 12M crystal, as the last board revision is from 2020. Not sure how I can verify that. But in the doubt, I already tried both overrides. 

Can it be caused by me using a Raspberry PI OS **64bit** image ?

hartkopp, 2022-04-28:
Hm, I'm not really an expert on this. Maybe @marckleinebudde can help here?

The termination resistor on the schematic is 120 Ohms. CAN usually needs 2x120 Ohms in parallel => 60 Ohms in total. You should add another 120 Ohms resistor. And do some measurement with a multimeter if you really got the 60 Ohms in your setup.

The frequency of the crystal should be printed on the silver housing on the PCB.

You wrote: "If I do `cangen can0` while `candump` is running, I can see some output."

Can you check whether your assigned interrupt 25 is increasing? (-> see `cat /proc/interrupts`)

And also post the output of  `ip -det link show can0` AFTER this attempt to write some (~20) CAN frames?

marckleinebudde, 2022-05-06:
I clicked a waveahre board with a 12 MHz oscillator, attached it to my rpi-4, using the raspi kernel v5.18 with the new "mcp2515" overlay and this entry in the `config.txt`:

```
dtoverlay=mcp2515,spi0-0,interrupt=25,oscillator=12000000,spimaxfrequency=2000000
```

It works out of the box.

loxK, 2022-05-11:
I am on kernel v5.15, tried both config which behave the same.

With loopback enabled, I can send and read what I send.

With `listen-only` off I get some data, but it doesn't seem to be what is expected: `RX errors` is increasing and keeps the same value as `frame` count.

<details>
    <summary>Candump output:</summary>

```
  can0  311   [8]  A5 57 24 92 49 25 E9 67
  can0  311   [6]  25 97 24 92 49 25
  can0  312   [8]  remote request
  can0  301   [0] 
  can0  314   [8]  remote request
  can0  301   [0] 
  can0  314   [8]  remote request
  can0  1B0   [8]  remote request
  can0  301   [0] 
  can0  304   [8]  remote request
  can0  1B0   [8]  remote request
  can0  0CD0158B   [0]  remote request
  can0       304   [8]  remote request
  can0       300   [8]  remote request
  can0  0CD0158B   [0]  remote request
  can0       311   [8]  AA AF 24 92 49 25 E9 67
  can0  0C45128B   [8]  remote request
  can0       301   [8]  AA AF 24 92 49 25 E9 67
  can0  0C45128B   [8]  remote request
  can0  0CD0158B   [0]  remote request
  can0       301   [8]  AA AF 24 92 49 25 E9 67
  can0       314   [8]  remote request
  can0       311   [8]  AA AF 24 92 49 25 E9 67
  can0       314   [8]  remote request
  can0  0C86128B   [8]  remote request
  can0       311   [8]  AA AF 24 92 49 25 E9 67
  can0       1B0   [8]  remote request
  can0  0C86128B   [8]  remote request
  can0       311   [8]  AA AF 24 92 49 25 E9 67
  can0       1B0   [8]  remote request
  can0  0C86128B   [8]  remote request
  can0  0CD09195   [8]  AA AF 24 92 49 25 E9 67
  can0  0C86128B   [8]  remote request
  can0       312   [6]  25 97 24 92 49 25
  can0       311   [8]  A4 AF 24 92 49 25 E9 67
  can0  0C45128B   [6]  25 1F 24 92 49 25
  can0  0C4B9195   [8]  A4 AF 24 92 49 25 E9 67
  can0  0C05128B   [6]  25 1F 24 92 49 25
  can0       314   [8]  remote request
  can0  0C05128B   [6]  25 1F 24 92 49 25
  can0       1B0   [8]  remote request
  can0  0C05128B   [6]  25 1F 24 92 49 25
  can0  0C06128B   [8]  remote request
  can0       1B0   [8]  remote request
  can0  0CD0128B   [6]  25 1F 24 92 49 25
  can0  0C06128B   [8]  remote request
  can0       300   [8]  remote request
  can0  0CD0128B   [6]  25 1F 24 92 49 25
  can0       311   [8]  89 5F 24 92 49 25 82 F3
  can0  0C45128B   [8]  remote request
  can0       301   [8]  89 5F 24 92 49 25 82 F3
  can0  0C45128B   [8]  remote request
  can0  0CD0128B   [6]  25 1F 24 92 49 25
  can0       301   [8]  89 5F 24 92 49 25 82 F3
  can0       314   [8]  remote request
  can0       311   [8]  89 5F 24 92 49 25 82 F3
  can0       314   [8]  remote request
  can0  0C87128B   [8]  remote request
  can0       1B0   [8]  remote request
  can0  0C87128B   [8]  remote request
[... 421 lines cut ...]
  can0       301   [8]  99 3F 24 92 49 25 FD FB
  can0       311   [8]  8C 4F 24 92 49 25 E9 67
  can0  0CD0924C   [8]  8C 4F 24 92 49 25 82 F3
  can0       301   [8]  99 3F 24 92 49 25 FD FB
  can0       314   [8]  remote request
  can0       311   [8]  99 3F 24 92 49 25 FD FB
  can0       314   [8]  remote request
  can0  0C872497   [8]  8C 4F 24 92 49 25 E9 67
  can0       311   [8]  99 3F 24 92 49 25 FD FB
  can0       1B0   [8]  remote request
  can0  0C872497   [8]  8C 4F 24 92 49 25 E9 67
  can0       311   [8]  99 3F 24 92 49 25 FD FB
  can0       1B0   [8]  remote request
  can0  0C872497   [8]  8C 4F 24 92 49 25 E9 67
  can0       301   [8]  99 3F 24 92 49 25 FD FB
  can0  0C872497   [8]  8C 4F 24 92 49 25 E9 67
  can0       1B0   [8]  remote request
  can0       312   [6]  99 3F 24 92 49 25
  can0       311   [8]  89 9F 24 92 49 25 E9 67
  can0       311   [8]  8C 4F 24 92 49 25 82 F3
  can0  0C4B9197   [8]  89 3F 24 92 49 25 E9 67
```
</details>

Without `listen-only`, there is no output.

marckleinebudde, 2022-05-11:
Maybe your bit timing doesn't match 100% the bit timing of the sender. The `mcp251x` driver limits the `tseg1_min1` to `3` resulting in a sample point a few % to early.

```
Bit timing parameters for mcp251x with 6.000000 MHz ref clock (12 MHz OSC) using algo 'v4.8'
 nominal                                  real  Bitrt    nom   real  SampP
 Bitrate TQ[ns] PrS PhS1 PhS2 SJW BRP  Bitrate  Error  SampP  SampP  Error   CNF1 CNF2 CNF3
 1000000    166   1    2    2   1   1  1000000   0.0%  75.0%  66.6%  11.2%   0x00 0x88 0x01
  800000 ***bitrate not possible***
  666666    166   3    3    2   1   1   666666   0.0%  80.0%  77.7%   2.9%   0x00 0x92 0x01
  500000    166   4    5    2   1   1   500000   0.0%  87.5%  83.3%   4.8%   0x00 0xa3 0x01
  250000    333   4    5    2   1   2   250000   0.0%  87.5%  83.3%   4.8%   0x01 0xa3 0x01
                                                                      ^^^^
~5% error in sample point
  125000    500   6    7    2   1   3   125000   0.0%  87.5%  87.5%   0.0%   0x02 0xb5 0x01
  100000    666   6    6    2   1   4   100000   0.0%  87.5%  86.6%   1.0%   0x03 0xad 0x01
   83333   1000   4    5    2   1   6    83333   0.0%  87.5%  83.3%   4.8%   0x05 0xa3 0x01
   50000   1333   6    6    2   1   8    50000   0.0%  87.5%  86.6%   1.0%   0x07 0xad 0x01
   33333   2000   6    6    2   1  12    33333   0.0%  87.5%  86.6%   1.0%   0x0b 0xad 0x01
   20000   3333   6    6    2   1  20    20000   0.0%  87.5%  86.6%   1.0%   0x13 0xad 0x01
   10000   6666   6    6    2   1  40    10000   0.0%  87.5%  86.6%   1.0%   0x27 0xad 0x01
```

I don't see this limitation in the data sheet, can you try the followin patch and see if it helps?

```
index fc747bff5eeb..7dbb1add55cb 100644
--- a/drivers/net/can/spi/mcp251x.c
+++ b/drivers/net/can/spi/mcp251x.c
@@ -209,7 +209,7 @@
 
 static const struct can_bittiming_const mcp251x_bittiming_const = {
        .name = DEVICE_NAME,
-       .tseg1_min = 3,
+       .tseg1_min = 2,
        .tseg1_max = 16,
        .tseg2_min = 2,
        .tseg2_max = 8,
```

Please check RX and post the output of

`ip -det link show can0`

loxK, 2023-04-22:
How should I apply the patch please, I am confused, I don't find the drivers/net/can/spi/mcp251x.c file.

marckleinebudde, 2023-04-24:
The patch is against the driver for the mcp251x chip, which is part of the Linux kernel, e.g. https://github.com/torvalds/linux/blob/master/drivers/net/can/spi/mcp251x.c

loxK, 2023-05-24:
I finally recompiled it as instructed : 

```bash
# ip -det link show can0
4: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state STOPPED restart-ms 0 
	  mcp251x: tseg1 2..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 6000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
```

marckleinebudde, 2023-05-24:
Ok, makes no difference with respect to the bit timing. Does it work now?

loxK, 2023-05-24:
`candump any -x` still outputs nothing.

If I run `cangen can0`, Bus goes off ...

```
4: can0: <NO-CARRIER,NOARP,UP,ECHO> mtu 16 qdisc pfifo_fast state DOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state BUS-OFF restart-ms 0 
	  bitrate 250000 sample-point 0.833 
	  tq 333 prop-seg 4 phase-seg1 5 phase-seg2 2 sjw 1
	  mcp251x: tseg1 2..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 6000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
```

marckleinebudde, 2023-05-24:
Have you checked the oscillator on your board? What does it say? Do you have a 2nd CAN controller you can test against?

marckleinebudde, 2023-10-03:
Any news on this? Feel free to re-open.

closed 2023-10-03 by marckleinebudde