linux-can/can-utils #361: BCM socket disregards extended CAN IDs (> 7FF)

michael-gosselin, 2022-04-28:
Hello, unclear to me what is going on here, but it appears only SFF frames are received by my BCM socket:

**_Symptom:_**

_SFF:_
```
cansend vcan0 7ff#11223344DEADBEEF
```
my program prints:
```
RX:  000007FF  [8]  11 22 33 44 DE AD BE EF
```
_EFF:_
```
cansend vcan0 00000800#11223344DEADBEEF
```
my program prints:
```
<nothing>
```
Just as a double-check I used `candump` to see what messages are coming on `vcan0`:

SFF:
```
vcan0  7FF   [8]  11 22 33 44 DE AD BE EF
```
EFF:
```
vcan0  00000800   [8]  11 22 33 44 DE AD BE EF
```

**_Code details:_**

_setup:_
(note: I removed all the checks on the return values of these setup function calls to make this more compact)
```
/* create and connect to socket */
socket_ = socket(PF_CAN, SOCK_DGRAM, CAN_BCM);
strcpy(ifr_.ifr_name, iface.c_str());
ioctl(socket_, SIOCGIFINDEX, &ifr_)
addr_.can_family = AF_CAN;
addr_.can_ifindex = ifr_.ifr_ifindex;
connect(socket_, (sockaddr *)&addr_, sizeof(addr_))

/* Set socket to non-blocking */
flags_ = fcntl(socket_, F_GETFL, 0);
fcntl(socket_, F_SETFL, flags_ | O_NONBLOCK)

/* header setup */
msg_.msg_head.opcode  = RX_SETUP;
msg_.msg_head.can_id  = arb_id;
msg_.msg_head.flags   = 0;
msg_.msg_head.nframes = 0;

if (write(socket_, &msg_, sizeof(msg_)) < 0)
{
    throw std::runtime_error("write: setup");
}
```
_try to receive:_

```
num_bytes_ = read(socket_, &msg_, sizeof(msg_));
if (num_bytes_ < 0) {
    if (errno != EAGAIN) {
        perror("read: unsuccessful");
    }
    return false;
}
else if (num_bytes_ < (ssize_t)sizeof(msg_)) {
    perror("read: incomplete message\n");
    return false;
}
else {
    can_frame * const frame = msg_.frame;
    print_frame(frame);
    return true;
}

```
_NOTE:_ `msg_` is an instance of:
```
struct can_msg {
	bcm_msg_head msg_head;
	can_frame frame[1];
};
```
I can see that [find_rcv_list](https://elixir.bootlin.com/linux/v2.6.25.12/source/net/can/af_can.c#L408) is supposed to setup what messages are received.

hartkopp, 2022-04-29:
```
...checking the header before setup:
BCM Message Header:
-------------------
opcode :  25024
flags :   0
count :   -735745152
can_id :  -737124830
nframes : 32722
can_frame: RX:  00000032  [0] 
```

Hm, what would you expect to happen with the `can_id :  -737124830` which is 0xD4105E22 as 32 bit unsigned int ??

That is an EFF & RTR frame with the CAN ID 0x14105E22.

You can try `cat /proc/net/can/rcvlist_*`  after doing a RX_SETUP to see what CAN ID the BCM has registered for.

I changed the CAN ID 0x042 in this test code https://github.com/linux-can/can-tests/blob/master/bcm/tst-bcm-filter.c to 0x80001000 which is a EFF Frame with CAN ID 0x1000 and it works as expected.

michael-gosselin, 2022-05-01:
Sorry - I am using linux 5.16.19 (not 2.6.25.12), it was my mistake to imply the wrong version in my original post.

I agree that `can_id` of `-737124830` makes no sense, in the context of CAN, generally (and same goes for `nframes` of `32722`).  I assumed that the bcm header struct values would not matter until the first `write()` call.  I actually don't know why the values of that header struct initialized like that (i.e. with a value of `-737124830` for the `can_id`, and so on.

I was trying to do something similar to what you do in your [test-bcm-filter.c](https://github.com/linux-can/can-tests/blob/f14512c96bb05ebd955ac05ef177e0828d2c5e92/bcm/tst-bcm-filter.c#L127), i.e. (1) set the values of the struct, and then (2) make a call to `write()`, but it seems I am doing something wrong.

hartkopp, 2022-05-01:
> I was trying to do something similar to what you do in your [test-bcm-filter.c](https://github.com/linux-can/can-tests/blob/f14512c96bb05ebd955ac05ef177e0828d2c5e92/bcm/tst-bcm-filter.c#L127), i.e. (1) set the values of the header struct, and then (2) make a call to `write()`, but it seems I am doing something wrong.

Hm, yes. I would assume some uninitialized or wrong dereferenced data structures.
In tst-bcm-filter.c the data structures are initialized by setting *all* the elements by hand (which could be improved too) ;-)

closed 2022-05-04 by hartkopp