linux-can/can-utils #362: Error: Unknown device type

xdoestech, 2022-05-12:
after going through [this](https://www.reddit.com/r/CarHacking/comments/ot3gjf/socketcancanutils_on_windows/) (https://www.reddit.com/r/CarHacking/comments/ot3gjf/socketcancanutils_on_windows/) guide on reddit 
I am able to run sudo modprobe vcan 
but ip link add dev vcan0 type vcan 
returns an error: unknown device type...

I am using linux on wsl2 for the first time so hopefully it is some basic issue?

any help is appreciated

hartkopp, 2022-05-12:
Hi as you started with your question here : https://github.com/linux-can/can-utils/issues/46#issuecomment-1125110859

I once build my own WSL2 kernel too to get the CAN support working.

But I'm pretty sure that there was no module loading support and the kernel was a single file.

So I enabled the CAN stuff by setting the selections to Y instead of M, so that the entire CAN support (including vcan) was built inside the kernel.

Or did you just build modules without building (and installing) the kernel?

hartkopp, 2022-05-12:
Hm, maybe the `ip` tool?

What does `ip -V` show?

And does `ip link help dev` show `vcan` and `vxcan` in the list of [TYPE]s ??

Maybe your `ip` tool is part of a multi-call binary like `busybox` which has a stripped configuration.

Then you should install the `iproute2` package to install the full blown version.

xdoestech, 2022-05-12:
Good looks that was the command I used... but unfortunately it shows both `vcan` and `vxcan` 

also looks like I already have iproute2 installed...

`TYPE := { vlan | veth | vcan | vxcan | dummy | ifb | 
         macvlan | macvtap |
           bridge | bond | team | ipoib | ip6tnl | ipip | sit | vxlan |
           gre | gretap | erspan | ip6gre | ip6gretap | ip6erspan |
           vti | nlmon | team_slave | bond_slave | bridge_slave |
           ipvlan | ipvtap | geneve | vrf | macsec | netdevsim | rmnet |
           xfrm }`

xdoestech, 2022-05-12:
I also installed debian (distro?) (from Microsoft store) to double-check that I did anything and it also shows the same output for `ip link help dev`, but it gives the original FATAL error telling me I made the new kernel correctly?

`modprobe: FATAL: Module vcan not found in directory /lib/modules/5.10.102.1-microsoft-standard-WSL2`

xdoestech, 2022-05-12:
would giving up on using the wsl and just booting linux solve this issue? At this point I'm ready to quit windows all together... make a boot USB and plug into my laptop?

hartkopp, 2022-05-12:
> would giving up on using the wsl and just booting linux solve this issue? At this point I'm ready to quit windows all together... make a boot USB and plug into my laptop?

Excellent choice! I'm using Xubuntu or Debian - installs within 10 minutes :-D

hartkopp, 2022-05-12:
> Good looks that was the command I used... but unfortunately it shows both `vcan` and `vxcan`
>
> also looks like I already have iproute2 installed...
>
> `TYPE := { vlan | veth | vcan | vxcan | dummy | [...]

The fact that `ip` shows up and also shows this help text does not necessarily mean that you have iproute2 installed.

On Debian `dpkg -l | grep iproute` should show if the package is really installed.

hartkopp, 2022-05-12:
> I also installed debian (distro?) (from Microsoft store) to double-check that I did anything and it also shows the same output for `ip link help dev`, but it gives the original FATAL error telling [...]

This is the problem with the module installation (and you need to run depmod -a after install).

That's why I built a WSL2 based Linux kernel without modules but with the CAN subsystem built-in.

xdoestech, 2022-05-12:
On the other side with linux mint... its not great so far, but I guess I just need to download everything

for debian I meant to say that I didn't add or install anything and was using it to test that the output was different from the ubuntu which I did change (I did the whole kernal nonsense including `depmod -a` on ubuntu and the fatal error went away, but it could not find `vcan0` even though `vcan` was installed

Im going with Linux Mint from here on out... If this is a terrible mistake let me know (It was recommend over anything else and I didn't want to have to spend 4 hours figuring it out... I ended up spending 5 hours)

hartkopp, 2022-05-13:
>  `depmod -a` on ubuntu and the fatal error went away, but it could not find `vcan0` even though `vcan` was installed

`vcan` is just the module.

Once loaded this `vcan` kernel module (e.g. with `modprobe vcan` ) the "software device" `vcan0` has to be created with `ip link add dev vcan0 type vcan` and then `ip link set vcan0 up`

Laskulappa, 2024-03-01:
once i run modprobe vcan i am getting an FATAL error: module vcan not found in directory

marckleinebudde, 2024-03-02:
You have to compile your kernel with `CONFIG_CAN_VCAN` enabled.