linux-can/can-utils #364: WSL2-linux-kernel can-subsystems not found in configuration

ursincantieni, 2022-06-10:
Hi, I want to make use of SocketCAN on windows, therefore I'm trying to compile a WSL2-Linux-Kernel, Ubuntu 20.04 based.
I'm following [these instructions](https://unix.stackexchange.com/questions/594470/wsl-2-does-not-have-lib-modules). Problem is, when running `$ make menuconfig`, the `Networking suppo
rt` submenu is empty, no can related stuff to find.

Any help is appreciated!

hartkopp, 2022-06-10:
I followed these instructions (which is similar to your reference):

https://gist.github.com/cerebrate/d40c89d3fa89594e1b1538b2ce9d2720

But I built my own WSL2 Kernel completely WITHOUT modules and compiled the CAN networking and driver stuff INTO the Kernel zImage which worked for me. So selecting "Y" instead of "M" in menuconfig.

ursincantieni, 2022-06-10:
I see, thanks for the help, that worked.

However, as soon as I get to `modprobe vcan`, I keep getting 
`modprobe: FATAL: Module vcan not found in directory /lib/modules/4.4.0-22000-Microsoft`,
which is unfortunate as this is what I'm trying to fix.

My menuconfig looks like as shown in the screens. Any idea what might be causing this?

[image]
[image]
[image]

hartkopp, 2022-06-12:
That is a correct behavior.

You can't `modprobe` a vcan module as it is no module anymore (when it is built-in).

You can directly start with creating a virtual CAN interface with `ip link add type vcan`.

closed 2022-06-30 by ursincantieni