linux-can/can-utils #367: No buffer space available

maltsevvv, 2022-06-26:
Hi I am using RPI3 and MCP2515 NiRen.  
#/boot/config.txt
```
dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=8000000,interrupt=25
dtoverlay=spi-bcm2835-overlay
```

candump can0   
```
can0 579#320800003300
can0 43A#080100000000
can0 621#01EB1D
can0 357#8000006400189B06
271#01
...
```
##
Sending a message in another window  
cansend can0 661#0001
`No buffer space available`

ip -details -statistic link show can0
```
4: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0
    can state BUS-OFF restart-ms 100
          bitrate 100000 sample-point 0.850
          tq 500 prop-seg 8 phase-seg1 8 phase-seg2 3 sjw 1
          mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
          clock 4000000
          re-started bus-errors arbit-lost error-warn error-pass bus-off
          0          0          0          26129      26130      0         numtxqueues 1 numrxqueues 1 gso_max_size 65                                                                                                                       536 gso_max_segs 65535
    RX: bytes  packets  errors  dropped overrun mcast
    101709     19750    0       2       0       0
    TX: bytes  packets  errors  dropped carrier collsns
    0          0        0       0       0       0

```

sd card 6 class

marckleinebudde, 2022-06-26:
You have a very high number of warning and passive errors.
* Is the configured bit rate of 100 kbit/s correct for your CAN bus?
* Please check the frequency of the oscillator on the your board. That's the big rectangular chip with a metallic enclosure.

maltsevvv, 2022-06-26:
metallic enclosure = 8.000

if the speed is incorrect, then there were no messages. Is not it?

marckleinebudde, 2022-06-26:
| metallic enclosure = 8.000

OK, then your `config.txt` entry is correct.

| if the speed is incorrect, then there were no messages. Is not it?

Usually you would not receive anything. I haven't a mcp2515 at hand to test how it behaves at running at half or double of the bus speed, but I think people have run into similar problems seeing CAN frames even if the bit rate if configured wrong. But on a correct bus, there should not be as many errors as on your bus.

Are you sure the bit rate of your CAN bus is 100 kbit/s?
Do the received CAN frames make any sense?

maltsevvv, 2022-06-27:
Thanks for the help!!!!!
The problem was in the MCP module

closed 2022-06-27 by maltsevvv

marckleinebudde, 2022-06-27:
Can you describe the problem and the solution? It might help other people having a similar problem.

maltsevvv, 2022-06-27:
### MCP2515 NiRen 8mhz
Connected to Raspberry PI
/boot/config.txt
`dtparam=spi=on
dtoverlay=mcp2515-can0,oscillator=8000000,interrupt=25
dtoverlay=spi-bcm2835-overlay`

check:
`dmesg | grep -i MCP2515`
'[    8.698783] mcp251x spi0.0 can0: MCP2515 successfully initialized.'

`sudo ip link set can0 up type can bitrate 100000 loopback on`
`cangen can0`
everything works fine, I see everything in the dump

**run** 
candump can0
`
can0 579#320800003300
can0 43A#080100000000
can0 621#01EB1D
can0 357#8000006400189B06
271#01
...
`
messages are correct

if I send a message to canbus, but he is not in сandump

marckleinebudde, 2022-06-28:
Was your MCP2515 NiRen hardware broken?

maltsevvv, 2022-06-28:
Yes