linux-can/can-utils #369: comparison of integers of different signs: 'int' and 'unsigned long'

marckleinebudde, 2022-06-30:
Sorry I didn't have time to dig more into the issue but https://github.com/linux-can/can-utils/pull/321 breaks the Android build:

```
external/can-utils/lib.c:575:28: error: comparison of integers of different signs: 'int' and 'unsigned long' [-Werror,-Wsign-compare]
                                if (tmp_n < 0 || tmp_n >= len - n){
                                                 ~~~~~ ^  ~~~~~~~
external/can-utils/lib.c:581:27: error: comparison of integers of different signs: 'int' and 'unsigned long' [-Werror,-Wsign-compare]
                        if (tmp_n < 0 || tmp_n >= len - n){
                                         ~~~~~ ^  ~~~~~~~
external/can-utils/lib.c:661:28: error: comparison of integers of different signs: 'int' and 'unsigned long' [-Werror,-Wsign-compare]
                                if (tmp_n < 0 || tmp_n >= len - n){
                                                 ~~~~~ ^  ~~~~~~~
external/can-utils/lib.c:667:27: error: comparison of integers of different signs: 'int' and 'unsigned long' [-Werror,-Wsign-compare]
                        if (tmp_n < 0 || tmp_n >= len - n){
                                         ~~~~~ ^  ~~~~~~~
4 errors generated.
```

_Originally posted by @gibsson in https://github.com/linux-can/can-utils/issues/368#issuecomment-1170868223_

weichslgartner, 2022-06-30:
will take care of it

sisco0, 2022-07-03:
@weichslgartner I agree with your solution if we should not consider the case that `n` be larger than `len`.
I could not see the comments from the `gh issue view` command, that is the reason why I opened other PR and closed it.

Cheers!

closed 2022-07-04 by marckleinebudde