linux-can/can-utils #373: Android no longer builds

tompee26, 2022-08-17:
I am not very familiar with this project but I use this a lot. With the recent changes to can calc timing, the build script for android doesn't work anymore. I could create a PR to update this line here in particular to point to the new directory but I am not sure how the new files should be reflected. Help would be appreciated. Thanks!

```
LOCAL_SRC_FILES := can-calc-bit-timing.c
```

marckleinebudde, 2022-08-17:
Sorry for the breakage, we should add an Android test to our testsuite.

Can you check if https://github.com/linux-can/can-utils/pull/374 fixes the problem.

Deishelon, 2022-08-17:
@marckleinebudde , applied the patch from #374 and it seems to be same:

```
$:~/projects/can-utils/build$ cmake -CMAKE_MAKE_PROGRAM=/home/nikita/AndroidNDK/android-ndk-r21d/prebuilt/linux-x86_64/bin/make  -DCMAKE_TOOLCHAIN_FILE=/home/nikita/AndroidNDK/android-ndk-r21d/build/cmake/android.toolchain.cmake -DANDROID_PLATFORM=android-21 -DANDROID_ABI=armeabi-v7a ..
loading initial cache file MAKE_MAKE_PROGRAM=/home/nikita/AndroidNDK/android-ndk-r21d/prebuilt/linux-x86_64/bin/make
CMake Error: Error processing file: /home/nikita/projects/can-utils/build/MAKE_MAKE_PROGRAM=/home/nikita/AndroidNDK/android-ndk-r21d/prebuilt/linux-x86_64/bin/make
-- Check for working C compiler: /home/nikita/AndroidNDK/android-ndk-r21d/toolchains/llvm/prebuilt/linux-x86_64/bin/clang
-- Check for working C compiler: /home/nikita/AndroidNDK/android-ndk-r21d/toolchains/llvm/prebuilt/linux-x86_64/bin/clang -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
CMake Error at CMakeLists.txt:90 (add_executable):
  add_executable cannot create target "can-calc-bit-timing" because another
  target with the same name already exists.  The existing target is an
  executable created in source directory "/home/nikita/projects/can-utils".
  See documentation for policy CMP0002 for more details.

-- Configuring incomplete, errors occurred!
See also "/home/nikita/projects/can-utils/build/CMakeFiles/CMakeOutput.log".
$:~/projects/can-utils/build$ 

```

Deishelon, 2022-08-17:
I see there is one more update from @marckleinebudde commit [e0945d8](https://github.com/marckleinebudde/can-utils/commit/e0945d867eeab54b2275e27f95f53a71b33dac31) above. 

Applied the patch on master branch, so:
```
$:~/projects/can-utils/build$ git diff
diff --git a/Android.mk b/Android.mk
index 85fbb8a..9aef576 100644
--- a/Android.mk
+++ b/Android.mk
@@ -181,7 +181,7 @@ include $(BUILD_EXECUTABLE)
 
 include $(CLEAR_VARS)
 
-LOCAL_SRC_FILES := can-calc-bit-timing.c
+LOCAL_SRC_FILES := calc-bit-timing/can-calc-bit-timing.c
 LOCAL_MODULE := can-calc-bit-timing
 LOCAL_MODULE_TAGS := optional
 LOCAL_C_INCLUDES := $(LOCAL_PATH)/include/
diff --git a/CMakeLists.txt b/CMakeLists.txt
index c317188..7088c51 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -46,7 +46,6 @@ set(PROGRAMS_J1939
 set(PROGRAMS
     ${PROGRAMS_CANLIB}
     bcmserver
-    can-calc-bit-timing
     canfdtest
     cangw
     cansniffer

```

And I can build now 🥳

closed 2022-08-17 by marckleinebudde