linux-can/can-utils #384: MCP2515 on Omega2

marckleinebudde, 2022-11-16:
I am customizing OpenWrt 22.03 (kernel 5.10) for my Omega2 hardware.
I connected MCP2515 to the SPI bus, on CS1 (GPIO11).
The DTS file looks like this.

```
&spi0 {
	status = "okay";

	pinctrl-names = "default";
	pinctrl-0 = <&spi_pins>, <&spi_cs1_pins>;

	flash0: flash@0 {
		compatible = "jedec,spi-nor";
		reg = <0>;
		spi-max-frequency = <40000000>;
		...
	};

	can0: can@1 {
		compatible = "microchip,mcp2515";
		reg = <1>;
		spi-max-frequency = <5000000>;
		clock-frequency  =  <20000000>;
		interrupt-parent = <&gpio>;
		interrupts = <11 IRQ_TYPE_EDGE_FALLING>;
	};
};
```

On MCP251x kernel module loading, `dmesg` says it  is initilaized succesfully.
I can enable `can0` and send messages using `cansend` command.
But I cannot get `candump` working.
If I connect external CAN device to send some message to the board, the INT pin falls to LOW, that means it is getting into some inturrupt status.

But the `candump` can read nothing. And from then, I cannnot make `cansend` working again.
I have to restart `can0` to make `cansend` working again.

When I connect two same board, and enable `can0` on board A then enable `can0` on board B, both boards can not send using `cansend`, just like the above. INT pins of both board falls to LOW and doesn't recover.

Here's detailed discussion on [Onion Omega2 Community](https://community.onion.io/topic/4892/can-bus-using-mcp2515-with-omega2).
There are what I've tried until now, e.g. using `listen-only` options, changing INT pin GPIO input directions, etc.

_Originally posted by @dumtux in https://github.com/linux-can/can-utils/issues/133#issuecomment-1317728661_

marckleinebudde, 2022-11-16:
FYI: https://github.com/dumtux

marckleinebudde, 2022-11-16:
```
interrupts = <11 IRQ_TYPE_EDGE_FALLING>;
```
You should use a level sensitive IRQ.

marckleinebudde, 2022-11-16:
Please post the output of `ip --details -s -s -s link show can0`

dumtux, 2022-11-17:
Hi @marckleinebudde , thanks for your effort helping me.
Here's the output before and after enabling `can0` device.

```
root@OpenWrt:~# echo 491 > /sys/class/gpio/export
root@OpenWrt:~# echo in > /sys/class/gpio/gpio491/direction
root@OpenWrt:~# echo 491 > /sys/class/gpio/unexport
root@OpenWrt:~# insmod /lib/modules/5.10.146/mcp251x.ko
root@OpenWrt:~# root@OpenWrt:~# ip --details -s -s -s link show can0
6: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state STOPPED restart-ms 0 
	  mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 10000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          0          0          0         numtxqueues 1 gso_max_size 65536 gso_max_segs 65535 
    RX:  bytes packets errors dropped  missed   mcast           
             0       0      0       0       0       0 
    RX errors:  length    crc   frame    fifo overrun           
                     0      0       0       0       0 
    TX:  bytes packets errors dropped carrier collsns           
             0       0      0       0       0       0 
    TX errors: aborted   fifo  window heartbt transns
                     0      0       0       0       0 
root@OpenWrt:~# ip link set can0 up type can bitrate 125000
root@OpenWrt:~# ip --details -s -s -s link show can0
6: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state ERROR-ACTIVE restart-ms 0 
	  bitrate 125000 sample-point 0.875 
	  tq 500 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
	  mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 10000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          0          0          0         numtxqueues 1 gso_max_size 65536 gso_max_segs 65535 
    RX:  bytes packets errors dropped  missed   mcast           
             0       0      0       0       0       0 
    RX errors:  length    crc   frame    fifo overrun           
                     0      0       0       0       0 
    TX:  bytes packets errors dropped carrier collsns           
             0       0      0       0       0       0 
    TX errors: aborted   fifo  window heartbt transns
                     0      0       0       0       1 
root@OpenWrt:~# 
```

marckleinebudde, 2022-11-17:
> Hi @marckleinebudde , thanks for your effort helping me. Here's the output before and after enabling `can0` device. It's just after new booting up, and nothing connected to CAN bus.
> 
> ```
> root@OpenWrt:~# echo 491 > /sys/class/gpio/export
> root@OpenWrt:~# echo in > /sys/class/gpio/gpio491/direction
> root@OpenWrt:~# echo 491 > /sys/class/gpio/unexport
> ```

What's this?

```
		interrupt-parent = <&gpio>;
		interrupts = <11 IRQ_TYPE_EDGE_FALLING>;
```

Your gpio 11 should be an interrupt pin not an input.

dumtux, 2022-11-17:
The GPIO11 is used as output High by Omega2's bootloader.
It is for powering on the Omega2 dock.

Of course, I am using my custom built board.
However, the Omega2 bootloader is doing that, so the GPIO11 pin is in output state HIGH, even after MCP2515 initializing.
Because of this, the GPIO11 is ~2V at interrupt status, because MCP2515 is trying to pull down it, while Omega2 is outputting high.

Once setting GPIO11 as input mode and releasing it,
GPIO11 is free from the Omega2's bootloader, so MCP2515 can pull down the pin to 0V correctly.

marckleinebudde, 2022-11-17:
You have to configure GPIO11 as an interrupt pin!

dumtux, 2022-11-17:
Whether I sent GPIO11 as input or output, the MCP251x module loading gives no error.
But if I change the pin number to something like 75 or 491 (wrong non-existing number), kernel loading says it cannot establish an interrupt.
(by `dmesg` I checked)

Thus, I thought this was working right.

So, should I find a way to set GPIO11 as an interrupt pin or use some else pin that is not occupied by the bootloader?

marckleinebudde, 2022-11-17:
> So, should I find a way to set GPIO11 as an interrupt pin or use some else pin that is not occupied by the bootloader?

You should modify your bootloader to not mess with this pin. You don't want to destroy the mcp2515. :)

Further you have to tell your SoC to route the pin 11 to the interrupt controller and not the GPIO controller.

dumtux, 2022-11-17:
> You don't want to destroy the mcp2515. :)
Ouch!  :P

I quickly surfed online and found some discussions for `kmod-gpio-irq` on Omega2 / OpenWrt.
It seems like there's no need to install an extra module for IRQ, and OpenWrt's DTS supports IRQ out of the box,
(like the above DTS file I posted)

Now I will look around for modifying Omega2's bootloader or solder flying wire to another GPIO pin for MCP2515's INT.
I'll update my trying results here soon.

marckleinebudde, 2022-11-17:
If the pin is driven high under Linux, the pin is obviously not configured for interrupt mode.

dumtux, 2022-11-18:
I found the GPIO11 behavior is managed at https://github.com/OnionIoT/omega2-bootloader/blob/master/lib_mips/board.c#L3028

```c
void gpio_init(void)
{
  ....
  RALINK_REG(RT2880_REG_PIODIR+0x04)=val;

  /*
  //zh@onion.io
  //setting GPIO 11 High, required for the reset button to work
  val=RALINK_REG(RT2880_REG_PIODIR);
  val|=1<<11;
  RALINK_REG(RT2880_REG_PIODIR) = val; // GPIO 11 direction output
  val=RALINK_REG(RT2880_REG_PIODATA);
  val|=1<<11;
  RALINK_REG(RT2880_REG_PIODATA) = val; // GPIO 11 High
  */

  //jeffzhou@onion.io
  //adding for read wifi MAC address.
  unsigned char macbuf[6];
	raspi_read(macbuf, CFG_FACTORY_ADDR - CFG_FLASH_BASE + 0x04, 6);
	printf("wifi mac address = %02X%02X%02X%02X%02X%02X.\n",
    ```

I commented on the GPIO11-related part in the bootloader source code and built it.
Just used the `make` command.

Unfortunately, after updating the bootloader, it didn't boot.
And I could not recover it.
It seemed like I missed some config options while building the Omega2's u-Boot..

I sent the board back to soldering technicians to replace it with a new Omega2 module.
I'll post more results here after the fixed board arrives.

dumtux, 2022-11-30:
Hi @marckleinebudde 
I disabled the GPIO11 pin usage of Omega2 bootloader.
Thus, GPIO11 is not occupied by anything.

Now the CAN bus status is like this.
```root@Omega-70C2:~# insmod mcp251x
module is already loaded - mcp251x
root@Omega-70C2:~# dmesg | grep -i MCP251x
[  106.334920] mcp251x spi0.1 can0: MCP2515 successfully initialized.
root@Omega-70C2:~# ip -det link show can0
5: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state STOPPED restart-ms 0 
	  mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 10000000 numtxqueues 1 gso_max_size 65536 gso_max_segs 65535 
root@Omega-70C2:~# ip link set can0 up type can bitrate 125000
root@Omega-70C2:~# ip --details -s -s -s link show can0
5: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state ERROR-ACTIVE restart-ms 0 
	  bitrate 125000 sample-point 0.875 
	  tq 500 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
	  mcp251x: tseg1 3..16 tseg2 2..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 10000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          0          0          0         numtxqueues 1 gso_max_size 65536 gso_max_segs 65535 
    RX:  bytes packets errors dropped  missed   mcast           
             0       0      0       0       0       0 
    RX errors:  length    crc   frame    fifo overrun           
                     0      0       0       0       0 
    TX:  bytes packets errors dropped carrier collsns           
             0       0      0       0       0       0 
    TX errors: aborted   fifo  window heartbt transns
                     0      0       0       0       1 
root@Omega-70C2:~# ```

marckleinebudde, 2022-11-30:
So far so good, for loading the driver no IRQ functionality is used. Can you send or receive CAN frames?

dumtux, 2022-11-30:
I can send CAN frames by using `cansend` command.
(it was able to operate before this bootloader customization, too.)

I can not read CAN frames from external sensor.
The INT pin GPIO11 is not triggered though there are repeating CAN frames from sensors.

I also connected two boards (one board = Omega2 + MCP2515).
From one, sending CAN frames by `cansend`, from another, trying to receive by `candump`.
But still same no result.
As the INT pin is not triggering (always high), it makes sense that `candump` giving no result.
Buty why is INT not triggered?

marckleinebudde, 2022-12-01:
Can you remove the INT pin of the mcp2515 from the GPIO11 of the SoC and measure if the mcp2515 drives the INT pin down after it received a CAN frame?

I just rechecked: the IRQ functionality is usually implemented by the GPIO controller, so configuring GPIO11 as a input is needed. Usually done via pinctrl. Can you try the 
```shell
echo 491 > /sys/class/gpio/export
echo in > /sys/class/gpio/gpio491/direction
echo 491 > /sys/class/gpio/unexport
```
hack again?

dumtux, 2022-12-01:
That's what I tried before.
With old bootloader from Omega2, the GPIO11 (bank number 491 in OpenWRT22) was output HIGH status.
In that case, MCP2515 was trying to drop the line as low.
As a result, the GPIO11 was ~ 2.0V.

As a quick fix, with such IO direction fix, the MCP2515 could pull the pin correctly down to LOW level ~ 0V.

Now I patched the default bootloader.
Thus, I don't need to do that IO direction fix anymore.
MCP2515 is pulling the GPIO11 line down well, I checked with an oscilloscope.

Now the problem is, MCP2515 is not triggering INT while there are coming in packets on the bus.

marckleinebudde, 2022-12-01:
Can you never the less try the hack? Have you changed the interrupt to level triggered in the DT?

dumtux, 2022-12-01:
Yes, I both tried `IRQ_TYPE_EDGE_FALLING` and `IRQ_TYPE_LEVEL_LOW`

dumtux, 2022-12-01:
The pin level is HIGH.
And using oscilloscope, it's 3.3V.

marckleinebudde, 2022-12-01:
high means no IRQ

dumtux, 2022-12-01:
Here I attach some waveforms.
On CAN bus, a AH125 angle sensor module is connected.
It is outputting CAN packets periodically.

Before `can0` enablinng.on Omega2-MCP2515 side, waveform is like this.

[image]
[image]

After enabling `can0` interface, the waveform becomes like this.
```sh
ip link set can0 up type can bitrate 125000
```

[image]

If I disable the `can0` bus now by
```sh
ip link set can0 down
```
, the waveform becomes same as the first state

[image]

---

At any stage, `candump can0` shows no result.
The GPIO11 pin is always HIGH 3.3V.

dumtux, 2022-12-01:
In my understanding, when the external sensor emits a CAN packet, the MCP2515 should trigger IRQ.
Then the Omega2 reads the receiving buffer of MCP2515 through SPI bus.
As no IRQ is triggered from MCP2515, `candump can0` shows no result.

marckleinebudde, 2022-12-01:
Before your bootloader change, you said the mcp2515 was pulling the IRQ line low. What's the difference now?

dumtux, 2022-12-01:
Yes, it was pulling down and never went back to HIGH till I reset power.
But now, it is not at all pulling down.

marckleinebudde, 2022-12-01:
Can you disconnect the IRQ pin of the mcp2515 from the SoC and check if it's going down. If this is not working, there's a different problem.

marckleinebudde, 2022-12-01:
Have a look at the [mcp2515 datasheet](https://ww1.microchip.com/downloads/en/DeviceDoc/MCP2515-Stand-Alone-CAN-Controller-with-SPI-20001801J.pdf) section "10.4 Loopback Mode":

> Loopback mode will allow internal transmission of messages from the transmit buffers to the receive buffers without actually transmitting messages on the CAN bus. This mode can be used in system development and testing.
> 
> In this mode, the ACK bit is ignored and the device willeallow incoming messages from itself, just as if they were coming from another node. The Loopback mode is a silent mode, meaning no messages will be transmitted while in this state (including error flags or Acknowledge signals). The TXCAN pin will be in a recessive state.

marckleinebudde, 2022-12-01:
> It is difficult to cut the trace, as it is routed in the 2nd layer of the 4-layer PCB.

You can de-solder the leg of the mcp2515 and lift it up.

> I understand what you need to test. However, as I wrote earlier when the Omega2 is configured to out HIGH on GPIO11 by the bootloader, I could see ~2.0V middle voltage when MCP2515 IRQ triggered. That means, if IRQ is triggered, we should be able to wee 0V or 2.0V for any internal state of GPIO11 from Omega2 config. Thus, I believe the IRQ is not triggered.

If you measure the unconnected IRQ pin of know it IRQs are triggered or not.

dumtux, 2022-12-01:
> Have a look at the [mcp2515 datasheet](https://ww1.microchip.com/downloads/en/DeviceDoc/MCP2515-Stand-Alone-CAN-Controller-with-SPI-20001801J.pdf) section "10.4 Loopback Mode":
> 
> > Loopback mode will allow internal transmission of messages from the transmit buffers to the receive buffers without actually transmitting messages on the CAN bus. This mode can be used in system development and testing.
> > In this mode, the ACK bit is ignored and the device willeallow incoming messages from itself, just as if they were coming from another node. The Loopback mode is a silent mode, meaning no messages will be transmitted while in this state (including error flags or Acknowledge signals). The TXCAN pin will be in a recessive state.
> 
> I think it works as intended.

Oh I didn't realise it.
So MCP2515 is correctly set to loopback mode when there's no electrical output.
I cannot see any data from `candump can0` command still, after running `cansend`.
Does it mean my `candump` command usage is wrong?

Or maybe `candump` is not properly working with my current setup?

marckleinebudde, 2022-12-02:
I'm not a native speaker, but I would phrase it like: If the MCP2515 is set to loopback mode, there's no electrical output.

`cansend` and `candump` are the correct commands to test, but if you don't get IRQs, you'll not see any CAN frames in `candump`.

So I still suggest that you de-solder the IRQ pin of the mcp2515 and check it's level.

dumtux, 2022-12-02:
I did it, and the IRQ is not triggering from MCP2515 side.
I can wonder if the MCP2515 is broken, but I have two boards and two are having the exactly same behavior.
I will think out some alternative solution.

marckleinebudde, 2023-10-03:
Any news on this? I'm closing this issue now. Feel free to reopen.

closed 2023-10-03 by marckleinebudde