linux-can/can-utils #385: candump can0 times out

meghancampbell9, 2022-12-05:
Im using SeedStudios CAN-BUS shield V2.0 attached to an Arduino to send CAN data to my computer from my sensor (Delphi ESR Radar). I'm using this Arduino sketch which makes a CAN-BUS shield into a CAN-USB adapter for Linux SocketCAN (can-utils) (https://github.com/kahiroka/slcanuino). I've got this setup running on a Jetson TX2 computer running Ubuntu 18.04, however when I follow the exact same setup on my laptop running Ubuntu 22.04 I receive a few frame for maybe ~10 seconds and then the signal stops and I don't receive any more data after running candump can0. I know its not my hardware setup since its able to run on a different computer. Ive checked the serial communication settings of the laptop and i was able to receive serial data from other devices. Does anyone know if this is an issue with Ubuntu 22.04, or perhaps a driver I am missing? When the data stops the LEDs on the CAN-BUS shield (TX and RX) stop flashing, and only the RX light stays on.

marckleinebudde, 2022-12-05:
I'm sorry, but I have to say this is one of the worst options to attach a CAN bus. You would be far better of with a [candlelight compatible device](https://github.com/candle-usb/candleLight_fw).

meghancampbell9, 2023-08-24:
The issue was with the laptop and not with the attachment method. The setup works great with a NVIDIA Jetson computer.

hartkopp, 2023-08-25:
Can you tell me the kernel versions of your two setups (Jetson & Ubuntu 22.04) ?

Btw. you should really consider to use the high performance GS_USB driver with candleLight hardware as suggested by @marckleinebudde .

I got two of those adapters for less than 20 Euros from China and flashed the latest candleLight firmware from https://github.com/candle-usb/candleLight_fw - which works really good. The delivered firmware on these interfaces was completely outdated and I only wanted to use trustworthy firmware.

[image]
[image]

meghancampbell9, 2023-08-28:
Sorry, it was for a university project and I can't remember the kernel versions. I don't have access to the hardware anymore.

I'd definitely upgrade to candleLight if we had the budget, however the hardware was bought by a previous year and we couldn't get approval from the school to replace it, so we made it work with what we had.

closed 2023-08-28 by meghancampbell9