linux-can/can-utils #398: Cannot build using ARMv7 toolchain

b-thomson, 2023-01-24:
I'm relatively new to this, so apologies if the answer is obvious.

After adding support for SO_TXTIME in cangen.c (https://github.com/linux-can/can-utils/commit/c6f2cf7c2f2786b0a769534d0a1f0716326a3494) it no longer builds with my ARMv7 toolchain:

```
brenton@ubuntu:~/can-utils$ cmake -DCMAKE_TOOLCHAIN_FILE=armv7-linux-gnueabi.cmake && make
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

-- The C compiler identification is GNU 7.4.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /opt/toolchain/gcc-conel-armv7-linux-gnueabi/bin/armv7-linux-gnueabi-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /home/brenton/can-utils
[  1%] Building C object CMakeFiles/can-calc-bit-timing.dir/calc-bit-timing/can-calc-bit-timing.c.o
[  2%] Linking C executable can-calc-bit-timing
[  2%] Built target can-calc-bit-timing
[  4%] Building C object CMakeFiles/mcp251xfd-dump.dir/mcp251xfd/mcp251xfd-dev-coredump.c.o
[  5%] Building C object CMakeFiles/mcp251xfd-dump.dir/mcp251xfd/mcp251xfd-dump.c.o
[  7%] Building C object CMakeFiles/mcp251xfd-dump.dir/mcp251xfd/mcp251xfd-main.c.o
[  8%] Building C object CMakeFiles/mcp251xfd-dump.dir/mcp251xfd/mcp251xfd-regmap.c.o
[ 10%] Linking C executable mcp251xfd-dump
[ 10%] Built target mcp251xfd-dump
[ 11%] Building C object CMakeFiles/j1939.dir/libj1939.c.o
[ 12%] Linking C static library libj1939.a
[ 12%] Built target j1939
[ 14%] Building C object CMakeFiles/can.dir/lib.c.o
[ 15%] Building C object CMakeFiles/can.dir/canframelen.c.o
[ 17%] Linking C static library libcan.a
[ 17%] Built target can
[ 18%] Building C object CMakeFiles/asc2log.dir/asc2log.c.o
[ 20%] Linking C executable asc2log
[ 20%] Built target asc2log
[ 21%] Building C object CMakeFiles/canbusload.dir/canbusload.c.o
[ 22%] Linking C executable canbusload
[ 22%] Built target canbusload
[ 24%] Building C object CMakeFiles/candump.dir/candump.c.o
[ 25%] Linking C executable candump
[ 25%] Built target candump
[ 27%] Building C object CMakeFiles/cangen.dir/cangen.c.o
/home/brenton/can-utils/cangen.c: In function 'setsockopt_txtime':
/home/brenton/can-utils/cangen.c:229:35: error: 'SO_TXTIME' undeclared (first use in this function); did you mean 'SI_TIMER'?
  ret = setsockopt(fd, SOL_SOCKET, SO_TXTIME,
                                   ^~~~~~~~~
                                   SI_TIMER
/home/brenton/can-utils/cangen.c:229:35: note: each undeclared identifier is reported only once for each function it appears in
/home/brenton/can-utils/cangen.c: In function 'do_send_one':
/home/brenton/can-utils/cangen.c:311:19: error: 'SCM_TXTIME' undeclared (first use in this function); did you mean '_SC_CPUTIME'?
   cm->cmsg_type = SCM_TXTIME;
                   ^~~~~~~~~~
                   _SC_CPUTIME
make[2]: *** [CMakeFiles/cangen.dir/build.make:76: CMakeFiles/cangen.dir/cangen.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:330: CMakeFiles/cangen.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
```
If I use the cangen.c before the above commit (https://github.com/linux-can/can-utils/commit/9dff4b6393673f1bcf351109a9cc77873dd1226e) it builds normally.

Could this be due to the older version of gcc used in the toolchain?
The toolchain can be found here: https://bitbucket.org/bbsmartworx/toolchains/src/master/

Any help would be greatly appreciated.
Thank you!

marckleinebudde, 2023-01-24:
Yes that due to a too old toolchain, I'll create a fix.

marckleinebudde, 2023-01-24:
Can you test https://github.com/marckleinebudde/can-utils/tree/fix-old-toolchains

closed 2023-01-24 by marckleinebudde