linux-can/can-utils #400: OSError: [Errno 105] No buffer space available

fabHO1, 2023-01-24:
Hello everyone I am working with a raspberry pi 3+ and a PICANDuo I am working with raspbian, I followed the steps in the user manual of the pican to make it work and then I tried to run a sample python code to receive data from the OBDII port but when I run it after a few seconds it sends me some errors including OSError: [Errno 105] No buffer space available even before running the code I modified the socketcan native to only socketcan since the first is no longer for version 4. 0 of python-can and also indicates me some file folders in which there is a file socketcan. py which indicates me some lines this last one do you think it is an error of the socketcan file or is it normal I hope you can help me thanks.

This is the python example I am running from my terminal.

#!/usr/bin/python3
#
## obdii_logger.py
# 
# This python3 program sends out OBDII request then logs the reply to the sd card.
# For use with PiCAN boards on the Raspberry Pi
# http://skpang.co.uk/catalog/pican2-canbus-board-for-raspberry-pi-2-p-1475.html
#
# Make sure Python-CAN is installed first http://skpang.co.uk/blog/archives/1220
#
#  24-08-16 SK Pang
#

import RPi.GPIO as GPIO
import can
import time
import os
import queue
from threading import Thread

led = 22
GPIO.setmode(GPIO.BCM)
GPIO.setwarnings(False)
GPIO.setup(led,GPIO.OUT)
GPIO.output(led,True)

# For a list of PIDs visit https://en.wikipedia.org/wiki/OBD-II_PIDs
ENGINE_COOLANT_TEMP = 0x05
ENGINE_RPM          = 0x0C
VEHICLE_SPEED       = 0x0D
MAF_SENSOR          = 0x10
O2_VOLTAGE          = 0x14
THROTTLE            = 0x11

PID_REQUEST         = 0x7DF
PID_REPLY           = 0x7E8

outfile = open('log.txt','w')

print('\n\rCAN Rx test')
print('Bring up CAN0....')

# Bring up can0 interface at 500kbps
os.system("sudo /sbin/ip link set can0 up type can bitrate 500000")
time.sleep(0.1)	
print('Ready')

try:
	bus = can.interface.Bus(channel='can0', bustype='socketcan')
except OSError:
	print('Cannot find PiCAN board.')
	GPIO.output(led,False)
	exit()

def can_rx_task():	# Receive thread
	while True:
		message = bus.recv()
		if message.arbitration_id == PID_REPLY:
			q.put(message)			# Put message into queue

def can_tx_task():	# Transmit thread
	while True:

		GPIO.output(led,True)
		# Sent a Engine coolant temperature request
		msg = can.Message(arbitration_id=PID_REQUEST,data=[0x02,0x01,ENGINE_COOLANT_TEMP,0x00,0x00,0x00,0x00,0x00], is_extended_id=False)
		bus.send(msg)
		time.sleep(0.05)

		# Sent a Engine RPM request
		msg = can.Message(arbitration_id=PID_REQUEST,data=[0x02,0x01,ENGINE_RPM,0x00,0x00,0x00,0x00,0x00], is_extended_id=False)
		bus.send(msg)
		time.sleep(0.05)

		# Sent a Vehicle speed  request
		msg = can.Message(arbitration_id=PID_REQUEST,data=[0x02,0x01,VEHICLE_SPEED,0x00,0x00,0x00,0x00,0x00],is_extended_id=False)
		bus.send(msg)
		time.sleep(0.05)		

		# Sent a Throttle position request
		msg = can.Message(arbitration_id=PID_REQUEST,data=[0x02,0x01,THROTTLE,0x00,0x00,0x00,0x00,0x00],extended_id=False)
		bus.send(msg)
		time.sleep(0.05)
		
		GPIO.output(led,False)
		time.sleep(0.1)
						
						
q = queue.Queue()
rx = Thread(target = can_rx_task)  
rx.start()
tx = Thread(target = can_tx_task)
tx.start()

temperature = 0
rpm = 0
speed = 0
throttle = 0
c = ''
count = 0

# Main loop
try:
	while True:
		for i in range(4):
			while(q.empty() == True):	# Wait until there is a message
				pass
			message = q.get()

			c = '{0:f},{1:d},'.format(message.timestamp,count)
			if message.arbitration_id == PID_REPLY and message.data[2] == ENGINE_COOLANT_TEMP:
				temperature = message.data[3] - 40			#Convert data into temperature in degree C

			if message.arbitration_id == PID_REPLY and message.data[2] == ENGINE_RPM:
				rpm = round(((message.data[3]*256) + message.data[4])/4)	# Convert data to RPM

			if message.arbitration_id == PID_REPLY and message.data[2] == VEHICLE_SPEED:
				speed = message.data[3]										# Convert data to km

			if message.arbitration_id == PID_REPLY and message.data[2] == THROTTLE:
				throttle = round((message.data[3]*100)/255)					# Conver data to %

		c += '{0:d},{1:d},{2:d},{3:d}'.format(temperature,rpm,speed,throttle)
		print('\r {} '.format(c))
		print(c,file = outfile) # Save data to file
		count += 1
			

 
	
except KeyboardInterrupt:
	#Catch keyboard interrupt
	GPIO.output(led,False)
	outfile.close()		# Close logger file
	os.system("sudo /sbin/ip link set can0 down")
	print('\n\rKeyboard interrtupt')

hartkopp, 2023-01-24:
Haha, are you in the same group as @Hector897 who posted the exact same question??

Please take a look here:

https://github.com/hardbyte/python-can/issues/1495

Thanks

closed 2023-01-24 by hartkopp

fabHO1, 2023-01-24:
> Jaja, ¿estás en el mismo grupo que@Hector897¿Quién publicó exactamente la misma pregunta?
> 
> Por favor, eche un vistazo aquí:
> 
> [hardbyte/python-can#1495](https://github.com/hardbyte/python-can/issues/1495)
> 
> Gracias
No but I am using the same example and others and when I run them I get the error OSError: [Errno 105] No buffer space available thank you for your answer.

marckleinebudde, 2023-01-25:
Hello @fabHO1, IMHO it will not help if you copy/paste this issue to random other issues in this project. As @hartkopp said, please use the existing issue: https://github.com/hardbyte/python-can/issues/1495.

Thanks

fabHO1, 2023-01-25:
> Hola@fabHO1, en mi humilde opinión, no ayudará si copia/pega este problema en otros problemas aleatorios en este proyecto. Como@hartkoppdicho, utilice el problema existente: [hardbyte/python-can#1495](https://github.com/hardbyte/python-can/issues/1495) .
> 
> Gracias

Hi thanks but I followed the steps to solve the existing problem and it did not work I keep getting the message OSError: [Errno 105] No buffer space available, another thing I want to connect the picanduo board to the OBDII port of a Toyota Hilux would you know if before connecting the OBDII cable you have to perform some configuration to receive the data that the code I attached to work properly.

marckleinebudde, 2023-01-25:
@fabHO1 please follow these suggestions:
https://github.com/hardbyte/python-can/issues/1495#issuecomment-1400884690
https://github.com/hardbyte/python-can/issues/1495#issuecomment-1400957065

fabHO1, 2023-01-25:
> @fabHO1siga estas sugerencias: [hardbyte/python-can#1495 (comentario) ](https://github.com/hardbyte/python-can/issues/1495#issuecomment-1400884690) [hardbyte/python-can#1495 (comentario)](https://github.com/hardbyte/python-can/issues/1495#issuecomment-1400957065)

I will try it and let you know if it works for me, thank you.