linux-can/can-utils #402: doubt: using testj1939 to populate struct struct can_frame

acode-x, 2023-01-28:
Hi all,

I am playing around with testj1939 to read some data.
https://github.com/linux-can/can-utils/blob/fe3717948b5be13e06bfee184bd7d8cc5d07d444/testj1939.c#L274-L276

I am curious if I can replace above logic to populate a `struct can_frame` instead.
I tried `read(sock, &frame, CAN_MTU);` but that doesn’t seem to workout. The can_id field is wrong and data is 0.

Requesting help on same.
Thanks!

marckleinebudde, 2023-01-28:
j1939 is a protocol on top of raw CAN. It makes no sense to read raw CAN frames from a j1939 socket.

acode-x, 2023-01-28:
Thanks a lot for confirmation and quick response.
I have another followup question, for which I couldn’t find relevant documentation / sample C code.

This is regarding handling J1939 messages having length > 8 bytes.  I am using https://github.com/nberlette/canbus/blob/main/dbc/j1939.dbc as a reference.  It has messages:
```
BO_ 2633939198 GTFI: 22 Vector__XXX
 SG_ ...

BO_ 2633832958 ILI: 200 Vector__XXX
 SG_ ...
```
How do I extend testj1939 / custom SocketCAN code to process/receive entire message 8 bytes at a time. Is there any specific way of processing/aggregating such frames? Is it possible to cansend a dummy payload of such message...

Would be great if you could throw some light on above.

Thanks again!

marckleinebudde, 2023-01-28:
The kernel will do the handling of TP or ETP messages for you. Call `receive()` with a buffer large enough to receive the while TP/ETP message.

acode-x, 2023-01-28:
Cool! Thank you so much

closed 2023-01-28 by acode-x