linux-can/can-utils #404: Trouble building from source in Termux

shaunmulligan, 2023-02-06:
Hi, perhaps a bit of an odd setup, but I am trying to build can-utils from source with in [termux](https://github.com/termux) running on a rooted android device. I have all the necessary kernel modules built into android and can run can-utils tools from a docker container within termux, but I need the tools in the termux user space so that I can forward can0 traffic to a vxcan interface for the containers to use. 

It almost builds but hits the following error, can you point me in the right direction to potentially patch this to build well within termux or how to just build what is necessary to use the `cangw` command. Thanks!

```
~/can-utils $ make
  CC       asc2log.o
  CC       lib.lo
  CC       canframelen.lo
  CCLD     libcan.la
  CCLD     asc2log
  CC       bcmserver.o
  CCLD     bcmserver
  CC       calc-bit-timing/can-calc-bit-timing.o
  CCLD     can-calc-bit-timing
  CC       canbusload.o
  CCLD     canbusload
  CC       candump.o
  CCLD     candump
  CC       canfdtest.o
  CCLD     canfdtest
  CC       cangen.o
  CCLD     cangen
  CC       cangw.o
  CCLD     cangw
  CC       canlogserver.o
  CCLD     canlogserver
  CC       canplayer.o
  CCLD     canplayer
  CC       cansend.o
  CCLD     cansend
  CC       cansequence.o
  CCLD     cansequence
  CC       cansniffer.o
  CCLD     cansniffer
  CC       isotpdump.o
  CCLD     isotpdump
  CC       isotpperf.o
  CCLD     isotpperf
  CC       isotprecv.o
  CCLD     isotprecv
  CC       isotpsend.o
  CCLD     isotpsend
  CC       isotpserver.o
  CCLD     isotpserver
  CC       isotpsniffer.o
  CCLD     isotpsniffer
  CC       isotptun.o
  CCLD     isotptun
  CC       j1939acd.o
j1939acd.c:470:2: error: use of undeclared identifier 'program_invocation_name'
        program_invocation_name = program_invocation_short_name;
        ^
j1939acd.c:470:28: error: use of undeclared identifier 'program_invocation_short_name'
        program_invocation_name = program_invocation_short_name;
                                  ^
j1939acd.c:489:18: error: use of undeclared identifier 'program_invocation_name'
                        if (asprintf(&program_invocation_name, "%s.%s",
                                      ^
j1939acd.c:490:10: error: use of undeclared identifier 'program_invocation_short_name'
                                     program_invocation_short_name, optarg) < 0)
                                     ^
4 errors generated.
make: *** [GNUmakefile:905: j1939acd.o] Error 1
```

marckleinebudde, 2023-02-06:
Does this fix your problem: https://github.com/marckleinebudde/can-utils/tree/program_invocation_short_name

shaunmulligan, 2023-02-06:
@marckleinebudde awesome, yes it does!!  I still have an issue with the `make install` because its trying to put it somewhere it shouldn't, but this gets things installed! 
Thanks so much!

marckleinebudde, 2023-02-06:
BTW: see https://github.com/linux-can/can-utils/blob/master/README.md#cmake-project-generator for Android and Android Studio integration.

shaunmulligan, 2023-02-06:
Awesome, I had seen that but really needed can-utils in termux to help me stand up a vxcan interface which I could use with in docker containers. Is https://github.com/marckleinebudde/can-utils/tree/program_invocation_short_name something you would consider merging in?

marckleinebudde, 2023-02-06:
Thinking of a better solution...which build system are you using to build the sources?

shaunmulligan, 2023-02-08:
Ultimately I would like to try get `can-utils` added to https://github.com/termux/termux-packages officially