linux-can/can-utils #408: candump: Why return 0 on signals

yashi, 2023-02-19:
Hi,

This is more like a question rather than an issue.

Currently [`candump` returns `0`](https://github.com/linux-can/can-utils/blob/master/candump.c#L318) when it gets a signal.  Is there any reason for that?  Many CLI tools return `128 + SIGNUM`.  https://www.gnu.org/savannah-checkouts/gnu/bash/manual/bash.html#Simple-Commands

Is it because `candump` is more of a tool for human and not for scripting?

Would you accept a patch to fix it?

marckleinebudde, 2023-02-19:
> Would you accept a patch to fix it?

Yes, please send a PR.

yashi, 2023-02-20:
It seems there are many tools in can-tuils using `signal()`.  Before converting everything, I want to make sure what I intend to do.  To show off, I'm attaching a diff.

The reason I `raise()` after the cleanup is that I thought exiting the loop was the intention.  If not, we don't actually need to handle the signals because every resource `candump` uses can be cleaned by the kernel.

Util-linux, OTOH, `raise()`s in the signal handler.  That's another way to do it.  It's using termios, which can't be restored by the kernel.  Thus, it's restoring it in the signal handler.
https://sources.debian.org/src/util-linux/2.38.1-5/disk-utils/fsck.minix.c/?hl=170#L163 

So, would you like
a) to wait for a loop (attached patch), or
b) to exit right away without handling the signals and let kernel clean up.

```diff
diff --git a/candump.c b/candump.c
index 82f75b1..036a7dd 100644
--- a/candump.c
+++ b/candump.c
@@ -118,6 +118,7 @@ static const char extra_m_info[4][4] = { "- -", "B -", "- E", "B E" };
 extern int optind, opterr, optopt;
 
 static volatile int running = 1;
+static volatile sig_atomic_t signal_num;
 
 static void print_usage(void)
 {
@@ -169,6 +170,7 @@ static void print_usage(void)
 static void sigterm(int signo)
 {
        running = 0;
+       signal_num = signo;
 }
 
 static int idx2dindex(int ifidx, int socket)
@@ -861,5 +863,10 @@ int main(int argc, char **argv)
        if (log)
                fclose(logfile);
 
+       if (signal_num) {
+               signal(signal_num, SIG_DFL);
+               raise(signal_num);
+       }
+
        return 0;
 }
```

yashi, 2023-02-20:
#54 and #38 are partially related.

hartkopp, 2023-02-20:
I would simply expect :
```
/* following bash rules for simple commands */
if (signal_num)
        return 128 + signal_num;
```
So why do you think `candump` should raise a signal itself?

yashi, 2023-02-20:
To run the kernel's default handler. 

It shouldn't matter for this particular case but that's [the recommended way](https://www.gnu.org/software/libc/manual/html_mono/libc.html#Termination-in-Handler) by the GNU C Library Reference Manual.

hartkopp, 2023-02-20:
Yes. We should not over-engineer the exit process here :-)

Would you like to send a PR starting with your two changes and ending with my simple return check?

closed 2023-03-22 by marckleinebudde