linux-can/can-utils #410: SLCAN and CAN-FD - Follow up

KhazAkar, 2023-02-21:
Hi!
I know, that there was similar threads, but as of today, we have at least 2 devices which use serial port as main communication channel and claim CAN FD support:
https://ucandevices.github.io/cfuc.html
https://canable.io/ (CANable 2.0)
Having support for CAN FD in SLCAN have slightly more sense today than in 2020.

I was scavenging through the internet to find really low cost CAN-FD capable interfaces (sorry, 200+€ for hobby is not low cost) which supports Linux and windows and those are the only ones I was able to find. Having CAN FD support in SLCAN would allow to make use of full potential of those devices, especially since newer cars have more and more CAN FD in them, for example it can be used for entry level Reverse Engineering toolkit or any other thing somebody would want to use for.

Thanks,
Damian

marckleinebudde, 2023-02-22:
Since the Linux driver "gs_usb" supports CAN-FD, it makes even less sense to use SLCAN from my point of view. It has no flow control, no proper bitrate configuration, no error handling and CAN-FD is not standardized.

If you want to enhance the Linux driver, go ahead. Have a look at this discussion: https://lore.kernel.org/all/B1BB0B55-5478-4C06-AAC2-B27AB986755B@gmail.com

KhazAkar, 2023-02-22:
If so, then I think at least ucandevices CFUC module, which I bought, needs porting to candleLight. 
For enhancing driver - sure, just need to learn bit more C before digging through code. Do you have some resources how to start hacking around for beginners? Would be great :)

Cheers!

marckleinebudde, 2023-02-22:
The easiest thing would be to run the candlelight firmware on supported hardware. Recompile the firmware, setup debugger, etc.... If that all works, you can jump to a new board :)

marckleinebudde, 2023-10-03:
I'll close this issue, if you want to hack on the candlelight FW, reopen this issue or open a new one on https://github.com/candle-usb/candleLight_fw.

closed 2023-10-03 by marckleinebudde