linux-can/can-utils #411: can0: Failed to set CAN interface configurations: Protocol error

smilecaozheng, 2023-02-22:
My 3D printer is connected to UBUNTU_22.04 through canbus.
It has been used normally before, but now it must be restarted to get can data,and it cannot be hot-plug.
`klipper@klipper-iWork8:~$ sudo systemctl status systemd-networkd
● systemd-networkd.service - Network Configuration
     Loaded: loaded (/lib/systemd/system/systemd-networkd.service; enabled; vendor preset: enabled)
     Active: active (running) since Thu 2023-02-23 00:22:23 CST; 1s ago
TriggeredBy: ● systemd-networkd.socket
       Docs: man:systemd-networkd.service(8)
   Main PID: 2984 (systemd-network)
     Status: "Processing requests..."
      Tasks: 1 (limit: 2200)
     Memory: 1.3M
        CPU: 191ms
     CGroup: /system.slice/systemd-networkd.service
             └─2984 /lib/systemd/systemd-networkd

2月 23 00:22:23 klipper-iWork8 systemd-networkd[2984]: wlan0: Link UP
2月 23 00:22:23 klipper-iWork8 systemd-networkd[2984]: wlan0: Gained carrier
2月 23 00:22:23 klipper-iWork8 systemd-networkd[2984]: lo: Link UP
2月 23 00:22:23 klipper-iWork8 systemd-networkd[2984]: lo: Gained carrier
2月 23 00:22:23 klipper-iWork8 systemd-networkd[2984]: wlan0: Gained IPv6LL
2月 23 00:22:23 klipper-iWork8 systemd-networkd[2984]: Enumeration completed
2月 23 00:22:23 klipper-iWork8 systemd[1]: Started Network Configuration.
2月 23 00:22:23 klipper-iWork8 systemd-networkd[2984]: wlan0: Connected WiFi access point: 1613 (00:00:00:00:00:00)
2月 23 00:22:23 klipper-iWork8 systemd-networkd[2984]: can0: Failed to set CAN interface configurations: Protocol error
2月 23 00:22:23 klipper-iWork8 systemd-networkd[2984]: can0: Failed
`

marckleinebudde, 2023-02-23:
Can you show us the output og `networkctl status can0` and the content of the `Link File` and the `Network File`.

smilecaozheng, 2023-02-23:
As follows.

```
klipper@klipper-iWork8:~$ networkctl status can0
● 3: can0                                                              
             Link File: /usr/lib/systemd/network/73-usb-net-by-mac.link
          Network File: /etc/systemd/network/80-can.network
                  Type: can
                 State: carrier (configured)
          Online state: offline
                  Path: pci-0000:00:14.0-usb-0:1:1.0
                Driver: gs_usb
                Vendor: OpenMoko, Inc.
                 Model: Geschwister Schneider CAN adapter
                   MTU: 16
                 QDisc: pfifo_fast
  Queue Length (Tx/Rx): 1/1
     Activation Policy: up
   Required For Online: yes

2月 23 17:42:04 klipper-iWork8 systemd-networkd[401]: can0: Link UP
2月 23 17:42:04 klipper-iWork8 systemd-networkd[401]: can0: Gained carrier
```
```

  GNU nano 6.2                               /usr/lib/systemd/network/73-usb-net-by-mac.link                                        
[Match]
Path=*-usb-*

[Link]
NamePolicy=mac

TransmitQueueLength=1024

```
```
  GNU nano 6.2                                     /etc/systemd/network/80-can.network                                              
[Match]
Name=can0

[CAN]
BitRate=500000
RestartSec=100ms
```

marckleinebudde, 2023-02-23:
Please enable logging in `systemd-networkd`: `systemctl edit systemd-networkd.service` will open an editor, please add:
```
[Service]
Environment=SYSTEMD_LOG_LEVEL=debug
```
Then remove the USB adapter, start logging: `journalctl -u systemd-networkd -f` in one terminal, do a `systemctl restart systemd-networkd` in another. Wait 10 seconds or so until the log is more or less quiet, the plug your USB CAN adapter. Send the log output.

smilecaozheng, 2023-02-23:
[image]
There is a lot of information。Sorry, but thank you very much

```
klipper@klipper-iWork8:~$ sudo systemctl edit systemd-networkd.service
klipper@klipper-iWork8:~$ journalctl -u systemd-networkd -f
2月 23 18:21:37 klipper-iWork8 systemd-networkd[396]: lo: Link UP
2月 23 18:21:37 klipper-iWork8 systemd-networkd[396]: lo: Gained carrier
2月 23 18:21:37 klipper-iWork8 systemd-networkd[396]: Enumeration completed
2月 23 18:21:37 klipper-iWork8 systemd[1]: Started Network Configuration.
2月 23 18:21:39 klipper-iWork8 systemd-networkd[396]: can0: Link UP
2月 23 18:21:39 klipper-iWork8 systemd-networkd[396]: can0: Gained carrier
2月 23 18:21:42 klipper-iWork8 systemd-networkd[396]: wlan0: Link UP
2月 23 18:21:52 klipper-iWork8 systemd-networkd[396]: wlan0: Connected WiFi access point: 1613 (00:00:00:00:00:00)
2月 23 18:21:52 klipper-iWork8 systemd-networkd[396]: wlan0: Gained carrier
2月 23 18:21:53 klipper-iWork8 systemd-networkd[396]: wlan0: Gained IPv6LL
2月 23 18:50:18 klipper-iWork8 systemd[1]: Stopping Network Configuration...
2月 23 18:50:18 klipper-iWork8 systemd[1]: systemd-networkd.service: Deactivated successfully.
2月 23 18:50:18 klipper-iWork8 systemd[1]: Stopped Network Configuration.
2月 23 18:50:18 klipper-iWork8 systemd[1]: Starting Network Configuration...
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: SELinux enabled state cached to: disabled
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: sd-device-monitor: Failed to stat PID1's netns, ignoring: No such file or directory
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: Bus bus-api-network: changing state UNSET → OPENING
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: sd-bus: starting bus bus-api-network by connecting to /run/dbus/system_bus_socket...
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: Bus bus-api-network: changing state OPENING → AUTHENTICATING
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: Registering bus object implementation for path=/org/freedesktop/LogControl1 iface=org.freedesktop.LogControl1
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: Loaded timestamp for '/etc/systemd/network'.
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: Loaded timestamp for '/run/systemd/network'.
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: No virtualization found in DMI vendor table.
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: Unable to read /sys/firmware/dmi/entries/0-0/raw, using the virtualization information found in DMI vendor table, ignoring: No such file or directory
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: UML virtualization not found in /proc/cpuinfo.
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: No virtualization found in CPUID
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: Virtualization XEN not found, /proc/xen does not exist
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: This platform does not support /proc/device-tree
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: This platform does not support /proc/sysinfo
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: Found VM virtualization none
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: /usr/lib/systemd/network/80-container-host0.network: Conditions in the file do not match the system environment, skipping.
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: wlan0: Link 3 added
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: wlan0: Saved original MTU 1500 (min: 68, max: 1500)
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: wlan0: Saved hardware address: 54:e4:bd:48:a2:d1
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: wlan0: Flags change: +UP +LOWER_UP +RUNNING +MULTICAST +BROADCAST
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: wlan0: Link UP
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: wlan0: Gained carrier
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: wlan0: udev initialized link
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: wlan0: State changed: pending -> initialized
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: can0: Permanent MAC address not found for new device, continuing without: Operation not supported
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: can0: Link 2 added
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: can0: Saved original MTU 16 (min: 0, max: 4294967295)
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: can0: Flags change: +UP +LOWER_UP +RUNNING +NOARP +ECHO
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: can0: Link UP
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: can0: Gained carrier
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: can0: udev initialized link
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: can0: State changed: pending -> initialized
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: lo: Failed to get driver, continuing without: Operation not supported
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: lo: Link 1 added
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: lo: Saved original MTU 65536 (min: 0, max: 4294967295)
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: lo: Saved hardware address: 00:00:00:00:00:00
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: lo: Flags change: +LOOPBACK +UP +LOWER_UP +RUNNING
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: lo: Link UP
2月 23 18:50:18 klipper-iWork8 systemd-networkd[2271]: lo: Gained carrier
[... cut ...]
2月 23 18:51:58 klipper-iWork8 systemd-networkd[2271]: wlan0: link_check_ready(): link is in unmanaged state.
2月 23 18:51:58 klipper-iWork8 systemd-networkd[2271]: wlan0: Remembering foreign route: dst: n/a, src: n/a, gw: fe80::faaf:5ff:fe6b:e803, prefsrc: n/a, scope: global, table: main(254), proto: ra, type: unicast, nexthop: 0, priority: 20600
2月 23 18:51:58 klipper-iWork8 systemd-networkd[2271]: wlan0: Forgetting route: dst: n/a, src: n/a, gw: fe80::faaf:5ff:fe6b:e803, prefsrc: n/a, scope: global, table: main(254), proto: ra, type: unicast, nexthop: 0, priority: 600

```

smilecaozheng, 2023-02-23:
In addition, due to the limitation of bios, I used 32bit boot for efi and installed ubuntu22.04 system.

marckleinebudde, 2023-02-23:
> In addition, due to the limitation of bios, I used 32bit boot for efi and installed ubuntu22.04 system.

That should be no problem.....

smilecaozheng, 2023-02-23:
I also had this problem before. I used the control variable method to reinstall the system and refresh the MCU firmware. Some variables were excluded, but the problem still occurred

marckleinebudde, 2023-02-23:
You can remove the debug from the `systemd-network` by commenting out the `Environment` in `systemctl edit systemd-networkd`:

```
[Service]
#Environment=SYSTEMD_LOG_LEVEL=debug
```

Note: If you remove everything, `systemctl` will ignore the edit.

Ok, let's try this instead of using `systemd-networkd`:

```
sudo ip link set dev can0 down
sudo ip link set can0 txqueuelen 1024 up type can bitrate 500000 restart-ms 100 berr-reporting off
```

smilecaozheng, 2023-02-23:
Additional content
After I commented out this point, I tried again, and the result is as follows

```
  GNU nano 6.2                                                                                           /etc/systemd/network/80-can.network                                                                                                     
[Match]
Name=can0

[CAN]
BitRate=500000
#RestartSec=100ms
```
```
klipper@klipper-iWork8:~$ sudo nano /etc/systemd/network/80-can.network
klipper@klipper-iWork8:~$ sudo ip link set dev can0 down
klipper@klipper-iWork8:~$ networkctl status can0
● 2: can0                                                              
             Link File: /usr/lib/systemd/network/73-usb-net-by-mac.link
          Network File: /etc/systemd/network/80-can.network
                  Type: can
                 State: off (failed)
          Online state: offline
                  Path: pci-0000:00:14.0-usb-0:1:1.0
                Driver: gs_usb
                Vendor: OpenMoko, Inc.
                 Model: Geschwister Schneider CAN adapter
                   MTU: 16
                 QDisc: pfifo_fast
  Queue Length (Tx/Rx): 1/1
     Activation Policy: up
   Required For Online: yes

2月 23 20:59:10 klipper-iWork8 systemd-networkd[394]: can0: Link UP
2月 23 20:59:10 klipper-iWork8 systemd-networkd[394]: can0: Gained carrier
2月 23 21:01:31 klipper-iWork8 systemd-networkd[394]: can0: Link DOWN
2月 23 21:01:31 klipper-iWork8 systemd-networkd[394]: can0: Lost carrier
2月 23 21:01:46 klipper-iWork8 systemd-networkd[394]: can0: Link UP
2月 23 21:01:46 klipper-iWork8 systemd-networkd[394]: can0: Gained carrier
2月 23 21:13:39 klipper-iWork8 systemd-networkd[394]: can0: Link DOWN
2月 23 21:13:39 klipper-iWork8 systemd-networkd[394]: can0: Lost carrier
2月 23 21:14:56 klipper-iWork8 systemd-networkd[2056]: can0: Failed to set CAN interface configurations: Protocol error
2月 23 21:14:56 klipper-iWork8 systemd-networkd[2056]: can0: Failed
klipper@klipper-iWork8:~$ sudo ip link set can0 txqueuelen 1024 up type can bitrate 500000 restart-ms 100 berr-reporting off
RTNETLINK answers: Protocol error
```

smilecaozheng, 2023-02-23:
The test results are as follows. I'm really sorry for letting you this trouble.
If it can be solved, I would appreciate it very much

```
klipper@klipper-iWork8:~$ sudo systemctl disable systemd-networkd
Removed /etc/systemd/system/network-online.target.wants/systemd-networkd-wait-online.service.
Removed /etc/systemd/system/sockets.target.wants/systemd-networkd.socket.
Removed /etc/systemd/system/sysinit.target.wants/systemd-network-generator.service.
Removed /etc/systemd/system/multi-user.target.wants/systemd-networkd.service.
Removed /etc/systemd/system/dbus-org.freedesktop.network1.service.
klipper@klipper-iWork8:~$ sudo mv -f /etc/systemd/network/80-can.network /etc/systemd/network/80-can.network-disabled
klipper@klipper-iWork8:~$ ls^C
klipper@klipper-iWork8:~$ ls /etc/systemd/network/
80-can.network-disabled
klipper@klipper-iWork8:~$ reboot
klipper@klipper-iWork8:~$ Connection closing...Socket close.

Connection closed by foreign host.

Disconnected from remote host(klipper) at 21:42:11.

Last login: Thu Feb 23 21:30:35 2023 from 192.168.0.6
klipper@klipper-iWork8:~$ ip --details -s -s -s link show can0
2: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 1024
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state STOPPED restart-ms 0 
	  gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp-inc 1
	  clock 48000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 parentbus usb parentdev 1-1:1.0 
    RX:  bytes packets errors dropped  missed   mcast           
             0       0      0       0       0       0 
    RX errors:  length    crc   frame    fifo overrun           
                     0      0       0       0       0 
    TX:  bytes packets errors dropped carrier collsns           
             0       0      0       0       0       0 
    TX errors: aborted   fifo  window heartbt transns
                     0      0       0       0       0 
klipper@klipper-iWork8:~$ sudo ip link set can0 txqueuelen 1024 up type can bitrate 500000 restart-ms 100 berr-reporting off

klipper@klipper-iWork8:~$ ip --details -s -s -s link show can0
2: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 16 qdisc pfifo_fast state UP mode DEFAULT group default qlen 1024
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state ERROR-ACTIVE restart-ms 100 
	  bitrate 500000 sample-point 0.875 
	  tq 125 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
	  gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp-inc 1
	  clock 48000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 parentbus usb parentdev 1-1:1.0 
    RX:  bytes packets errors dropped  missed   mcast           
           943     130      0       0       0       0 
    RX errors:  length    crc   frame    fifo overrun           
                     0      0       0       0       0 
    TX:  bytes packets errors dropped carrier collsns           
             0       0      0       0       0       0 
    TX errors: aborted   fifo  window heartbt transns
                     0      0       0       0       1 

klipper@klipper-iWork8:~$ sudo ip link set dev can0 down
klipper@klipper-iWork8:~$ ip --details -s -s -s link show can0
2: can0: <NOARP,ECHO> mtu 16 qdisc pfifo_fast state DOWN mode DEFAULT group default qlen 1024
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state ERROR-ACTIVE restart-ms 100 
	  bitrate 500000 sample-point 0.875 
	  tq 125 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
	  gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp-inc 1
	  clock 48000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 parentbus usb parentdev 1-1:1.0 
    RX:  bytes packets errors dropped  missed   mcast           
         21449    2895      0       0       0       0 
    RX errors:  length    crc   frame    fifo overrun           
                     0      0       0       0       0 
    TX:  bytes packets errors dropped carrier collsns           
          1664     345      0      10       0       0 
    TX errors: aborted   fifo  window heartbt transns
                    10      0       0       0       1 
klipper@klipper-iWork8:~$ sudo ip link set can0 txqueuelen 1024 up type can bitrate 500000 restart-ms 100 berr-reporting off
RTNETLINK answers: Protocol error

```

smilecaozheng, 2023-02-23:
[image]

Hardware connection

marckleinebudde, 2023-02-23:
So this boils down to:
```shell
klipper@klipper-iWork8:~$ ip --details -s -s -s link show can0
# DOWN: OK!
klipper@klipper-iWork8:~$ sudo ip link set can0 txqueuelen 1024 up type can bitrate 500000 restart-ms 100 berr-reporting off
# DOWN -> UP
klipper@klipper-iWork8:~$ ip --details -s -s -s link show can0
# UP: OK!

klipper@klipper-iWork8:~$ sudo ip link set dev can0 down
# UP -> DOWN
klipper@klipper-iWork8:~$ ip --details -s -s -s link show can0
# DOWN: OK!

klipper@klipper-iWork8:~$ sudo ip link set can0 txqueuelen 1024 up type can bitrate 500000 restart-ms 100 berr-reporting off
RTNETLINK answers: Protocol error
# DOWN -> UP: FAIL!
```

Does it always Fail after after the 2nd UP?

Can you unplug and replug the USB after the device is in the fail state?

smilecaozheng, 2023-02-23:
I think this may have to be tested tomorrow.

marckleinebudde, 2023-02-23:
ok, gn8

smilecaozheng, 2023-02-24:
I tried it. It's not like yesterday. It is strange that the error has disappeared and cannot be reproduced today.
I noticed that the power supply mode of my PC has changed today. I temporarily determined that it was a hardware problem, such as USB signal interference.
I'm going back to returning the systemd-network method to connect.
If I find that this problem cannot be solved by myself, please allow me to contact you again.
Let's close this issue.
Finally, thank you for your help in your busy schedule.
The highest respect!!!

marckleinebudde, 2023-02-24:
You're welcome!

closed 2023-02-24 by marckleinebudde