linux-can/can-utils #412: SLCAN - Why is BTR limited to 6 characters?

rnd-ash, 2023-02-23:
Just a question, as seen here:
https://github.com/linux-can/can-utils/blob/master/slcand.c#L247

It seems the limit to the BTR reg is 6 characters.

Why is this exactly?

A lot of ATMEL MCUs use a 32bit BTR register which (If printed in hex) would need 8 characters. This makes it rather difficult to set a custom baud rate.

I am happy to open a PR to change this if needed (To support hex printed 32bit integers in BTR reg)

marckleinebudde, 2023-02-23:
The [SLCAN "spec" by LAWICEL](http://www.can232.com/docs/canusb_manual.pdf) says `sxxyy[CR]`, which is just 4 characters. Don't know why we limit to 6 here. @hartkopp do you know where the `6` come from? I don't mind a patch that increases that to 8 chars.

On the other hand, think about using a proper protocol, like the [candlelight](https://github.com/candle-usb/candleLight_fw/blob/master/include/gs_usb.h) it has a proper Linux driver, supports multiple channels, proper bit rate configuration and even CAN-FD.

hartkopp, 2023-02-23:
The SLCAN driver is created to support different SLCAN flavours, see:
https://github.com/linux-can/can-doc/blob/master/slcan/SLCAN-API.pdf

And the CANHack tool supported 6 chars at that time. When the Atmel MCUs need 8 chars I have no objections to increase this value to 8.

@rnd-ash Please feel free to send a PR (as you already phrased a proper commit description).

rnd-ash, 2023-02-23:
Thanks @hartkopp. PR requested.

PR #413 (2023-02-23): SLCAN - Increase BTR char limit to 8 for 32bit BTR
Resolves #412
closed 2023-02-23 by hartkopp