linux-can/can-utils #415: Receive multiple j1939 messages using multiple sockets

maniu1994, 2023-03-15:
My goal is to receive j1939 messages with two different PGNs from two different addresses. 
As proposed in #282, I use multiple (two) sockets for this case and set the appropriate filters.

Unfortunately, the program does not work all the time as expected. Usually, right after building the code, both messages are received and printed to the console. If I kill the program and start it again, either both or only one or no message is received anymore.

Here is a minimal example:

```
#include <stdio.h>
#include <net/if.h>
#include <linux/can/j1939.h>

int main(int argc, char * argv[])
{
  int value = 1;
  int maxfd = 0;
  int ret;
  int sock1, sock2;
  struct sockaddr_can baddr1, baddr2;
  struct j1939_filter filter1, filter2;
  fd_set readfds;
  u_int8_t data[8];

  baddr1.can_family = AF_CAN;
  baddr1.can_ifindex = if_nametoindex("vcan0");
  baddr1.can_addr.j1939.name = J1939_NO_NAME;
  baddr1.can_addr.j1939.pgn = J1939_NO_PGN;
  baddr1.can_addr.j1939.addr = J1939_NO_ADDR;

  baddr2.can_family = AF_CAN;
  baddr2.can_ifindex = if_nametoindex("vcan0");
  baddr2.can_addr.j1939.name = J1939_NO_NAME;
  baddr2.can_addr.j1939.pgn = J1939_NO_PGN;
  baddr2.can_addr.j1939.addr = J1939_NO_ADDR;

  filter1.addr_mask = ~0;
  filter1.pgn_mask = ~0;
  filter1.addr = 0x03;
  filter1.pgn = 0xf001;

  filter2.addr_mask = ~0;
  filter2.pgn_mask = ~0;
  filter2.addr = 0x04;
  filter2.pgn = 0xf002;

  sock1 = socket(PF_CAN, SOCK_DGRAM, CAN_J1939);
  setsockopt(sock1, SOL_SOCKET, SO_BROADCAST, &value, sizeof(value));
  setsockopt(sock1, SOL_CAN_J1939, SO_J1939_FILTER, &filter1, sizeof(filter1));
  bind(sock1, (struct sockaddr *)&baddr1, sizeof(baddr1));

  sock2 = socket(PF_CAN, SOCK_DGRAM, CAN_J1939);
  setsockopt(sock2, SOL_SOCKET, SO_BROADCAST, &value, sizeof(value));
  setsockopt(sock2, SOL_CAN_J1939, SO_J1939_FILTER, &filter2, sizeof(filter2));
  bind(sock2, (struct sockaddr *)&baddr2, sizeof(baddr2));

  maxfd = sock1 > maxfd ? sock1 : maxfd;
  maxfd = sock2 > maxfd ? sock2 : maxfd;

  while(true)
  {
    FD_ZERO(&readfds);
    FD_SET(sock1, &readfds);
    FD_SET(sock2, &readfds);

    select(maxfd + 1, &readfds, NULL, NULL, NULL);

    if(FD_ISSET(sock1, &readfds)) 
    {
      ret = recv(sock1, data, sizeof(data), 0);
      printf("received %d bytes\n", ret);
    }

    if(FD_ISSET(sock2, &readfds)) 
    {
      ret = recv(sock2, data, sizeof(data), 0);
      printf("received %d bytes\n", ret);
    }
  }

  return 0;
}
```

It prints the number of received bytes from two messages (pgn1: 0xf001, sa1: 0x03, pgn2: 0xf002, sa2: 0x04) from virtual can vcan0. Since both are PDU2 (broadcast) messages, I called setsockopt() with SO_BROADCAST to be able to receive them. The socket fds are handled using select() within the infinite loop (I also tried using poll() and non-blocking sockets, so this seems not to be the problem). 

In another terminal I send the required messages with the following commands:
`cansend vcan0 18f00103#0123456789ABCDEF` 
`cansend vcan0 18f00204#0123456789ABCDEF`

Hopefully, someone can reproduce the problem and help me. Many thanks in advance.

olerem, 2023-03-16:
Hi  @maniu1994 , thank you for your report. I'll take a look on this this Friday or the next week.

olerem, 2023-03-16:
Before I'll do deeper investigate, please change you example to a poll() or epoll() based. select() do not supports file descriptors that are negative or equal to or larger than FD_SETSIZE (1024).

olerem, 2023-03-16:
If you like, you can provide simple testing application for your use case. So it can be integrated for regular testing.

maniu1994, 2023-03-16:
Sure, here is the working application using poll() instead of select():

```
#include <stdio.h>
#include <string.h>
#include <net/if.h>
#include <linux/can/j1939.h>
#include <poll.h>

int main(int argc, char * argv[])
{
  int value = 1;
  int ret;
  struct sockaddr_can baddr1, baddr2;
  struct j1939_filter filter1, filter2;
  struct pollfd pfds[2];
  u_int8_t data[8];

  memset(&filter1, 0, sizeof(filter1));
  memset(&filter2, 0, sizeof(filter2));

  baddr1.can_family = AF_CAN;
  baddr1.can_ifindex = if_nametoindex("vcan0");
  baddr1.can_addr.j1939.name = J1939_NO_NAME;
  baddr1.can_addr.j1939.pgn = J1939_NO_PGN;
  baddr1.can_addr.j1939.addr = J1939_NO_ADDR;

  baddr2.can_family = AF_CAN;
  baddr2.can_ifindex = if_nametoindex("vcan0");
  baddr2.can_addr.j1939.name = J1939_NO_NAME;
  baddr2.can_addr.j1939.pgn = J1939_NO_PGN;
  baddr2.can_addr.j1939.addr = J1939_NO_ADDR;

  filter1.addr_mask = ~0;
  filter1.pgn_mask = ~0;
  filter1.addr = 0x03;
  filter1.pgn = 0xf001;

  filter2.addr_mask = ~0;
  filter2.pgn_mask = ~0;
  filter2.addr = 0x04;
  filter2.pgn = 0xf002;

  pfds[0].fd = socket(PF_CAN, SOCK_DGRAM, CAN_J1939);
  setsockopt(pfds[0].fd, SOL_SOCKET, SO_BROADCAST, &value, sizeof(value));
  setsockopt(pfds[0].fd, SOL_CAN_J1939, SO_J1939_FILTER, &filter1, sizeof(filter1));
  bind(pfds[0].fd, (struct sockaddr *)&baddr1, sizeof(baddr1));

  pfds[1].fd = socket(PF_CAN, SOCK_DGRAM, CAN_J1939);
  setsockopt(pfds[1].fd, SOL_SOCKET, SO_BROADCAST, &value, sizeof(value));
  setsockopt(pfds[1].fd, SOL_CAN_J1939, SO_J1939_FILTER, &filter2, sizeof(filter2));
  bind(pfds[1].fd, (struct sockaddr *)&baddr2, sizeof(baddr2));

  pfds[0].events = POLLIN;
  pfds[1].events = POLLIN;

  while(true)
  {
    poll(pfds, 2, -1);

    if(pfds[0].revents & POLLIN) 
    {
      ret = recv(pfds[0].fd, data, sizeof(data), 0);
      printf("received %d bytes\n", ret);
    }

    if(pfds[1].revents & POLLIN) 
    {
      ret = recv(pfds[1].fd, data, sizeof(data), 0);
      printf("received %d bytes\n", ret);
    }
  }

  return 0;
}
```

closed 2023-03-17 by maniu1994