linux-can/can-utils #420: CAN-Interface not showing up under armhf

NiDaKuDE, 2023-03-27:
Hey there,
I am currently facing the issue, that my "Kvaser BlackBird SemiPro" does not show up ("ip addr" does not list the canX interfaces), when plugged into a RaspberryPi 3 or 4. However it works just fine using an x86 machine. Another "no name" interface works fine as well on all three tested devices.

I simply installed the can-utils package via apt and ran an "apt-get upgrade" to be safe. Additionally I tried to rule out a power issue by connecting both Raspberrys to a beefy power supply (no voltage errors were logged).

Since 
    » I did nothing else on my x86-Machine to get this to work
    » none of my web searches revealed someone having the same or a similar problem
    » there are no obvious errors to trace logged under the standard /var/log path
    » I talked to @hartkopp some days ago, and he recommended to create a new issue, if the problem persists
I decided to open this issue (:

Below I attached some information of the devices as well as the output of "lsusb" and the "kern.log" when connected to the Pi4.

### x86-Desktop
```
OS: Debian Bullseye
Kernel: 5.10.0-18-amd64
DPKG: ii  can-utils   2020.11.0-1  amd64   SocketCAN userspace utilities and tools
```

### RaspberryPi 3
```
OS: Debian Bullseye
Kernel: 5.15.84-v7+
DPKG: ii  can-utils   2020.11.0-1  armhf   SocketCAN userspace utilities and tools
```

### RaspberryPi 4
```
OS: Debian Buster
Kernel: 5.10.103-v7l+
DPKG: ii  can-utils   2018.02.0-1  armhf   SocketCAN userspace utilities and tools
```

/var/log/kern.log
```
[   20.093147] fuse: init (API version 7.32)
[  350.989971] usb 1-1.2: new high-speed USB device number 3 using xhci_hcd
[  351.121871] usb 1-1.2: New USB device found, idVendor=0bfd, idProduct=001d, bcdDevice= 0.01
[  351.121894] usb 1-1.2: New USB device strings: Mfr=0, Product=2, SerialNumber=0
[  351.121913] usb 1-1.2: Product: Kvaser BlackBird SemiPro
```
<details><summary>Output of lsusb</summary>

```
Bus 001 Device 003: ID 0bfd:001d Kvaser AB 
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass          255 Vendor Specific Class
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0        64
  idVendor           0x0bfd Kvaser AB
  idProduct          0x001d 
  bcdDevice            0.01
  iManufacturer           0 
  iProduct                2 Kvaser BlackBird SemiPro
  iSerial                 0 
  bNumConfigurations      2
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              400mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass         0 
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength       0x0027
    bNumInterfaces          1
    bConfigurationValue     2
    iConfiguration          0 
    bmAttributes         0xc0
      Self Powered
    MaxPower              400mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           3
      bInterfaceClass         0 
      bInterfaceSubClass      0 
      bInterfaceProtocol      0 
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x01  EP 1 OUT
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x82  EP 2 IN
        bmAttributes            2
          Transfer Type            Bulk
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0200  1x 512 bytes
        bInterval               1
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Device Status:     0x0000
  (Bus Powered)
```

</details>

marckleinebudde, 2023-03-27:
I think the raspi kernel is lacking the `kvaser_usb` module. Re-compile your kernel with `CONFIG_CAN_KVASER_USB` enabled or as a module.

hartkopp, 2023-03-27:
First I was not sure whether product ID 0x001d is supported.

```
  idVendor           0x0bfd Kvaser AB
  idProduct          0x001d 
```

But 0x001d is 29 decimal and can be found here:
https://elixir.bootlin.com/linux/latest/source/drivers/net/can/usb/kvaser_usb/kvaser_usb_core.c#L53

So I would also assume that the module is simply missing on that system.

@marckleinebudde I tend to post a patch that brings the USB IDs into a hex representation. Decimal USB IDs is really uncommon.

hartkopp, 2023-03-27:
And yes, CONFIG_CAN_KVASER_USB is missing here:

https://github.com/raspberrypi/linux/blob/rpi-5.15.y/arch/arm/configs/bcm2711_defconfig#L415

NiDaKuDE, 2023-03-29:
> I think the raspi kernel is lacking the `kvaser_usb` module. Re-compile your kernel with `CONFIG_CAN_KVASER_USB` enabled or as a module.

Did not think about missing modules. This seems to have solved the problem.

Thank you!