linux-can/can-utils #425: sending frame, receiving error frame

illegal-instruction-co, 2023-04-11:
i am trying to test can interfaces. scenerio is like this:

2 terminal windows are on the screen.

1. terminal: 
candump any,0~0,#FFFFFFFF -e -x -ta

2. terminal
cansend  can0 123#DEADBEEF
cansend  can1 123#DEADBEEF
cansend  can2 123#DEADBEEF

whenever i send frame to can0 or can2 there is no error frames.  but for same message, can1 output on 1.terminal:
 (1681229878.879041)  can1  RX - -  20000004   [8]  00 00 00 00 00 00 00 00   ERRORFRAME
	controller-problem{}

I am really not sure why it is happening

marckleinebudde, 2023-04-12:
* Which CAN controllers are you using?
* Which Kernel are you using?
* Please describe your physical CAN bus layout

illegal-instruction-co, 2023-04-12:
> * Which CAN controllers are you using?
i will edit here
> * Which Kernel are you using?
i will edit here
> * Please describe your physical CAN bus layout
CAN-controller->PC->Device

illegal-instruction-co, 2023-04-16:
Closed the issue because i explored it is happend because of wrong hw setup.

closed 2023-04-16 by illegal-instruction-co