linux-can/can-utils #427: Mcp2515 And Raspiberry Pi 4B

fy8207345, 2023-04-26:
I have a mcp2515 module and Raspiberry Pi 4B. And I installed the can-utils. Is there A Circuit between the Mcp2515 and Raspberry GPIOs? Thanks.

AndreiCherniaev, 2023-04-30:
> 

> Mcp2515 and Raspberry GPIOs?

Take [it](https://github.com/linux-can/can-utils/issues/427#issue-1684217025)

arturca, 2023-04-30:
If by MCP2515 module you mean PCB that is SPI to CAN interface, which includes MCP2515 (its just a Stand-Alone CAN Controller with SPI Interface - google it) and TJA1050 (5V CAN transiver) you still need to make one adjustment that is described here:
https://forums.raspberrypi.com//viewtopic.php?t=141052.
But if you actually mean the MCP2515 CAN controller module, you can figure out connections based on this link.

marckleinebudde, 2023-10-03:
Seems to be solved, I'll close this issue.

closed 2023-10-03 by marckleinebudde