linux-can/can-utils #428: installing on cent 7 64 bit fails

yannS2016, 2023-04-30:
i runthis command inside build folder
make CFLAGS='-std=c99
i got these errors
j1939cat.c:35:12: error: expected ‘;’ before ‘_min1’
  typeof(x) _min1 = (x);   \
            ^
j1939cat.c:431:13: note: in expansion of macro ‘min’
  buf_size = min(priv->max_transfer, count);
             ^
j1939cat.c:36:12: error: expected ‘;’ before ‘_min2’
  typeof(y) _min2 = (y);   \
            ^
j1939cat.c:431:13: note: in expansion of macro ‘min’
  buf_size = min(priv->max_transfer, count);
             ^
j1939cat.c:37:11: error: ‘_min1’ undeclared (first use in this function)
  (void) (&_min1 == &_min2);  \
           ^
j1939cat.c:431:13: note: in expansion of macro ‘min’
  buf_size = min(priv->max_transfer, count);
             ^
j1939cat.c:37:11: note: each undeclared identifier is reported only once for each function it appears in
  (void) (&_min1 == &_min2);  \
           ^
j1939cat.c:431:13: note: in expansion of macro ‘min’
  buf_size = min(priv->max_transfer, count);
             ^
j1939cat.c:37:21: error: ‘_min2’ undeclared (first use in this function)
  (void) (&_min1 == &_min2);  \
                     ^
j1939cat.c:431:13: note: in expansion of macro ‘min’
  buf_size = min(priv->max_transfer, count);
             ^
j1939cat.c:35:12: error: expected ‘;’ before ‘_min1’
  typeof(x) _min1 = (x);   \
            ^
j1939cat.c:455:13: note: in expansion of macro ‘min’
   to_read = min(buf_size, count);
             ^
j1939cat.c:36:12: error: expected ‘;’ before ‘_min2’
  typeof(y) _min2 = (y);   \
            ^
j1939cat.c:455:13: note: in expansion of macro ‘min’
   to_read = min(buf_size, count);
             ^
make: *** [j1939cat.o] Error 1
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE  -c -o j1939spy.o j1939spy.c
cc   j1939spy.o libj1939.o   -o j1939spy
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE  -c -o j1939sr.o j1939sr.c
cc   j1939sr.o libj1939.o   -o j1939sr
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE  -c -o testj1939.o testj1939.c
cc   testj1939.o libj1939.o   -o testj1939
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE   slcan_attach.c   -o slcan_attach
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE   slcand.c   -o slcand
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE  -c -o asc2log.o asc2log.c
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE  -c -o lib.o lib.c
cc   asc2log.o lib.o   -o asc2log
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE   bcmserver.c   -o bcmserver
cc  calc-bit-timing/can-calc-bit-timing.o  -o can-calc-bit-timing
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE  -c -o canbusload.o canbusload.c
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE  -c -o canframelen.o canframelen.c
cc   canbusload.o canframelen.o   -o canbusload
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE  -c -o candump.o candump.c
cc   candump.o lib.o   -o candump
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE   canfdtest.c   -o canfdtest
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE  -c -o cangen.o cangen.c
cc   cangen.o lib.o   -o cangen
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE  -c -o cansequence.o cansequence.c
cc   cansequence.o lib.o   -o cansequence
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE  -c -o canlogserver.o canlogserver.c
[... 1089 lines cut ...]
include/linux/kernel.h:77:18: error: called object is not a function or function pointer
   (typeof(_mask))(((_reg) & (_mask)) >> __bf_shf(_mask)); \
                  ^
include/linux/kernel.h:24:41: note: in expansion of macro ‘FIELD_GET’
 #define pr_info(...) fprintf(stdout, ## __VA_ARGS__)
                                         ^
mcp251xfd/mcp251xfd-dump.c:92:2: note: in expansion of macro ‘pr_info’
  pr_info("%16s = " fmt "\t\t%s\n", \
  ^
mcp251xfd/mcp251xfd-dump.c:767:3: note: in expansion of macro ‘__dump_mask’
   __dump_mask(hw_tx_obj->flags, MCP251XFD_OBJ_FLAGS, SEQ_MCP2518FD, "0x%06lx", "Sequence (MCP2518)");
   ^
include/linux/kernel.h:77:18: error: called object is not a function or function pointer
   (typeof(_mask))(((_reg) & (_mask)) >> __bf_shf(_mask)); \
                  ^
mcp251xfd/mcp251xfd-dump.c:780:9: note: in expansion of macro ‘FIELD_GET’
   dlc = FIELD_GET(MCP251XFD_OBJ_FLAGS_DLC, hw_rx_obj->flags);
         ^
make: *** [mcp251xfd/mcp251xfd-dump.o] Error 1
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE  -c -o mcp251xfd/mcp251xfd-main.o mcp251xfd/mcp251xfd-main.c
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE  -c -o mcp251xfd/mcp251xfd-regmap.o mcp251xfd/mcp251xfd-regmap.c
cc -std=c99 -I. -Iinclude -DAF_CAN=PF_CAN -DPF_CAN=29 -DSO_RXQ_OVFL=40 -DSCM_TIMESTAMPING_OPT_STATS=54 -DCLOCK_TAI=11 -DSO_TXTIME=61 -DSCM_TXTIME=SO_TXTIME -D_FILE_OFFSET_BITS=64 -D_GNU_SOURCE  -c -o slcanpty.o slcanpty.c
cc   slcanpty.o lib.o   -o slcanpty
make: Target `all' not remade because of errors.
[ess-dev@localhost can-utils]$

marckleinebudde, 2023-04-30:
```
i runthis command inside build folder
make CFLAGS='-std=c99
```

`can-utils` support several build systems, use only one of them:

1. `make`
2. `configure && make`
3. `mkdir build && cd build && cmake -GNinja .. && ninja`

yannS2016, 2023-04-30:
i started by just running make but it return error asking to add -std=c99
after that i got there errors i posted

marckleinebudde, 2023-04-30:
I've pushed a fix to master, please retry.

yannS2016, 2023-04-30:
getting the same issue

cangen.c: In function ‘main’:
cangen.c:797:4: error: ‘for’ loop initial declarations are only allowed in C99 mode
    for (int i = 0; i < frame.len; i++) {
    ^
cangen.c:797:4: note: use option -std=c99 or -std=gnu99 to compile your code
make: *** [cangen.o] Error 1

marckleinebudde, 2023-04-30:
Sorry, try again.

closed 2023-04-30 by marckleinebudde