linux-can/can-utils #433: Limit Hops fails when value is set to anything but 1

derek-will, 2023-05-26:
I am currently working on adding support for configuring the CAN Gateway into [SocketCAN#](https://github.com/derek-will/SocketCANSharp) and I have run into a minor snag. 

Whenever I try to set Limit Hops to anything other than 1, I receive back EINVAL (errno 22) in response.

```
NL Request: 2C-00-00-00-18-00-05-00-00-00-00-00-00-00-00-00-1D-01-01-00-08-00-09-00-04-00-00-00-08-00-0A-00-05-00-00-00-05-00-0D-00-02-00-00-00

NL ACK: 40-00-00-00-02-00-00-00-00-00-00-00-65-10-01-00-EA-FF-FF-FF-2C-00-00-00-18-00-05-00-00-00-00-00-00-00-00-00-1D-01-01-00-08-00-09-00-04-00-00-00-08-00-0A-00-05-00-00-00-05-00-0D-00-02-00-00-00
```

If I try the official `cangw` utility, it behaves the same:

```
sudo cangw -A -s vcan0 -d vcan1 -l 2
netlink error -22 (Invalid argument)
```

I am running Ubuntu 22.04 and my kernel is 5.18.10-76051810-generic. The process has the `CAP_NET_ADMIN` capability. I have not run into issues with setting any other attributes. What am I missing here? Thanks!

hartkopp, 2023-05-27:
The limitation is set in the can-gw module and not in the can-gw rule.

When loading the module try:

```
# rmmod can_gw
# modprobe can_gw max_hops=3
# cat /sys/module/can_gw/parameters/max_hops 
3
```

This enables the hop-limit up to 3 hops in the can-gw rules.

derek-will, 2023-05-28:
Eureka! It works as expected now! I appreciate the support as always! Thanks!

```
NL Request: 2C-00-00-00-18-00-05-00-00-00-00-00-00-00-00-00-1D-01-01-00-08-00-09-00-04-00-00-00-08-00-0A-00-05-00-00-00-05-00-0D-00-02-00-00-00
NL ACK: 24-00-00-00-02-00-00-01-00-00-00-00-05-93-01-00-00-00-00-00-2C-00-00-00-18-00-05-00-00-00-00-00-00-00-00-00
```

closed 2023-05-28 by derek-will