linux-can/can-utils #434: Device not found (CAN via MCP2515)

YuBer0, 2023-05-30:
Hi, i'm trying to use my RPI4B to send CAN messages via MCP2515 (modified where i swapped out the TJA1050 for VP230).

I was using it just fine, but i took a few days break, and when i came back the CAN module could not be detected via 
raspberrypi:~ $ ifconfig can0
> can0: error fetching interface information: Device not found

Here are a few of the configurations i have.
Physical connection:
> MCP2515 VCC (RPI pin 17)
> MCP2515 SI (RPI pin 19)
> MCP2515 GND (RPI pin 20)
> MCP2515 SO (RPI pin 21)
> MCP2515 INT (RPI pin 22)
> MCP2515 SCK (RPI pin 23)
> MCP2515 CS (RPI pin 24)

uname -a
> Linux raspberrypi 5.15.84-v7l+ #1613 SMP Thu Jan 5 12:01:26 GMT 2023 armv7l GNU/Linux

in (sudo nano /boot/config.txt)
> dtparam=spi=on
> dtoverlay=mcp2515-can0,cs0_pin=8,oscillator=8000000,interrupt=25

in (raspberrypi:~ $ sudo nano /etc/network/interfaces) 
> source /etc/network/interfaces.d/*
> 
> auto can0
> iface can0 inet manual
>     #pre-up /sbin/ip link set can0 type can bitrate 500000 triple-sampling on restart-ms 100
>     pre-up /sbin/ip link set can0 type can bitrate 500000 restart-ms 100
>     up /sbin/ifconfig can0 up
>     down /sbin/ifconfig can0 down

raspberrypi:~ $ lsmod | grep "can"
> can_dev                32768  1 mcp251x

raspberrypi:~ $ dmesg | grep -i can
> [    9.949427] CAN device driver interface

marckleinebudde, 2023-05-30:
Look for `mcp25` in the `demsg` output.

YuBer0, 2023-05-30:
Here is the output of the command for mcp251x

> [   10.351758] mcp251x spi0.0: MCP251x didn't enter in conf mode after reset
> [   10.352043] mcp251x spi0.0: Probe failed, err=110

marckleinebudde, 2023-05-30:
The mcp251x is not detected. Check cabling, mechanical contacts, power, etc... Are you running the stock raspi kernel?

YuBer0, 2023-05-30:
I see, my kernel came with the raspi image.
my wiring is quite straightforward, for now i've connected the touch display and MCP2515 only.
i have also tested the wires connectivity via a DMM.
[image]

as for wrong configuration with DT overlay, how can i check it?

YuBer0, 2023-05-30:
The wiring to pin can be found on the 1st comment.
Physical wiring can be seen on top image, its via a 7 ribbon wire.
I tried to use the pinout via terminal, not sure if its what you would mean as well.
[image]

> Hi, i'm trying to use my RPI4B to send CAN messages via MCP2515 (modified where i swapped out the TJA1050 for VP230).
>
> I was using it just fine, but i took a few days break, and when i came back the [...]

marckleinebudde, 2023-05-30:
Where is the Vcc connected to?

YuBer0, 2023-05-30:
pin 17, the 3V3 pin

YuBer0, 2023-05-30:
Yup! for MCP if the CAN transceiver is  TJA1050 than it would require 5v, but i modified it to VP230. 
here is a snippet from the MCP2515 datasheet for the min value for supply voltage
[image]

marckleinebudde, 2023-05-30:
Can you connect a scope to SCK, SI and SO, and CS?

YuBer0, 2023-05-30:
Not right now, however i tried to swap a few MCP2515 (modified as well) and it doesnt seem to link up as well

marckleinebudde, 2023-05-30:
Can you give me output of `cat /sys/kernel/debug/pinctrl/fe200000.gpio-pinctrl-bcm2711/pinmux-pins ` and `cat /sys/kernel/debug/pinctrl/fe200000.gpio-pinctrl-bcm2711/pins`.

YuBer0, 2023-05-30:
cat /sys/kernel/debug/pinctrl/fe200000.gpio-pinctrl-bcm2711/pinmux-pins 
sudo cat /sys/kernel/debug/pinctrl/fe200000.gpio-pinctrl-bcm2711/pinmux-pins
> Pinmux settings per pin
> Format: pin (name): mux_owner gpio_owner hog?
> pin 0 (gpio0): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 1 (gpio1): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 2 (gpio2): fe804000.i2c (GPIO UNCLAIMED) function alt0 group gpio2
> pin 3 (gpio3): fe804000.i2c (GPIO UNCLAIMED) function alt0 group gpio3
> pin 4 (gpio4): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 5 (gpio5): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 6 (gpio6): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 7 (gpio7): fe204000.spi pinctrl-bcm2711:7 function gpio_out group gpio7
> pin 8 (gpio8): fe204000.spi pinctrl-bcm2711:8 function gpio_out group gpio8
> pin 9 (gpio9): fe204000.spi (GPIO UNCLAIMED) function alt0 group gpio9
> pin 10 (gpio10): fe204000.spi (GPIO UNCLAIMED) function alt0 group gpio10
> pin 11 (gpio11): fe204000.spi (GPIO UNCLAIMED) function alt0 group gpio11
> pin 12 (gpio12): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 13 (gpio13): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 14 (gpio14): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 15 (gpio15): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 16 (gpio16): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 17 (gpio17): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 18 (gpio18): fe204c00.spi pinctrl-bcm2711:18 function gpio_out group gpio18
> pin 19 (gpio19): fe204c00.spi (GPIO UNCLAIMED) function alt3 group gpio19
> pin 20 (gpio20): fe204c00.spi (GPIO UNCLAIMED) function alt3 group gpio20
> pin 21 (gpio21): fe204c00.spi (GPIO UNCLAIMED) function alt3 group gpio21
> pin 22 (gpio22): (MUX UNCLAIMED) pinctrl-bcm2711:22
> pin 23 (gpio23): (MUX UNCLAIMED) pinctrl-bcm2711:23
> pin 24 (gpio24): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 25 (gpio25): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 26 (gpio26): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 27 (gpio27): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 28 (gpio28): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 29 (gpio29): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 30 (gpio30): fe201000.serial (GPIO UNCLAIMED) function alt3 group gpio30
> pin 31 (gpio31): fe201000.serial (GPIO UNCLAIMED) function alt3 group gpio31
> pin 32 (gpio32): fe201000.serial (GPIO UNCLAIMED) function alt3 group gpio32
> pin 33 (gpio33): fe201000.serial (GPIO UNCLAIMED) function alt3 group gpio33
> pin 34 (gpio34): fe300000.mmcnr (GPIO UNCLAIMED) function alt3 group gpio34
> pin 35 (gpio35): fe300000.mmcnr (GPIO UNCLAIMED) function alt3 group gpio35
> pin 36 (gpio36): fe300000.mmcnr (GPIO UNCLAIMED) function alt3 group gpio36
> pin 37 (gpio37): fe300000.mmcnr (GPIO UNCLAIMED) function alt3 group gpio37
> pin 38 (gpio38): fe300000.mmcnr (GPIO UNCLAIMED) function alt3 group gpio38
> pin 39 (gpio39): fe300000.mmcnr (GPIO UNCLAIMED) function alt3 group gpio39
> pin 40 (gpio40): bcm2835_audio (GPIO UNCLAIMED) function alt0 group gpio40
> pin 41 (gpio41): bcm2835_audio (GPIO UNCLAIMED) function alt0 group gpio41
> pin 42 (gpio42): (MUX UNCLAIMED) pinctrl-bcm2711:42
> pin 43 (gpio43): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 44 (gpio44): soc:i2c0mux (GPIO UNCLAIMED) function alt1 group gpio44
> pin 45 (gpio45): soc:i2c0mux (GPIO UNCLAIMED) function alt1 group gpio45
> pin 46 (gpio46): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 47 (gpio47): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 48 (gpio48): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 49 (gpio49): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 50 (gpio50): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 51 (gpio51): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 52 (gpio52): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 53 (gpio53): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 54 (gpio54): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 55 (gpio55): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 56 (gpio56): (MUX UNCLAIMED) (GPIO UNCLAIMED)
> pin 57 (gpio57): (MUX UNCLAIMED) (GPIO UNCLAIMED)

cat /sys/kernel/debug/pinctrl/fe200000.gpio-pinctrl-bcm2711/pins

> registered pins: 58
> pin 0 (gpio0) 0:pinctrl-bcm2711 function gpio_in in hi; irq 0 (none)
> pin 1 (gpio1) 1:pinctrl-bcm2711 function gpio_in in hi; irq 0 (none)
> pin 2 (gpio2) 2:pinctrl-bcm2711 function alt0 in hi; irq 0 (none)
> pin 3 (gpio3) 3:pinctrl-bcm2711 function alt0 in hi; irq 0 (none)
[... 29 lines cut ...]
> pin 33 (gpio33) 33:pinctrl-bcm2711 function alt3 in hi; irq 0 (none)
> pin 34 (gpio34) 34:pinctrl-bcm2711 function alt3 in lo; irq 0 (none)
> pin 35 (gpio35) 35:pinctrl-bcm2711 function alt3 in hi; irq 0 (none)
> pin 36 (gpio36) 36:pinctrl-bcm2711 function alt3 in hi; irq 0 (none)
> pin 37 (gpio37) 37:pinctrl-bcm2711 function alt3 in hi; irq 0 (none)
> pin 38 (gpio38) 38:pinctrl-bcm2711 function alt3 in hi; irq 0 (none)
> pin 39 (gpio39) 39:pinctrl-bcm2711 function alt3 in hi; irq 0 (none)
> pin 40 (gpio40) 40:pinctrl-bcm2711 function alt0 in hi; irq 0 (none)
> pin 41 (gpio41) 41:pinctrl-bcm2711 function alt0 in hi; irq 0 (none)
> pin 42 (gpio42) 42:pinctrl-bcm2711 function gpio_out in hi; irq 0 (none)
> pin 43 (gpio43) 43:pinctrl-bcm2711 function gpio_in in hi; irq 0 (none)
> pin 44 (gpio44) 44:pinctrl-bcm2711 function alt1 in hi; irq 0 (none)
> pin 45 (gpio45) 45:pinctrl-bcm2711 function alt1 in hi; irq 0 (none)
> pin 46 (gpio46) 46:pinctrl-bcm2711 function gpio_in in lo; irq 0 (none)
> pin 47 (gpio47) 47:pinctrl-bcm2711 function gpio_in in lo; irq 0 (none)
> pin 48 (gpio48) 48:pinctrl-bcm2711 function gpio_in in lo; irq 0 (none)
> pin 49 (gpio49) 49:pinctrl-bcm2711 function gpio_in in lo; irq 0 (none)
[... cut ...]
> pin 51 (gpio51) 51:pinctrl-bcm2711 function gpio_in in lo; irq 0 (none)
> pin 52 (gpio52) 52:pinctrl-bcm2711 function gpio_in in lo; irq 0 (none)
> pin 53 (gpio53) 53:pinctrl-bcm2711 function gpio_in in lo; irq 0 (none)
> pin 54 (gpio54) 54:pinctrl-bcm2711 function gpio_in in lo; irq 0 (none)
> pin 55 (gpio55) 55:pinctrl-bcm2711 function gpio_in in lo; irq 0 (none)
> pin 56 (gpio56) 56:pinctrl-bcm2711 function gpio_in in lo; irq 0 (none)
> pin 57 (gpio57) 57:pinctrl-bcm2711 function gpio_in in lo; irq 0 (none)

YuBer0, 2023-05-31:
Hi! @marckleinebudde  thank you so much for your help, it seems like it is an hardware issue.
I used another RPI and am able to detect the device just fine.
Might be issues with the GPIO pins.

closed 2023-05-31 by YuBer0