linux-can/can-utils #435: SocketCan filter bug

KozAAAAA, 2023-06-08:
My library attempts to put an abstraction layer on top of the standard Socketcan functionality. The problem is that whenever I try to **### read()** in a while loop with filters set to different can_id I receive a totally random frame. It works as if filters were ignored or unable to switch as fast (??????). I tested this on two Linux kernel versions, 5.19 and 5.10, as well as on virtual can, USB-CAN, and KVASER. This bug mainly occurs while debugging. I can't confirm it yet, but It seems like it worked on the previous version of Socketcan. Here is the library: https://github.com/PUT-Motorsport/PUTM_DV_CAN_LIBRARY. 

**Here is how I connect:**
`    inline CanState CAN::connect(const char *ifname)
    {
        ifreq ifr;
        sockaddr_can addr;
        private_socket = socket(PF_CAN, SOCK_RAW, CAN_RAW);
        if (private_socket == -1)
        {
            return CanState::CAN_SOCKET_ERROR;
        }
        (void)std::strncpy(ifr.ifr_name, ifname, sizeof(ifr.ifr_name));
        if (ioctl(private_socket, SIOCGIFINDEX, &ifr) == -1)
        {
            return CanState::CAN_IOCTL_ERROR;
        }
        addr.can_family = AF_CAN;
        addr.can_ifindex = ifr.ifr_ifindex;
        if (bind(private_socket, (sockaddr *)&addr, sizeof(addr)) == -1)
        {
            return CanState::CAN_BIND_ERROR;
        }

        return CanState::CAN_OK;
    }`

**Receive messages:**
`    template <typename T>
    inline CanState CAN::receive(T &rx_frame, const time_t timeout_in_seconds)
    {
        static_assert(can_id<T> != INVALID_CAN_ID, "Can id has not been specified");

        can_frame frame;
        can_filter filter = {can_id<T>, CAN_SFF_MASK};

        timeval tv = {timeout_in_seconds, 0};
        if (setsockopt(private_socket, SOL_SOCKET, SO_RCVTIMEO, (const char *)&tv, sizeof tv) != 0)
        {
            return CanState::CAN_SET_TIMEOUT_ERROR;
        }

        if (setsockopt(private_socket, SOL_CAN_RAW, CAN_RAW_FILTER, &filter, sizeof(filter)) != 0)
        {
            return CanState::CAN_SET_MASK_ERROR;
        }

        if (read(private_socket, &frame, sizeof(can_frame)) < (ssize_t)sizeof(can_frame))
        {
            return CanState::CAN_READ_ERROR;
        }
        std::cout<<frame.can_id<<"vs"<< can_id<T> <<"\n";
        (void)std::memcpy(&rx_frame, frame.data, sizeof(T));

        return CanState::CAN_OK;
    }`

**And my main:**
`int main()
{
    /* CONNECT TO THE CAN INTERFACE */
    CAN can;
    if (can.connect("slcan0") != CanState::CAN_OK)
    {
        std::cout << "Error" << std::endl;
        return 1;
    }

    /* RECEIVE DATA */
    Apps_main apps_frame;
    BMS_LV_main lv_frame;

    while (1)
    {
        if (can.receive(apps_frame, NO_TIMEOUT) != CanState::CAN_OK)
        {
            std::cout << "Error" << std::endl;
            return 1;
        }
        if (can.receive(lv_frame, NO_TIMEOUT) != CanState::CAN_OK)
        {
            std::cout << "Error" << std::endl;
            return 1;
        }
    }
}`

hartkopp, 2023-06-08:
Hi Mateusz!

The sequence is:

- create socket()
- do all the sockopt()'s like CAN_RAW_FILTER and others
- retrieve interface index
- do bind() with this interface index (which also applies the filters)

After bind() you can read() and write() on the socket file descriptor.

When you change the filters after bind(), the internal sequence is to copy and activate the new filter set and then remove the old filter set.  So what is the "random CAN ID" you receive? A CAN ID which is neither in the old nor in the new filter?

Two other remarks:
1. ` can_filter filter = {can_id<T>, CAN_SFF_MASK};` should be `can_filter filter = {can_id<T>, CAN_SFF_MASK | CAN_EFF_FLAG | CAN_RTR_FLAG};`  as you might get RTR frames or 29 bit CAN IDs otherwise
2. Especially for automotive use-cases which handle cyclic messages (send/receive) and have to take care about timeouts or send CAN frames sequences with a fix time gap, the CAN_BCM protocol has been developed. CAN_RAW is not the best approach to handle timers and filters for cyclic CAN traffic

Btw. when there is a real problem with CAN_RAW this needs to be fixed anyway.

KozAAAAA, 2023-06-09:
Hi Oliver,

Thank you for your fast response.

If I understand correctly, I should set the setsockopt() to receive every frame without a timeout before bind(), and then modify these settings in read().

I have made the changes to the filters as you suggested, but no luck here.

Most of the time, the received IDs are either the ID of the device from the first receive or the last breakpoint. I have received frames that shouldn't have been received as well.

If it turns out that this issue is not easily fixable, I will give CAN_BCM a try.

hartkopp, 2023-06-09:
The idea is to provide a complete filter set (of up to 512 filter) before bind().

See here: https://github.com/linux-can/can-tests/blob/master/raw/tst-raw.c#L117

E.g.

```
	rfilter[0].can_id   = 0x123;
	rfilter[0].can_mask = CAN_SFF_MASK | CAN_EFF_FLAG | CAN_RTR_FLAG;
	rfilter[1].can_id   = 0x200;
	rfilter[1].can_mask = CAN_SFF_MASK | CAN_EFF_FLAG | CAN_RTR_FLAG;
	rfilter[2].can_id   = 0x400;
	rfilter[2].can_mask = CAN_SFF_MASK | CAN_EFF_FLAG | CAN_RTR_FLAG;
	rfilter[3].can_id   = 0x651;
	rfilter[3].can_mask = CAN_SFF_MASK | CAN_EFF_FLAG | CAN_RTR_FLAG;

	setsockopt(s, SOL_CAN_RAW, CAN_RAW_FILTER, &rfilter, sizeof(rfilter));
```

And after reading from the socket you can use  a `switch(frame.can_id){}` statement with `frame.can_id` to handle that specific frame.

It makes no sense to apply such filter directly before the `read()` statement.

KozAAAAA, 2023-06-09:
So setting the filter right before read() is undefined?

KozAAAAA, 2023-06-09:
I think that in our case this behavior is expected as read waits until ID occurs on the can bus. We don't use read for safety-critical applications, mainly logging data. There are multiple ros nodes in our system and It's easier to set this filter on each can.receive() for the user of the library. To wrap It up, do I do something wrong or is there a bug in CAN_RAW?

hartkopp, 2023-06-09:
You are doing it wrong.

You open ONE socket and have different readers where each of these readers manipulates the filter for this single socket for his own needs. This simply does not work.

You can either open as much sockets as you have CAN IDs to monitor and apply the filter for that CAN ID on those sockets.

Or you do it as suggested above by me with the filter set of multiple CAN IDs and the switch() statement.

Or you take a look at CAN_BCM sockets which is very likely what you need for cyclic messages with timeout management and content filtering.

KozAAAAA, 2023-06-10:
Thanks a lot! Everything is crystal clear now.

closed 2023-06-10 by hartkopp