linux-can/can-utils #437: CAN remote frame with DLC greater than 8

a-matos00, 2023-06-09:
Is it possible to send a remote request frame with a DLC greater than 8 by using cansend ? The CAN FD specification says that CAN FD devices use CAN 2.0 format for remote request frames. I noticed that this is currently not implemented.

hartkopp, 2023-06-09:
Oh!
There really seems to be an issue with `cansend` ... I'll take a look at it.

But regarding `candump` -> please try the `-8` option, e.g. `candump -8 can0` to see those DLCs with `candump`.
`candump -L can0` prints the logfile format whoch also includes DLC > 8 stuff.

hartkopp, 2023-06-13:
@a-matos00 can this issue considered to be solved for you?

a-matos00, 2023-06-13:
Thank you so much for this information. This solves the issue then.

closed 2023-06-13 by a-matos00