linux-can/can-utils #440: Error configuring CAN interface: ioctl SIOCSIFNAME rename: File exists

jensenr30, 2023-07-06:
When I try to configure a serial device as a CAN interface, I sometimes get the following error:
```
ioctl SIOCSIFNAME rename: File exists
```
The command I am running is:
```
sudo slcand -o -c -s8 /dev/serial/by-id/my_serial_to_can_adapter_id can0
```

This issue only seems to happen after my computer is suspended using `systemctl suspend`.  The error is resolved by rebooting my computer.  So in the meantime, I will avoid suspending my computer.  But why does suspending my computer cause an error when re-configuring a can interface?

Machine information:
Arch Linux x86_64 6.1.37-1-lts #1 SMP PREEMPT_DYNAMIC Sat, 01 Jul 2023 17:31:09 +0000 x86_64 GNU/Linux
13th Gen Intel(R) Core(TM) i5-13600K

hartkopp, 2023-07-07:
> When I try to configure a serial device as a CAN interface, I sometimes get the following error:
> 
> ```
> ioctl SIOCSIFNAME rename: File exists
> ```
> 
> The command I am running is:
> 
> ```
> sudo slcand -o -c -s8 /dev/serial/by-id/my_serial_to_can_adapter_id can0
> ```
> 

In this example you force `slcand` to create `can0` as CAN device name.

The CAN interface name (here: `can0`) is an optional parameter.

So if you want to assign `can0` here you have to make sure, that there is no `can0` in your system at this point.
Is it possible that your `slcand` from *before* the suspend is still running?

jensenr30, 2023-07-07:
@hartkopp 
Thanks for the reply.  That is possible.  I've tried to delete the `can0` interface but failed.  I've tried disabling it with commands such as `ifconfig can0 down` and the interface is down, but it still exists on my system (virtual interfaces are still present and `ifconfig -a` still shows `can0`).

But as far as I can tell, `slcand` doesn't care if the interface is already running; I can run my script as many times as I want and it always produces a functioning CAN interface.  It is only when my computer goes through a suspend cycle that I encounter the error message.

my script is:
```
sudo ifconfig can0 down
sudo slcand -o -c -s8 /dev/serial/by-id/my_device can0
sudo ifconfig can0 up
sudo ifconfig can0 txqueuelen 1000
```

hartkopp, 2023-07-07:
Bringing a CAN interface "down" just means it is not in operation. This (intentionally) does not remove the interface.

I assume you only have a slcan device, right?

When there is a `can0` interface, there has to be a `slcand`instance too.
Can you check for a running slcand when waking up from suspend?

Maybe things are still working and you do not need to start a new slcand instance?!?

If it is not working then stop the slcand and start a new instance ...

jensenr30, 2023-07-08:
@hartkopp Thank you for pointing that out!  I tested it again, and after suspending my computer and waking up, there was an `slcand` process still running.  The can0 interface was down, and I couldn't bring it back up.  So I killed the `slcand` process, and re-ran my setup script.  That did the trick!  Thanks again for your help.  I didn't realize that slcand ran in the background all the time - I wrongly assumed it was only responsible for initial configuration of the interface.

closed 2023-07-08 by jensenr30