linux-can/can-utils #442: j1939 error: j1939_tp_rxtimer rx timeout, send abort

in-text, 2023-07-13:
Hello!

I catch errors in dmesg, when send large frames (> 8 bytes): `while true; do sleep 1; testj1939 -B -s128 can0:0x2a can0:,0xff01; done` (with frames <= 8 all okay).

Errors:
```shell
[23112.067826] flexcan 2090000.flexcan can0: j1939_tp_rxtimer: 0xf2cf7c70: rx timeout, send abort
[23114.297614] flexcan 2090000.flexcan can0: j1939_tp_rxtimer: 0xcee12276: rx timeout, send abort
[23116.528282] flexcan 2090000.flexcan can0: j1939_tp_rxtimer: 0xcee12276: rx timeout, send abort
[23118.761639] flexcan 2090000.flexcan can0: j1939_tp_rxtimer: 0xcee12276: rx timeout, send abort
[23120.989034] flexcan 2090000.flexcan can0: j1939_tp_rxtimer: 0xcee12276: rx timeout, send abort
[23123.218722] flexcan 2090000.flexcan can0: j1939_tp_rxtimer: 0x56694c17: rx timeout, send abort
```

Maybe I do something wrong?

I use [this](https://github.com/SolidRun/linux-stable/tree/imx_5.4.47_2.2.0-sr) kernel for i.MX6DL processor (v5.4)

in-text, 2023-07-13:
Same error on vcan iface:
```shell
[23539.842874] can1: j1939_tp_rxtimer: 0x0f062086: rx timeout, send abort
[23542.070520] can1: j1939_tp_rxtimer: 0x7d4f212b: rx timeout, send abort
```

```shell
8: can1: <NOARP,UP,LOWER_UP> mtu 72 qdisc noqueue state UNKNOWN group default qlen 1000
    link/can
```

olerem, 2023-07-13:
Hm... I just took your example command, replaced can0 to vcan0 and tested It on stock ubuntu, uname:
 6.2.0-24-generic #24-Ubuntu SMP PREEMPT_DYNAMIC Fri Jun 16 12:03:50 UTC 2023 x86_64 x86_64 x86_64
 
 It works as expected.

olerem, 2023-07-13:
From the error message, I would say, the J1939 stack didn't not got an echo packet from the CAN stack. If i remember it correctly, we mainlined some fix for this.

If you wont to use old kernels, i would recommend to backport most of this fixes https://github.com/SolidRun/linux-stable/commits/linux-6.3.y-cex7?before=4030ddfd0e381814b52f00a07b98239c2f58c754+35&branch=linux-6.3.y-cex7&path%5B%5D=net&path%5B%5D=can&path%5B%5D=j1939&qualified_name=refs%2Fheads%2Flinux-6.3.y-cex7

in-text, 2023-07-14:
> From the error message, I would say, the J1939 stack didn't not got an echo packet from the CAN stack. If i remember it correctly, we mainlined some fix for this.
>
> If you wont to use old kernels, i [...]

I found this [patch](https://patchwork.kernel.org/project/netdevbpf/patch/20210906094219.95924-1-william.xuanziyang@huawei.com/#24439087), and apply it. No more errors come in dmesg.

I be able to apply this [patches](https://github.com/SolidRun/linux-stable/commits/linux-6.3.y-cex7?before=4030ddfd0e381814b52f00a07b98239c2f58c754+35&branch=linux-6.3.y-cex7&path%5B%5D=net&path%5B%5D=can&path%5B%5D=j1939&qualified_name=refs%2Fheads%2Flinux-6.3.y-cex7) and backport to 5.4, but I'm not sure that it will not be necessary to update the other modules of the kernel

marckleinebudde, 2023-07-14:
Use at least v5.4.156, it contains the patch you've mentioned.

closed 2023-07-14 by in-text