linux-can/can-utils #450: Linux terminal CAN issue

soorajs1997, 2023-08-18:
Hello,

My name is Sooraj. I am new to this forum and to CAN communication. I have established a CAN communication between a curtis motor controller and a Jetson AGX Orin. when I type ip -details -statistic link show can0, initially, it shows a proper connection without any issues. However, after sending the first CAN message, it shows 24 bytes and 3 packets received 0 packets transmitted and No-carrier and the bus turns off. the network however is still up. I am using a SN65 as a transceiver in between the motor and the Jetson to convert the TX and RX of the Jetson to CAN Hi and CAN Lo. Any help would be appreciated

hartkopp, 2023-08-20:
1. Can you please post the output of `ip -det link show can0` before and after sending the first message?
2. Do you have a termination resistor (60 Ohms) between CAN_L and CAN_H lines?

rusoku, 2023-08-21:
120 Ohms ?

>  1. Do you have a termination resistor (60 Ohms) between CAN_L and
>     CAN_H lines?
>
> Message ID: ***@***.***>
>

hartkopp, 2023-08-21:
Yes, 2 times a 120 Ohm resistor (on each end of the CAN segment!) sums up to 60 Ohms which can be measured e.g. with a MultiMeter.
https://www.allaboutcircuits.com/tools/parallel-resistance-calculator/

soorajs1997, 2023-08-22:
Hello,

@hartkopp thank you for the immediate response. Yes, I had put 2 120 Ohm resistors summing up to 60 Ohms. Though, I solved the issue. 

For any one in the future who might face a similar situation - the solution was to check the pinmux registers. the values are supposed to be 454 and 400 for the can-in and can-out. I set them as 458 and 400 and then as the default values 454 and 404.

I am going to need follow up help regarding the python code, if you could. I am trying to write a python code to send and receive can message using python-can. Are there any example codes you know regarding these?

hartkopp, 2023-08-22:
> @hartkopp thank you for the immediate response. Yes, I had put 2 120 Ohm resistors summing up to 60 Ohms. Though, I solved the issue.

Great!

> For any one in the future who might face a similar situation - the solution was to check the pinmux registers. the values are supposed to be 454 and 400 for the can-in and can-out. I set them as 458 and 400 and then as the default values 454 and 404.

Ok, this is really  Jetson AGX Orin specific then ...

> I am going to need follow up help regarding the python code, if you could. I am trying to write a python code to send and receive can message using python-can. Are there any example codes you know regarding these?

You should better try Google for this - and swap on GitHub to this project: https://github.com/hardbyte/python-can
They have tons of documentation and examples there.

closed 2023-08-22 by soorajs1997