linux-can/can-utils #451: Detect Error Frames and Data frames both in SocketCAN

abhiroopnray, 2023-08-24:
Hi,

I am using socket can to communicate CAN-FD from application to a separate board.
I can detect data frames.
I would also like to receive error frames like TX-PASSIVE, TX_WARNING etc.

I have gone through this:
https://elixir.bootlin.com/linux/v4.9/source/Documentation/networking/can.txt#L521

My code is like below:

```
can_err_mask_t err_mask = (CAN_ERR_CRTL_TX_OVERFLOW | CAN_ERR_CRTL_TX_PASSIVE |
                               CAN_ERR_CRTL_TX_WARNING | CAN_ERR_TX_TIMEOUT |
                               CAN_ERR_CRTL_RX_OVERFLOW | CAN_ERR_ACK |
                               CAN_ERR_CRTL_RX_PASSIVE | CAN_ERR_MASK | CAN_ERR_FLAG |
                               CAN_ERR_CRTL_RX_WARNING | CAN_ERR_BUSOFF | CAN_ERR_CRTL_ACTIVE);

    if (setsockopt(sock, SOL_CAN_RAW, CAN_RAW_ERR_FILTER, &err_mask, sizeof(err_mask))) {
        LOGE("error while enabling network problem notification");
    }

filter[0].can_id = 0x60;
#if ENABLE_EXTENDED_FRAMES
    filter[0].can_mask = CAN_EFF_MASK;
#else
    filter[0].can_mask = CAN_SFF_MASK;
#endif

    filter[1].can_id = 0x70;
#if ENABLE_EXTENDED_FRAMES
    filter[1].can_mask = CAN_EFF_MASK;
#else
    filter[1].can_mask = CAN_SFF_MASK;
#endif

    if(setsockopt(sock, SOL_CAN_RAW, CAN_RAW_FILTER, &filter, sizeof(filter))) {
        LOGE("Error while applying the filters");
    }
if (bind(sock, (struct sockaddr *) &addr, sizeof(addr)) < 0) {
        LOGE("Error on Binding to the socket addr");
        return;
    }
    LOGD("Binding done");

    while (running) {
        number_of_bytes = read(sock, &frame, sizeof(struct canfd_frame));
        /* Read the frames */
        LOGD("Reading a frame");

        jbyte readDataFrame[frame.len];
        if (number_of_bytes < 0) {
            LOGE("No bytes written");
            const char *error_message = "No bytes written";
            jstring error_message_string = env->NewStringUTF(error_message);
            env->CallVoidMethod(callbackObject, onErrorCallbackMethod, error_message_string);
            continue;
        }

        LOGD("Debug flags on read Frame id = 0x%X, Frame length = %d, Frame flags = 0x%X", frame.can_id, frame.len,
             frame.flags);

        for (i = 0; i < frame.len; i++) {
            readDataFrame[i] = (jbyte) (frame.data[i]);
            //LOGD("data[%d] = %02X ", i, frame.data[i]);
            if (frame.can_id & CAN_ERR_FLAG) {
                LOGD("CAN received an error frame");
            }
            LOGD("data[%d] = %02X ", i, frame.data[i]);
            LOGD("read data[%d] = %02X flag = %X", i, frame.data[i], frame.flags);
        }
}
```
But I am still not getting the error frames. I am trying to inject error frames via PCAN analyzer.
Can you tell me how can I receive both data frames and error frames with SocketCAN.

Thanks,
Abhiroop

hartkopp, 2023-08-24:
Hi,

first of all your filter setup here is not correct:

> filter[0].can_id = 0x60;
> #if ENABLE_EXTENDED_FRAMES
>     filter[0].can_mask = CAN_EFF_MASK;
> #else
>     filter[0].can_mask = CAN_SFF_MASK;
> #endif

If you want to get EFF frames only you have to set CAN_EFF_FLAG too in can_id and can_mask:

e.g. 
 
 #if ENABLE_EXTENDED_FRAMES
     filter[1].can_id = 0x70 | CAN_EFF_FLAG;
     filter[1].can_mask = CAN_EFF_MASK | CAN_EFF_FLAG;
 #else
     filter[1].can_id = 0x70;
     filter[1].can_mask = CAN_EFF_MASK | CAN_EFF_FLAG;
 #endif
 

> But I am still not getting the error frames. I am trying to inject error frames via PCAN analyzer. Can you tell me how can I receive both data frames and error frames with SocketCAN.

```
can_err_mask_t err_mask = (CAN_ERR_CRTL_TX_OVERFLOW | CAN_ERR_CRTL_TX_PASSIVE |
                               CAN_ERR_CRTL_TX_WARNING | CAN_ERR_TX_TIMEOUT |
                               CAN_ERR_CRTL_RX_OVERFLOW | CAN_ERR_ACK |
                               CAN_ERR_CRTL_RX_PASSIVE | CAN_ERR_MASK | CAN_ERR_FLAG |
                               CAN_ERR_CRTL_RX_WARNING | CAN_ERR_BUSOFF | CAN_ERR_CRTL_ACTIVE);
```

These are the wrong values!

The error mask uses only these flags:
```
/* error class (mask) in can_id */
#define CAN_ERR_TX_TIMEOUT   0x00000001U /* TX timeout (by netdevice driver) */
#define CAN_ERR_LOSTARB      0x00000002U /* lost arbitration    / data[0]    */
#define CAN_ERR_CRTL         0x00000004U /* controller problems / data[1]    */
#define CAN_ERR_PROT         0x00000008U /* protocol violations / data[2..3] */
#define CAN_ERR_TRX          0x00000010U /* transceiver status  / data[4]    */
#define CAN_ERR_ACK          0x00000020U /* received no ACK on transmission */
#define CAN_ERR_BUSOFF       0x00000040U /* bus off */
#define CAN_ERR_BUSERROR     0x00000080U /* bus error (may flood!) */
#define CAN_ERR_RESTARTED    0x00000100U /* controller restarted */
#define CAN_ERR_CNT          0x00000200U /* TX error counter / data[6] */
					 /* RX error counter / data[7] */
```

And then the other values like CAN_ERR_CRTL_RX_OVERFLOW can be found in the data[] of the received error message frame.

You can also use `candump any,#FFFFFFFF` to experiment with error messages (see `candump -h`).

abhiroopnray, 2023-08-25:
Hi,

Thanks for the response. 
ENABLE_EXTENDED_FRAMES is disabled now, so I have made like this:

```
filter[0].can_id = 0x60;
#if ENABLE_EXTENDED_FRAMES
filter[0].can_mask = CAN_EFF_MASK;
#else
filter[0].can_mask = CAN_SFF_MASK;
#endif
```
Below one:

```
can_err_mask_t err_mask = (CAN_ERR_CRTL_TX_OVERFLOW | CAN_ERR_CRTL_TX_PASSIVE |
                               CAN_ERR_CRTL_TX_WARNING | CAN_ERR_TX_TIMEOUT |
                               CAN_ERR_CRTL_RX_OVERFLOW | CAN_ERR_ACK |
                               CAN_ERR_CRTL_RX_PASSIVE | CAN_ERR_MASK | CAN_ERR_FLAG |
                               CAN_ERR_CRTL_RX_WARNING | CAN_ERR_BUSOFF | CAN_ERR_CRTL_ACTIVE);
```

I have changed accordingly to this:

```
can_err_mask_t err_mask = (CAN_ERR_TX_TIMEOUT | CAN_ERR_LOSTARB | CAN_ERR_CRTL | CAN_ERR_PROT |
                               CAN_ERR_TRX | CAN_ERR_ACK | CAN_ERR_BUSOFF | CAN_ERR_BUSERROR |
                               CAN_ERR_RESTARTED | CAN_ERR_CNT);
```

After this is this enough to check the error frames if I inject error frames via PCAN analyzer:

```
if (frame.can_id & CAN_ERR_FLAG) {
    LOGD("CAN received an error frame");
}
```

Also do we need to create a new socket for the CAN_RAW_ERR_FILTER or it can be added as socketopt along with CAN_RAW_FD_FRAMES? 

I have created the socket like below:

```
if ((sock = socket(PF_CAN, SOCK_RAW, CAN_RAW)) < 0) {
        LOGE("Socket creation failed in read");
        return;
    }
```

Let me know about the above queries.

Thanks,
Abhiroop

hartkopp, 2023-08-25:
> Hi,
> 
> Thanks for the response. ENABLE_EXTENDED_FRAMES is disabled now, so I have made like this:
> 
> ```
> filter[0].can_id = 0x60;
> #if ENABLE_EXTENDED_FRAMES
> filter[0].can_mask = CAN_EFF_MASK;
> #else
> filter[0].can_mask = CAN_SFF_MASK;
> #endif
> ```
> 

The mask is again wrong! Please read the Kernel documentation why you need to set the CAN_EFF_FLAG in the can_mask!
https://www.kernel.org/doc/html/latest/networking/can.html#raw-socket-option-can-raw-filter

> Below one:
>
> ```
> can_err_mask_t err_mask = (CAN_ERR_CRTL_TX_OVERFLOW | CAN_ERR_CRTL_TX_PASSIVE |
>                                CAN_ERR_CRTL_TX_WARNING | CAN_ERR_TX_TIMEOUT [...]

Good!

> After this is this enough to check the error frames if I inject error frames via PCAN analyzer:
> 
> ```
> if (frame.can_id & CAN_ERR_FLAG) {
>     LOGD("CAN received an error frame");
> }
> ```
> 

Instead of changing your code you better check the behaviour with `candump` first.
And when you see the messages as expected you  should continue with programming.

> Also do we need to create a new socket for the CAN_RAW_ERR_FILTER or it can be added as socketopt along with CAN_RAW_FD_FRAMES?
> 
> I have created the socket like below:
> 
> ```
> if ((sock = socket(PF_CAN, SOCK_RAW, CAN_RAW)) < 0) {
>         LOGE("Socket creation failed in read");
>         return;
>     }
> ```

After creating the socket you need to execute TWO setsockopt() calls.

One for CAN_RAW_ERR_FILTER and one for CAN_RAW_FD_FRAMES.

See here:
https://github.com/linux-can/can-utils/blob/master/candump.c#L575
https://github.com/linux-can/can-utils/blob/master/candump.c#L593

abhiroopnray, 2023-08-28:
Hi

Thanks for clarification.

I am trying to send data and then inject error both via PCAN.

As you can see the below candump:

```
127|gui_p1:/ # candump can0
[  743.952434][  T377] healthd: battery l=85 v=3 t=35.0 h=2 st=2 c=400000 fc=4000000 cc=32 chg=a
  can0  060   [8]  00 00 00 00 00 00 00 01
  can0  060   [8]  00 00 00 00 00 00 00 01
  can0  060   [8]  00 00 00 00 00 00 00 01
  can0  060   [8]  00 00 00 00 00 00 00 01
  can0  060   [8]  00 00 00 00 00 00 00 01
  can0  060   [8]  00 00 00 00 00 00 00 01
  can0  060   [8]  00 00 00 00 00 00 00 01
  can0  060   [8]  00 00 00 00 00 00 00 01
  can0  060   [8]  00 00 00 00 00 00 00 01
[  803.952439][  T377] healthd: battery l=85 v=3 t=35.0 h=2 st=2 c=400000 fc=4000000 cc=32 chg=a

```
The data is going fine few times I sent. Then when I inject the error the data simply don't pass any more. But neither it is showing any error flag in candump.

In ADB logs I can get this:

`08-30 22:48:10.632     0     0 D         : [    C0] flexcan 308c0000.can can0: Controller changed from Error Active State (0) into Error Warning State (1).
`
 I need the error frames in the can socket. Can you tell me how can I proceed and what else I need to cehck?

Thanks and regards,
Abhiroop Nandi Ray

hartkopp, 2023-08-28:
You still only write:
`127|gui_p1:/ # candump can0`

Enable the error messages (and the human readable output) with:
`127|gui_p1:/ # candump -e can0,#FFFFFFFF`

abhiroopnray, 2023-08-28:
Hi,

I have used candump -e can0 

I got the below output:

```
gui_p1:/ # candump -e can0,#FFFFFFFF
  can0  060   [8]  00 00 00 00 00 00 00 00
  can0  20000004   [8]  00 04 00 00 00 00 00 00   ERRORFRAME
        controller-problem{rx-error-warning}
  can0  123  [08]  11 22 FF 11 00 00 00 00
  can0  123  [08]  11 22 FF 11 00 00 00 00
  can0  060   [8]  00 00 00 00 00 00 00 00
  can0  20000004   [8]  00 10 00 00 00 00 00 00   ERRORFRAME
        controller-problem{rx-error-passive}
[  955.841089][    T1] init: Untracked pid 6996 exited with status 0
[  955.893979][    T1] init: Untracked pid 6998 exited with status 0
[  980.001994][  T369] healthd: battery l=85 v=3 t=35.0 h=2 st=2 c=400000 fc=4000000 cc=32 chg=a
  can0  123  [08]  11 22 FF 11 00 00 00 00
  can0  20000004   [8]  00 04 00 00 00 00 00 00   ERRORFRAME
        controller-problem{rx-error-warning}
  can0  123  [08]  11 22 FF 11 00 00 00 00
  can0  060   [8]  00 00 00 00 00 00 00 00
  can0  20000004   [8]  00 10 00 00 00 00 00 00   ERRORFRAME
        controller-problem{rx-error-passive}

```
These two:
  ```
can0  123  [08]  11 22 FF 11 00 00 00 00
  can0  060   [8]  00 00 00 00 00 00 00 00
```

are data frames I am sending from pcan analyzer.
I am also getting error frames.after generating error multiple times.

Can you tell me is this the correct approach and how to get this error in socket?

Thanks and regards,
Abhiroop Nandi Ray

hartkopp, 2023-08-28:
Excellent!

As you can see in the PCAN-View screenshot, the error frame is something that is monitored and detected at the **_sending_** node, which also increases the tx error counter "**TxErr**".

It depends on the CAN controller - but if you want to see something similar on Linux, you need so "**send**" frames from the Linux system and destroy them with the Windows tool - as you made it with the PCAN-View tool.

abhiroopnray, 2023-08-29:
Hi

Thanks for the support.
Now I am also getting error detection in CAN-Socket.

```
08-31 22:21:17.839  2374  2398 D SocketCAN:    CAN received an error frame
08-31 22:21:17.839  2374  2398 D SocketCAN:    data[0] = 00
08-31 22:21:17.839  2374  2398 D SocketCAN:    read data[0] = 0x0 flag = 0x0
08-31 22:21:17.839  2374  2398 D SocketCAN:    data[1] = 04
08-31 22:21:17.839  2374  2398 D SocketCAN:    read data[1] = 0x4 flag = 0x0
08-31 22:21:17.839  2374  2398 D SocketCAN:    data[2] = 00
08-31 22:21:17.839  2374  2398 D SocketCAN:    read data[2] = 0x0 flag = 0x0
08-31 22:21:17.839  2374  2398 D SocketCAN:    data[3] = 00
08-31 22:21:17.839  2374  2398 D SocketCAN:    read data[3] = 0x0 flag = 0x0
08-31 22:21:17.839  2374  2398 D SocketCAN:    data[4] = 00
08-31 22:21:17.839  2374  2398 D SocketCAN:    read data[4] = 0x0 flag = 0x0
08-31 22:21:17.839  2374  2398 D SocketCAN:    data[5] = 00
08-31 22:21:17.839  2374  2398 D SocketCAN:    read data[5] = 0x0 flag = 0x0
08-31 22:21:17.839  2374  2398 D SocketCAN:    data[6] = 00
08-31 22:21:17.839  2374  2398 D SocketCAN:    read data[6] = 0x0 flag = 0x0
08-31 22:21:17.839  2374  2398 D SocketCAN:    data[7] = 00
08-31 22:21:17.839  2374  2398 D SocketCAN:    read data[7] = 0x0 flag = 0x0
[  505.464828][    T1] init: Untracked pid 2430 exited with status 0
[  505.517932][    T1] init: Untracked pid 2432 exited with status 0
08-31 22:21:19.531  2440  2464 E SocketCAN: No error while enabling network problem notification    0

08-31 22:21:54.338  2440  2464 D SocketCAN:    data[0] = 11
08-31 22:21:54.338  2440  2464 D SocketCAN:    read data[0] = 0x11 flag = 0x0
08-31 22:21:54.338  2440  2464 D SocketCAN:    data[1] = 22
08-31 22:21:54.338  2440  2464 D SocketCAN:    read data[1] = 0x22 flag = 0x0
08-31 22:21:54.338  2440  2464 D SocketCAN:    data[2] = FF
08-31 22:21:54.338  2440  2464 D SocketCAN:    read data[2] = 0xFF flag = 0x0
08-31 22:21:54.338  2440  2464 D SocketCAN:    data[3] = 11
08-31 22:21:54.338  2440  2464 D SocketCAN:    read data[3] = 0x11 flag = 0x0
08-31 22:21:54.338  2440  2464 D SocketCAN:    data[4] = 00
08-31 22:21:54.338  2440  2464 D SocketCAN:    read data[4] = 0x0 flag = 0x0
08-31 22:21:54.338  2440  2464 D SocketCAN:    data[5] = 00
08-31 22:21:54.338  2440  2464 D SocketCAN:    read data[5] = 0x0 flag = 0x0
08-31 22:21:54.338  2440  2464 D SocketCAN:    data[6] = 00
08-31 22:21:54.338  2440  2464 D SocketCAN:    read data[6] = 0x0 flag = 0x0
08-31 22:21:54.338  2440  2464 D SocketCAN:    data[7] = 00
08-31 22:21:54.338  2440  2464 D SocketCAN:    read data[7] = 0x0 flag = 0x0

08-31 22:21:54.339  2440  2464 D SocketCAN:    CAN received an error frame
08-31 22:21:54.339  2440  2464 D SocketCAN:    data[0] = 00
08-31 22:21:54.339  2440  2464 D SocketCAN:    read data[0] = 0x0 flag = 0x0
08-31 22:21:54.339  2440  2464 D SocketCAN:    data[1] = 04
08-31 22:21:54.339  2440  2464 D SocketCAN:    read data[1] = 0x4 flag = 0x0
08-31 22:21:54.339  2440  2464 D SocketCAN:    data[2] = 00
08-31 22:21:54.339  2440  2464 D SocketCAN:    read data[2] = 0x0 flag = 0x0
08-31 22:21:54.339  2440  2464 D SocketCAN:    data[3] = 00
08-31 22:21:54.339  2440  2464 D SocketCAN:    read data[3] = 0x0 flag = 0x0
08-31 22:21:54.339  2440  2464 D SocketCAN:    data[4] = 00
08-31 22:21:54.339  2440  2464 D SocketCAN:    read data[4] = 0x0 flag = 0x0
08-31 22:21:54.339  2440  2464 D SocketCAN:    data[5] = 00
08-31 22:21:54.339  2440  2464 D SocketCAN:    read data[5] = 0x0 flag = 0x0
08-31 22:21:54.339  2440  2464 D SocketCAN:    data[6] = 00
08-31 22:21:54.339  2440  2464 D SocketCAN:    read data[6] = 0x0 flag = 0x0
08-31 22:21:54.339  2440  2464 D SocketCAN:    data[7] = 00
08-31 22:21:54.339  2440  2464 D SocketCAN:    read data[7] = 0x0 flag = 0x0

```
But now I need to filter the errors.
I am trying this way:

```
if (frame.can_id & CAN_ERR_FLAG) {
            jstring error_message_string = env->NewStringUTF(
                    reinterpret_cast<const char *>(CAN_ERR_FLAG));
            LOGD("CAN received an error frame ");
            LOGD("error frame data %X", frame.data);
            LOGD("error frame %X", CAN_ERR_FLAG);
            if (frame.can_id & CAN_ERR_CRTL_RX_PASSIVE) {
                LOGD("CAN received an error frame : CAN_ERR_CRTL_RX_PASSIVE: %X",
                     CAN_ERR_CRTL_RX_PASSIVE);
            } else if (frame.can_id & CAN_ERR_CRTL_RX_WARNING) {
                     LOGD("CAN received an error frame : CAN_ERR_CRTL_RX_WARNING: %X",
                     CAN_ERR_CRTL_RX_WARNING);
            }
        }

```

But Is this the correct way? I am not getting the exact error. 
Can you tell me how to filter particular error.

Thanks and regards,
Abhiroop Nandi Ray

marckleinebudde, 2023-08-29:
Have a look at `linux/can/error.h`:

```
/* error status of CAN-controller / data[1] */                                                                                                                                                                                                                  
#define CAN_ERR_CRTL_UNSPEC      0x00 /* unspecified */                                                                                                                                                                                                         
#define CAN_ERR_CRTL_RX_OVERFLOW 0x01 /* RX buffer overflow */                                                                                                                                                                                                  
#define CAN_ERR_CRTL_TX_OVERFLOW 0x02 /* TX buffer overflow */                                                                                                                                                                                                  
#define CAN_ERR_CRTL_RX_WARNING  0x04 /* reached warning level for RX errors */                                                                                                                                                                                 
#define CAN_ERR_CRTL_TX_WARNING  0x08 /* reached warning level for TX errors */                                                                                                                                                                                 
#define CAN_ERR_CRTL_RX_PASSIVE  0x10 /* reached error passive status RX */                                                                                                                                                                                     
#define CAN_ERR_CRTL_TX_PASSIVE  0x20 /* reached error passive status TX */                                                                                                                                                                                     
                                      /* (at least one error counter exceeds */                                                                                                                                                                                 
                                      /* the protocol-defined level of 127)  */                                                                                                                                                                                 
#define CAN_ERR_CRTL_ACTIVE      0x40 /* recovered to error active state */       
```

It says you should look at `data[1]`.

hartkopp, 2023-08-29:
Or just take a look into `lib.c` how `candump` decodes the error message frames.

https://github.com/linux-can/can-utils/blob/master/lib.c#L636

abhiroopnray, 2023-08-31:
Thank you I am able to get the different types of error while checking the data field [1].

Can you tell me any way to detect the ERROR_ACTIVE state,
if the bus goes back from PASSIVE/WARNING to ACTIVE state?

marckleinebudde, 2023-08-31:
```
#define CAN_ERR_CRTL_ACTIVE      0x40 /* recovered to error active state */
```