linux-can/can-utils #459: In can-calc-bit-timing -? the --tq parameter is not described

EnricoMontecaggi, 2023-09-19:
Hi all,
if I run [can-calc-bit-timing -?](https://github.com/linux-can/can-utils/blob/master/calc-bit-timing/can-calc-bit-timing.c#L95-L119) the **--tq** parameter is not described
```
./can-calc-bit-timing -?
can-calc-bit-timing - calculate CAN bit timing parameters.
Usage: can-calc-bit-timing [options] [<CAN-contoller-name>]
Options:
        -q             don't print header line
        -l             list all support CAN controller names
        -b <bitrate>   arbitration bit-rate in bits/sec
        -d <bitrate>   data bit-rate in bits/sec
        -s <samp_pt>   sample-point in one-tenth of a percent
                       or 0 for CIA recommended sample points
        -c <clock>     real CAN system clock in Hz
        --alg <alg>    choose specified algorithm for bit-timing calculation

Or supply low level bit timing parameters to decode them:

        --prop-seg     Propagation segment in TQs
        --phase-seg1   Phase buffer segment 1 in TQs
        --phase-seg2   Phase buffer segment 2 in TQs
        --sjw          Synchronisation jump width in TQs
        --brp          Bit-rate prescaler
        --tseg1        Time segment 1 = prop-seg + phase-seg1
        --tseg2        Time segment 2 = phase_seg2
```

marckleinebudde, 2023-09-19:
What do you think about adding this to the help output:

```
        --tq           Time quantum in ns
```

EnricoMontecaggi, 2023-09-19:
Sounds good!

closed 2023-09-19 by marckleinebudde