linux-can/can-utils #463: cangw persistent rules

pompushko, 2023-10-01:
Hello

How can I make cangw persistent rules?

Thank you!

hartkopp, 2023-10-01:
When you start your system you need to configure the CAN interfaces.
I would suggest to configure the cangw rules at the same time, so that they are enabled, when the CAN interfaces get up. 

If you want to preserve your current cangw config just type `cangw -L` which outputs the current routing jobs in a text representation that you can use as shell script as-is.

pompushko, 2023-10-01:
> When you start your system you need to configure the CAN interfaces. I would suggest to configure the cangw rules at the same time, so that they are enabled, when the CAN interfaces get up.
>
> If you [...]

Thank you for your response!

In short - better to add this rules in to post-up?

pompushko, 2023-10-03:
Well. I facing issue, when if I add rules to **post-up**, they not applied ;(
Only **cangw -F** and adding them again - solve issue....

hartkopp, 2023-10-03:
IIRC it is possible to define can-gw rules when the two interfaces are existing. They should not necessarily need to be in 'UP' state for that.

pompushko, 2023-10-03:
> IIRC it is possible to define can-gw rules when the two interfaces are existing. They should not necessarily need to be in 'UP' state for that.

 Could you, please,  help me with that?

hartkopp, 2023-10-04:
I just wanted to point out that there should be no difference whether the CAN interface is UP or DOWN when you set up the cangw rules.

pompushko, 2023-10-04:
> I just wanted to point out that there should be no difference whether the CAN interface is UP or DOWN when you set up the cangw rules.

Well.. Than I facing strange issue...  I see packets on physical interface, but they are not forwarded to virtual interface... weird...

hartkopp, 2023-10-06:
How do you "see" the packets on the physical interface? With `candump can0` on your Linux box?
If so, your defined cangw rules are very likely wrong.

pompushko, 2023-10-07:
> How do you "see" the packets on the physical interface? With `candump can0` on your Linux box? If so, your defined cangw rules are very likely wrong.

Hello. Sorry for long delay.

Yes. I can see that packets when I type **candump can0**
Even if I add cangw rules like:
`
	post-up /usr/bin/cangw -A -s vcan0 -d can0 -e -f 18DA00F1:FFFF00FF
	post-up /usr/bin/cangw -A -s can0 -d vcan0 -e -f 18DAF100:FFFFFF00
`

hartkopp, 2023-10-07:
You have not set CAN_EFF_FLAG  ( = 0x80000000) in the can_mask.can_id !

Please try

```
	post-up /usr/bin/cangw -A -s vcan0 -d can0 -e -f 98DA00F1:FFFF00FF
	post-up /usr/bin/cangw -A -s can0 -d vcan0 -e -f 98DAF100:FFFFFF00
```

pompushko, 2023-10-07:
Thank you!

Works great!

closed 2023-10-07 by pompushko