linux-can/can-utils #467: Use of J1939, can-utils, RaspberryPi, ValueCan4

Craxer01, 2023-10-16:
Hello.
I just started working with this system:
RaspberryPi 3 B+
ValueCan 4

I just need to show the CANframes and save to a file.

When I run:
sudo modprobe can-j1939  There are no errors or any other messages so I assume it is all fine.

I followed all the instructions and commands from intrepidcs and the system works fine for the regular/common OBD.
I can see CAN traffic and save to a log file.

These are the steps I followed:

sudo modprobe can
sudo modprobe can_raw
sudo modprobe can_dev
sudo insmod intrepid.ko   **** I think this is to work with intrepidcs tools.

Then I run a deamon to detect intrepid tool connected.

Finally:

ip link set up can0
candump can0

And voila! This displays CAN frames  but ONLY for "Regular" OBD *** connected to a Chevy.
If I try this on a j1939 vehicle nothing appears on the screen.

So I think I need to enable J1939.

My question is, what commands do I need for J1939? 
Is this correct?

sudo modprobe can
sudo modprobe can_raw
sudo modprobe can_dev
sudo modprobe can-j1939
sudo insmod intrepid.ko

ip link set up can0
candump can0

Thanks a lot!

marckleinebudde, 2023-10-17:
Hello @Craxer01,

J1939 is a higher level CAN protocol on top of CAN frames with an extended ID. You can use `candump` to display connection-less J1939 traffic.

Maybe the J1939 traffic is on a different CAN bus on the OBD connector. How many CAN channels does the ValueCan 4 support?

Last but not least, the ValueCan 4 CAN driver is not part of the Mainline Linux kernel, but the drivers supplied by the vendor seem to work here.

regards,
Marc

Craxer01, 2023-10-17:
Hi.
Thank you for the answer.
ValueCan4 has two channels and I know for sure that the pins for j1939 on these vehicles are 6 & 14. Just the same as the Chevy I tested.

I know I can read regular CAN with (because it works fine):
```
sudo modprobe can
sudo modprobe can_raw
sudo modprobe can_dev
sudo insmod intrepid.ko **** I think this is to work with intrepidcs tools.
```

If I add: sudo modeprobe j1939, am I already stating to use j1939?
What do you think?

The frames I got with a PCAN, and that I would like to read are like: c560bd0x, 1823d0d9x, 18eaffd7x, 1821d0d9x, 1872d7d4x, 10ff2021x, 18ff25e4x

I will have access to the vehicle in some hours but I want to have something that has good chances to work as I will have just a couple (maybe less) of hours to work on it.

Thank you again.

marckleinebudde, 2023-10-17:
Does the vehicle send the J1939 messages on its own, or do you have to request them?

marckleinebudde, 2023-10-17:
If the device sends
1. J1939 on its own

and

2. on the same channel as the OBD messages

Then you already receive j1939 messages. Try `j1939spy -P can0`.

Thanks for the pointer to the intrepidcs website. I just wanted to mention, that if there is a problem with the driver (here seems to be everything OK) then you should ask the intrepidcs people, as the code is not part of the Linux kernel. Or better ask intrepidcs to bring their driver into the Linux kernel. 😸

olerem, 2023-10-17:
> It sends them by its own. My intention is to build a stand alone device that will save the logs during several hours. By the way, and with all the respect, this is the page I got instructions for the ValueCan if you want to take a look: https://intrepidcs.com/socketcan-vcan4/

Probably you wont something like in this discussion:
https://github.com/linux-can/can-utils/issues/381

Craxer01, 2023-10-18:
> > It sends them by its own. My intention is to build a stand alone device that will save the logs during several hours. By the way, and with all the respect, this is the page I got instructions for the ValueCan if you want to take a look: https://intrepidcs.com/socketcan-vcan4/
> 
> Probably you want something like in this discussion: #381

Thanks a lot for the reference. I saw it but honestly I didn´t read it because I thouhgt it wasn´t related but now that i think about it, maybe it is.
I didn´t have any chance to work on the vehicle today but I´ll try to post any progress.
Thanks again.

Craxer01, 2023-10-24:
woohoooo! it worked!
I can see j1939 frames from my vehicles. :)

marckleinebudde, 2023-10-24:
Hooray!

Can you summarize you working setup (for reference)?

Craxer01, 2023-10-25:
Yep!
So working with an Intrepidcs ValueCAN4 I followed their webpage to set up the intrepidcs dirvers in linux and the socket:
[Intrepid ValueCan Socket](https://intrepidcs.com/socketcan-vcan4/)

then I added the line modeprobe can-j1939 to thei modprobe´s:

sudo modprobe can
sudo modprobe can_raw
sudo modprobe can_dev
sudo modprobe can-j1939
sudo insmod intrepid.ko

ip link set up can0
candump can0

That was it. Now I will work on filtering messages because the file is too big. I am recording about 8 hours.

Now you can add the ValuCAN to the devices that work with can-utils !!

marckleinebudde, 2023-10-25:
Thanks for the summary!

closed 2023-10-25 by marckleinebudde