linux-can/can-utils #469: Candump dont show delta

Kuzma30, 2023-10-17:
My system is CM4 with MCP2515. 
Linux b2qt-raspberrypi4 6.1.54-rt15-v7l #1 SMP PREEMPT_RT Wed Sep 20 14:13:53 UTC 2023 armv7l GNU/Linux
Example of my log. (with -t delta option it show 0)
candump -t a -a can0
```
 (1697535023.000000)  can0  18EF8382   [7]  01 11 00 00 00 80 4F      '......O'
 (1697535023.000000)  can0  18EF8382   [7]  01 12 00 00 00 80 4F      '......O'
 (1697535023.000000)  can0  18EF8382   [7]  01 13 00 00 00 80 4F      '......O'
 (1697535023.000000)  can0  18EF8382   [7]  01 14 00 00 00 80 4F      '......O'
 (1697535023.000000)  can0  18EF8382   [7]  01 15 00 00 00 80 4F      '......O'
 (1697535023.000000)  can0  18EF8382   [7]  01 16 00 00 00 80 4F      '......O'
 (1697535023.000000)  can0  18EF8382   [7]  01 17 00 00 00 80 4F      '......O'
 (1697535023.000000)  can0  18EF8382   [7]  01 18 00 00 00 80 4F      '......O'
 (1697535023.000000)  can0  18EF8382   [7]  01 91 00 00 00 00 00      '.......'

```
But in real packet receive with delays. And I want to measure it. Such error I start receive after system update.

marckleinebudde, 2023-10-17:
What was your previous kernel version?

Kuzma30, 2023-10-17:
6.1.28

marckleinebudde, 2023-10-17:
There was no change in the `mcp251x` driver between these 2 kernel versions.

hartkopp, 2023-10-17:
Can you try to set `arm_64bit=0` in config.txt if this changes the behaviour?

https://github.com/hardbyte/python-can/issues/1652#issuecomment-1693764224

Kuzma30, 2023-10-17:
do not help

Kuzma30, 2023-10-19:
I check with latest non-RT Rasbain 32-bit on Rpi4b. candump works fine (show delta).

marckleinebudde, 2023-10-19:
`b2qt-raspberrypi4 6.1.54-rt15-v7l` looks like 32 bit, as well.

marckleinebudde, 2023-10-19:
can you only update the kernel to `b2qt-raspberrypi4 6.1.54-rt15-v7l` keeping the same root file system?

Kuzma30, 2023-10-19:
b2qt is Yocto based distro.
I rebuild my image without RT-patch and get same results - no delta info. 
Linux b2qt-raspberrypi4 6.1.54-v7l #1 SMP PREEMPT Wed Sep 20 14:13:53 UTC 2023 armv7l GNU/Linux

I use custom config for kernel. What options I must check in kernel. (Or I can upload my current config)

marckleinebudde, 2023-10-19:
Can you compile `2qt-raspberrypi4 6.1.54-v7l` with the problematic config and install it on the raspi image?

marckleinebudde, 2023-10-19:
Another option is to make sure, that you are using the latest `can-utils` in your yocto system.

Kuzma30, 2023-10-19:
I use can-utils  with SRCREV = "41ea759237bbd156d753251c573c280d96873528" as latest commit to build.

Kuzma30, 2023-10-19:
I make small patch to output debug info
```
diff --git a/candump.c b/candump.c
index 8e4444f..680dd60 100644
--- a/candump.c
+++ b/candump.c
@@ -766,6 +766,7 @@ int main(int argc, char **argv)
                                         */
                                        tv.tv_sec = stamp[2].tv_sec;
                                        tv.tv_usec = stamp[2].tv_nsec / 1000;
+                                       fprintf(stdout,"stamp[0].tv_nsec = %lld\nstamp[1].tv_nsec = %lld\n nstamp[2].tv_nsec = %lld\n", stamp[0].tv_nsec,stamp[1].tv_nsec,stamp[2].tv_nsec);
                                } else if (cmsg->cmsg_type == SO_RXQ_OVFL) {
                                        memcpy(&obj->dropcnt, CMSG_DATA(cmsg), sizeof(__u32));
                                }
```

When I run
candump -t delta -H can0
I see such log
```
 (000.000000)  can0  18EF83EF   [7]  01 36 00 00 00 00 00
stamp[0].tv_nsec = 276327893
stamp[1].tv_nsec = 0
 nstamp[2].tv_nsec = 0
 (000.000000)  can0  18EF83EF   [7]  01 37 00 00 00 00 00
stamp[0].tv_nsec = 277147204
stamp[1].tv_nsec = 0
 nstamp[2].tv_nsec = 0
 (000.000000)  can0  18EF83EF   [7]  01 61 00 BA A8 FF 3F
stamp[0].tv_nsec = 278121161
stamp[1].tv_nsec = 0
 nstamp[2].tv_nsec = 0
 (000.000000)  can0  18EF83EF   [7]  01 52 00 B1 68 DB 41
stamp[0].tv_nsec = 279125952
stamp[1].tv_nsec = 0
 nstamp[2].tv_nsec = 0
 (000.000000)  can0  18EF83EF   [7]  01 64 00 8F 39 19 40
stamp[0].tv_nsec = 280117373
stamp[1].tv_nsec = 0
 nstamp[2].tv_nsec = 0
 (000.000000)  can0  18EF83EF   [7]  01 53 00 8F 39 19 40
stamp[0].tv_nsec = 281221274
stamp[1].tv_nsec = 0
 nstamp[2].tv_nsec = 0
 (000.000000)  can0  18EF83EF   [7]  01 71 00 66 C6 C3 42
stamp[0].tv_nsec = 282229417
stamp[1].tv_nsec = 0
 nstamp[2].tv_nsec = 0
 (000.000000)  can0  18EF83EF   [7]  01 78 00 00 00 00 00
stamp[0].tv_nsec = 283140856
stamp[1].tv_nsec = 0
 nstamp[2].tv_nsec = 0
 (000.000000)  can0  18EF83EF   [7]  01 74 00 00 00 00 00
stamp[0].tv_nsec = 284144610
stamp[1].tv_nsec = 0
 nstamp[2].tv_nsec = 0
 (000.000000)  can0  18EF83EF   [7]  01 83 00 00 00 00 00
```

May be this will be usefull

marckleinebudde, 2023-10-19:
The driver doesn't support hardware timestamps. This is why you get `(000.000000)` as the timestamp if you start with the `-H` option.

marckleinebudde, 2023-10-19:
```
stamp[0].tv_nsec = 276327893
stamp[0].tv_nsec = 277147204
stamp[0].tv_nsec = 278121161
stamp[0].tv_nsec = 279125952
stamp[0].tv_nsec = 280117373
stamp[0].tv_nsec = 281221274
stamp[0].tv_nsec = 282229417
stamp[0].tv_nsec = 283140856
stamp[0].tv_nsec = 284144610
```

What's the supposed delay between 2 CAN frames?

Kuzma30, 2023-10-19:
Delay - 1 ms.
When I run without -H option I get 0 in delta
```
candump -t delta can0
 (000.000000)  can0  18EF8283   [3]  03 FF 00
 (000.000000)  can0  18EF8382   [7]  01 11 00 00 00 80 4F
 (000.000000)  can0  18EF8382   [7]  01 12 00 00 00 80 4F
 (000.000000)  can0  18EF8382   [7]  01 13 00 00 00 80 4F
 (000.000000)  can0  18EF8382   [7]  01 14 00 00 00 80 4F
 (000.000000)  can0  18EF8382   [7]  01 15 00 00 00 80 4F
 (000.000000)  can0  18EF8382   [7]  01 16 00 00 00 80 4F
 (000.000000)  can0  18EF8382   [7]  01 17 00 00 00 80 4F
 (000.000000)  can0  18EF8382   [7]  01 18 00 00 00 80 4F
 (000.000000)  can0  18EF8382   [7]  01 91 00 00 00 00 00
 (000.000000)  can0  18EFEF83   [3]  03 00 00

```

marckleinebudde, 2023-10-19:
Can you re-run with your debug output enabled?

Kuzma30, 2023-10-19:
https://github.com/linux-can/can-utils/issues/469#issuecomment-1770565548 
Logs from here is with debug output enabled.

marckleinebudde, 2023-10-19:
Please run `strace -o /tmp/log ./candump any,0:0,#FFFFFFFF -cexdtA` and send me the log output, it's in `/tmp/log`.

Kuzma30, 2023-10-19:
```
strace -o /tmp/log candump any,0:0,#FFFFFFFF -cexdtA
 (2023-10-19 11:31:44.4294967295)  can0  TX - -  18EF8183   [3]  03 FF 00
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8381   [7]  01 11 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8381   [7]  01 12 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8381   [7]  01 13 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8381   [7]  01 14 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8381   [7]  01 15 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8381   [7]  01 16 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8381   [7]  01 17 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8381   [7]  01 18 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8381   [7]  01 91 00 00 00 00 00
 (2023-10-19 11:31:44.3068705531)  can0  TX - -  18EF8483   [3]  03 FF 00
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8384   [7]  01 11 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8384   [7]  01 12 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8384   [7]  01 13 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8384   [7]  01 14 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8384   [7]  01 15 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8384   [7]  01 16 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8384   [7]  01 17 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8384   [7]  01 18 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8384   [7]  01 91 00 00 00 00 00
 (2023-10-19 11:31:44.3068705531)  can0  TX - -  18EF8283   [3]  03 FF 00
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8382   [7]  01 11 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8382   [7]  01 12 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8382   [7]  01 13 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8382   [7]  01 14 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8382   [7]  01 15 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8382   [7]  01 16 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8382   [7]  01 17 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8382   [7]  01 18 00 00 00 80 4F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF8382   [7]  01 91 00 00 00 00 00
 (2023-10-19 11:31:44.3068705531)  can0  TX - -  18EFEF83   [3]  03 00 00
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF83EF   [7]  01 35 00 BB BD BF 40
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF83EF   [7]  01 36 00 00 00 00 00
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF83EF   [7]  01 37 00 00 00 00 00
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF83EF   [7]  01 61 00 A4 A7 FF 3F
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF83EF   [7]  01 52 00 2F 18 0D 42
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF83EF   [7]  01 64 00 8F 39 19 40
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF83EF   [7]  01 53 00 8F 39 19 40
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF83EF   [7]  01 71 00 66 C6 C3 42
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF83EF   [7]  01 78 00 00 00 00 00
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF83EF   [7]  01 74 00 00 00 00 00
 (2023-10-19 11:31:44.3068705531)  can0  RX - -  18EF83EF   [7]  01 83 00 00 00 00 00
^Croot@b2qt-raspberrypi4:~cat /tmp/log
execve("/usr/bin/candump", ["candump", "any,0:0,#FFFFFFFF", "-cexdtA"], 0xbef47d00 /* 17 vars */) = 0
brk(NULL)                               = 0x223a000
mmap2(NULL, 8192, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0xb6f62000
access("/etc/ld.so.preload", R_OK)      = -1 ENOENT (No such file or directory)
openat(AT_FDCWD, "/etc/ld.so.cache", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=31226, ...}) = 0
mmap2(NULL, 31226, PROT_READ, MAP_PRIVATE, 3, 0) = 0xb6f5a000
close(3)                                = 0
openat(AT_FDCWD, "/usr/lib/libc.so.6", O_RDONLY|O_LARGEFILE|O_CLOEXEC) = 3
read(3, "\177ELF\1\1\1\0\0\0\0\0\0\0\0\0\3\0(\0\1\0\0\0\271\346\1\0004\0\0\0"..., 512) = 512
statx(3, "", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT|AT_EMPTY_PATH, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0755, stx_size=1146976, ...}) = 0
mmap2(NULL, 1184344, PROT_READ|PROT_EXEC, MAP_PRIVATE|MAP_DENYWRITE, 3, 0) = 0xb6e38000
mmap2(0xb6f35000, 110592, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_DENYWRITE, 3, 0xfc000) = 0xb6f35000
mmap2(0xb6f50000, 37464, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_FIXED|MAP_ANONYMOUS, -1, 0) = 0xb6f50000
close(3)                                = 0
set_tls(0xb6f630a0)                     = 0
set_tid_address(0xb6f62c08)             = 20953
set_robust_list(0xb6f62c0c, 12)         = 0
rseq(0xb6f63080, 0x20, 0, 0xe7f5def3)   = 0
mprotect(0xb6f35000, 106496, PROT_READ) = 0
mprotect(0x4d6000, 4096, PROT_READ)     = 0
mprotect(0xb6f7f000, 8192, PROT_READ)   = 0
ugetrlimit(RLIMIT_STACK, {rlim_cur=8192*1024, rlim_max=RLIM_INFINITY}) = 0
munmap(0xb6f5a000, 31226)               = 0
[... cut ...]
statx(AT_FDCWD, "/etc/localtime", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=111, ...}) = 0
write(1, " (2023-10-19 11:31:44.3068705531"..., 99) = 99
epoll_wait(3, 0xbecaf860, 1, -1)        = -1 EINTR (Interrupted system call)
--- SIGINT {si_signo=SIGINT, si_code=SI_KERNEL} ---
sigreturn({mask=[]})                    = -1 EINTR (Interrupted system call)
close(4)                                = 0
close(3)                                = 0
exit_group(130)                         = ?
+++ exited with 130 +++

```

marckleinebudde, 2023-10-19:
Which C library are you using? glibc? Which version?

marckleinebudde, 2023-10-19:
The difference between your strace output and mine is the different `cmsg_type=SO_TIMESTAMP*`. You have `cmsg_type=SO_TIMESTAMP_NEW`, I have `cmsg_type=SO_TIMESTAMP_OLD`.

```
recvmsg(4, {msg_name={sa_family=AF_CAN, sa_data="\0\0\10\0\0\0"}, msg_namelen=24 => 8, msg_iov=[{iov_base="\2\0\0\0\1\0\0\0\22\0\0\0\0\0\0\0", iov_len=72}], msg_iovlen=1, msg_control=[{cmsg_len=32, cmsg_level=SOL_SOCKET, cmsg_type=SO_TIMESTAMP_OLD, cmsg_data={tv_sec=1697716746, tv_usec=662521}}], msg_controllen=32, msg_flags=MSG_DONTROUTE}, 0) = 16
```

Kuzma30, 2023-10-19:
glibc 2.38
```
cat glibc-version.inc 
SRCBRANCH ?= "release/2.38/master"
PV = "2.38+git"
SRCREV_glibc ?= "1aed90c9c8f8be9f68b58e96b6e4cd0fc08eb2b1"
SRCREV_localedef ?= "e0eca29583b9e0f62645c4316ced93cf4e4e26e1"

GLIBC_GIT_URI ?= "git://sourceware.org/git/glibc.git;protocol=https"

UPSTREAM_CHECK_GITTAGREGEX = "(?P<pver>\d+\.\d+(\.(?!90)\d+)*)"

```

marckleinebudde, 2023-10-19:
Can you add this patch, rerun with `strace` like before and the the output.

```
index 8e4444fc51c4..6e6b21b5fe30 100644
--- a/candump.c
+++ b/candump.c
@@ -752,6 +752,7 @@ int main(int argc, char **argv)
                         for (cmsg = CMSG_FIRSTHDR(&msg);
                              cmsg && (cmsg->cmsg_level == SOL_SOCKET);
                              cmsg = CMSG_NXTHDR(&msg,cmsg)) {
+                                printf("%s: cmsg_type=%d, SO_TIMESTAMP=%d\n", __func__, cmsg->cmsg_type, SO_TIMESTAMP);
                                 if (cmsg->cmsg_type == SO_TIMESTAMP) {
                                         memcpy(&tv, CMSG_DATA(cmsg), sizeof(tv));
                                 } else if (cmsg->cmsg_type == SO_TIMESTAMPING) {
```

marckleinebudde, 2023-10-19:
please add the patch from https://github.com/linux-can/can-utils/issues/469#issuecomment-1770855399, compile and re-run.

Kuzma30, 2023-10-19:
Sorry 
```
strace -o /tmp/log candump any,0:0,#FFFFFFFF -cexdtA
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.4294967295)  can0  TX - -  18EF8183   [3]  03 FF 00
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8381   [7]  01 11 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8381   [7]  01 12 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8381   [7]  01 13 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8381   [7]  01 14 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8381   [7]  01 15 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8381   [7]  01 16 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8381   [7]  01 17 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8381   [7]  01 18 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8381   [7]  01 91 00 00 00 00 00
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  TX - -  18EF8483   [3]  03 FF 00
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8384   [7]  01 11 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8384   [7]  01 12 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8384   [7]  01 13 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8384   [7]  01 14 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8384   [7]  01 15 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8384   [7]  01 16 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8384   [7]  01 17 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8384   [7]  01 18 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8384   [7]  01 91 00 00 00 00 00
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  TX - -  18EF8283   [3]  03 FF 00
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8382   [7]  01 11 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8382   [7]  01 12 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8382   [7]  01 13 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8382   [7]  01 14 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8382   [7]  01 15 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8382   [7]  01 16 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8382   [7]  01 17 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8382   [7]  01 18 00 00 00 80 4F
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF8382   [7]  01 91 00 00 00 00 00
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  TX - -  18EFEF83   [3]  03 00 00
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF83EF   [7]  01 35 00 8C BE BF 40
main: cmsg_type=63, SO_TIMESTAMP=63
 (2023-10-19 12:43:41.3068484347)  can0  RX - -  18EF83EF   [7]  01 36 00 00 00 00 00
main: cmsg_type=63, SO_TIMESTAMP=63
[... 259 lines cut ...]
write(1, " (2023-10-19 12:43:41.3068484347"..., 99) = 99
epoll_wait(3, [{events=EPOLLIN, data={u32=4931680, u64=4931680}}], 1, -1) = 1
recvmsg(4, {msg_name={sa_family=AF_CAN, sa_data="\0\0\2\0\0\0"}, msg_namelen=24 => 8, msg_iov=[{iov_base="\357\203\357\230\7\0\0\0\1x\0\0\0\0\0\0", iov_len=72}], msg_iovlen=1, msg_control=[{cmsg_len=28, cmsg_level=SOL_SOCKET, cmsg_type=SO_TIMESTAMP_NEW, cmsg_data={tv_sec=1697719421, tv_usec=281371}}], msg_controllen=28, msg_flags=0}, 0) = 16
write(1, "main: cmsg_type=63, SO_TIMESTAMP"..., 36) = 36
statx(AT_FDCWD, "/etc/localtime", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=111, ...}) = 0
write(1, " (2023-10-19 12:43:41.3068484347"..., 99) = 99
epoll_wait(3, [{events=EPOLLIN, data={u32=4931680, u64=4931680}}], 1, -1) = 1
recvmsg(4, {msg_name={sa_family=AF_CAN, sa_data="\0\0\2\0\0\0"}, msg_namelen=24 => 8, msg_iov=[{iov_base="\357\203\357\230\7\0\0\0\1t\0\0\0\0\0\0", iov_len=72}], msg_iovlen=1, msg_control=[{cmsg_len=28, cmsg_level=SOL_SOCKET, cmsg_type=SO_TIMESTAMP_NEW, cmsg_data={tv_sec=1697719421, tv_usec=282284}}], msg_controllen=28, msg_flags=0}, 0) = 16
write(1, "main: cmsg_type=63, SO_TIMESTAMP"..., 36) = 36
[... cut ...]
statx(AT_FDCWD, "/etc/localtime", AT_STATX_SYNC_AS_STAT|AT_NO_AUTOMOUNT, STATX_BASIC_STATS, {stx_mask=STATX_BASIC_STATS|STATX_MNT_ID, stx_attributes=0, stx_mode=S_IFREG|0644, stx_size=111, ...}) = 0
write(1, " (2023-10-19 12:43:41.3068484347"..., 99) = 99
epoll_wait(3, 0xbe8f5860, 1, -1)        = -1 EINTR (Interrupted system call)
--- SIGINT {si_signo=SIGINT, si_code=SI_KERNEL} ---
sigreturn({mask=[]})                    = -1 EINTR (Interrupted system call)
close(4)                                = 0
close(3)                                = 0
exit_group(130)                         = ?
+++ exited with 130 +++

```

Kuzma30, 2023-10-19:
I try your branch.
Still no delta info.

marckleinebudde, 2023-10-19:
I've updated the branch, it should output some debug information.

```
main: sizeof(time_t)=8, sizeof(struct timeval)=16, sizeof(struct __kernel_old_timeval)=16, sizeof(struct __kernel_sock_timeval)=16
SO_TIMESTAMP_OLD=29: tv=1697725454.334298, c_tv=1697725454.334298
 (2023-10-19 16:24:14.334298)  vcan0  TX - -  002   [1]  00
SO_TIMESTAMP_OLD=29: tv=1697725462.246376, c_tv=1697725462.246376
 (2023-10-19 16:24:22.246376)  vcan0  TX - -  002   [1]  00
```

Kuzma30, 2023-10-19:
```
candump -t delta can0
main: sizeof(time_t)=8, sizeof(struct timeval)=16, sizeof(struct __kernel_old_timeval)=8, sizeof(struct __kernel_sock_timeval)=16
SO_TIMESTAMP_NEW=29: tv=28.1697726000, c_tv=122780.1697726000
 (000.000000)  can0  18EF8183   [3]  03 FF 00
SO_TIMESTAMP_NEW=29: tv=28.1697726000, c_tv=123478.1697726000
 (000.000000)  can0  18EF8381   [7]  01 11 00 00 00 80 4F
SO_TIMESTAMP_NEW=29: tv=28.1697726000, c_tv=124537.1697726000
 (000.000000)  can0  18EF8381   [7]  01 12 00 00 00 80 4F
SO_TIMESTAMP_NEW=29: tv=28.1697726000, c_tv=125589.1697726000
 (000.000000)  can0  18EF8381   [7]  01 13 00 00 00 80 4F
SO_TIMESTAMP_NEW=29: tv=28.1697726000, c_tv=126656.1697726000
 (000.000000)  can0  18EF8381   [7]  01 14 00 00 00 80 4F
SO_TIMESTAMP_NEW=29: tv=28.1697726000, c_tv=127699.1697726000
 (000.000000)  can0  18EF8381   [7]  01 15 00 00 00 80 4F
SO_TIMESTAMP_NEW=29: tv=28.1697726000, c_tv=128763.1697726000
 (000.000000)  can0  18EF8381   [7]  01 16 00 00 00 80 4F
SO_TIMESTAMP_NEW=29: tv=28.1697726000, c_tv=129812.1697726000
 (000.000000)  can0  18EF8381   [7]  01 17 00 00 00 80 4F
SO_TIMESTAMP_NEW=29: tv=28.1697726000, c_tv=130863.1697726000
 (000.000000)  can0  18EF8381   [7]  01 18 00 00 00 80 4F
SO_TIMESTAMP_NEW=29: tv=28.1697726000, c_tv=131930.1697726000
 (000.000000)  can0  18EF8381   [7]  01 91 00 00 00 00 00
SO_TIMESTAMP_NEW=29: tv=28.1697726000, c_tv=173302.1697726000
 (000.000000)  can0  18EF8483   [3]  03 FF 00

```

Kuzma30, 2023-10-19:
```
candump -t delta can0
main: sizeof(time_t)=8, sizeof(struct timeval)=16, sizeof(struct __kernel_old_timeval)=8, sizeof(struct __kernel_sock_timeval)=16
SO_TIMESTAMP_NEW=63: tv=4796948.1697731440, c_tv=496503.1697731440 0x00079377 0x00000000 0x65315370 0x00000000
 (000.000000)  can0  18EAFFFE   [3]  00 EE 00
SO_TIMESTAMP_NEW=63: tv=4796948.1697731440, c_tv=497153.1697731440 0x00079601 0x00000000 0x65315370 0x00000000
 (000.000000)  can0  18EEFF81   [8]  29 00 E0 FF 00 85 0A A0
SO_TIMESTAMP_NEW=63: tv=4796948.1697731440, c_tv=497712.1697731440 0x00079830 0x00000000 0x65315370 0x00000000
 (000.000000)  can0  18EEFF83   [8]  64 00 E0 FF 09 80 00 A0
SO_TIMESTAMP_NEW=63: tv=4796948.1697731441, c_tv=314122.1697731441 0x0004cb0a 0x00000000 0x65315371 0x00000000
 (000.000000)  can0  18EA8281   [3]  00 EE 00
SO_TIMESTAMP_NEW=63: tv=4796948.1697731441, c_tv=314488.1697731441 0x0004cc78 0x00000000 0x65315371 0x00000000
 (000.000000)  can0  18EA8284   [3]  00 EE 00
SO_TIMESTAMP_NEW=63: tv=4796948.1697731441, c_tv=314865.1697731441 0x0004cdf1 0x00000000 0x65315371 0x00000000
 (000.000000)  can0  18EAED84   [3]  00 EE 00
SO_TIMESTAMP_NEW=63: tv=4796948.1697731441, c_tv=315247.1697731441 0x0004cf6f 0x00000000 0x65315371 0x00000000
 (000.000000)  can0  18EAEF84   [3]  00 EE 00
SO_TIMESTAMP_NEW=63: tv=4796948.1697731441, c_tv=426721.1697731441 0x000682e1 0x00000000 0x65315371 0x00000000
 (000.000000)  can0  18EA82ED   [3]  00 EE 00
SO_TIMESTAMP_NEW=63: tv=4796948.1697731441, c_tv=427102.1697731441 0x0006845e 0x00000000 0x65315371 0x00000000
 (000.000000)  can0  18EAEFED   [3]  00 EE 00
SO_TIMESTAMP_NEW=63: tv=4796948.1697731441, c_tv=427670.1697731441 0x00068696 0x00000000 0x65315371 0x00000000
 (000.000000)  can0  18EEFF82   [8]  2C 00 F5 FF 00 85 0A A0
SO_TIMESTAMP_NEW=63: tv=4796948.1697731441, c_tv=676686.1697731441 0x000a534e 0x00000000 0x65315371 0x00000000
 (000.000000)  can0  18EAEF83   [3]  00 EE 00
SO_TIMESTAMP_NEW=63: tv=4796948.1697731441, c_tv=677160.1697731441 0x000a5528 0x00000000 0x65315371 0x00000000
 (000.000000)  can0  18EAED83   [3]  00 EE 00
SO_TIMESTAMP_NEW=63: tv=4796948.1697731441, c_tv=677790.1697731441 0x000a579e 0x00000000 0x65315371 0x00000000
 (000.000000)  can0  18EEFFED   [8]  2B 00 FD FF 00 81 0A A0
SO_TIMESTAMP_NEW=63: tv=4796948.1697731441, c_tv=756291.1697731441 0x000b8a43 0x00000000 0x65315371 0x00000000
 (000.000000)  can0  18EEFFEF   [8]  E9 03 E0 FF 00 80 0A A0

```

marckleinebudde, 2023-10-19:
Can you send me the output of `pahole candump.o`. You can use `pahole` of your host distribution. I'm especially interested in the `struct *timeval*`.

```
struct timeval {
        __time_t                   tv_sec;               /*     0     8 */
        __suseconds_t              tv_usec;              /*     8     8 */

        /* size: 16, cachelines: 1, members: 2 */
        /* last cacheline: 16 bytes */
};
struct __kernel_old_timeval {
        __kernel_long_t            tv_sec;               /*     0     8 */
        __kernel_long_t            tv_usec;              /*     8     8 */

        /* size: 16, cachelines: 1, members: 2 */
        /* last cacheline: 16 bytes */
};
struct __kernel_sock_timeval {
        __s64                      tv_sec;               /*     0     8 */
        __s64                      tv_usec;              /*     8     8 */

        /* size: 16, cachelines: 1, members: 2 */
        /* last cacheline: 16 bytes */
};
```

Kuzma30, 2023-10-20:
```
struct timeval {
        __time64_t                 tv_sec;               /*     0     8 */
        __suseconds64_t            tv_usec;              /*     8     8 */

        /* size: 16, cachelines: 1, members: 2 */
        /* last cacheline: 16 bytes */
};
struct __kernel_old_timeval {
        __kernel_long_t            tv_sec;               /*     0     4 */
        __kernel_long_t            tv_usec;              /*     4     4 */

        /* size: 8, cachelines: 1, members: 2 */
        /* last cacheline: 8 bytes */
};
struct __kernel_sock_timeval {
        __s64                      tv_sec;               /*     0     8 */
        __s64                      tv_usec;              /*     8     8 */

        /* size: 16, cachelines: 1, members: 2 */
        /* last cacheline: 16 bytes */
};
```

marckleinebudde, 2023-10-20:
I've updated the branch, it should output even more debug information.

Kuzma30, 2023-10-20:
```
candump -t delta can0
main: sizeof(time_t)=8, sizeof(struct timeval)=16, sizeof(struct __kernel_old_timeval)=8, sizeof(struct __kernel_sock_timeval)=16
SO_TIMESTAMP_NEW=63: tv=4911636.1697785647, c_tv=231528.1697785647 0x00038868 0x00000000 0x6532272f 0x00000000
 (000.000000)  can0  18EF8183   [3]  03 FF 00
SO_TIMESTAMP_NEW=63: tv=4911636.1697785647, c_tv=232241.1697785647 0x00038b31 0x00000000 0x6532272f 0x00000000
 (000.000000)  can0  18EF8381   [7]  01 11 00 00 00 80 4F
SO_TIMESTAMP_NEW=63: tv=4911636.1697785647, c_tv=233298.1697785647 0x00038f52 0x00000000 0x6532272f 0x00000000
 (000.000000)  can0  18EF8381   [7]  01 12 00 00 00 80 4F
SO_TIMESTAMP_NEW=63: tv=4911636.1697785647, c_tv=234352.1697785647 0x00039370 0x00000000 0x6532272f 0x00000000
 (000.000000)  can0  18EF8381   [7]  01 13 00 00 00 80 4F

```

marckleinebudde, 2023-10-20:
yet another update

Kuzma30, 2023-10-20:
```
candump -t delta can0
main: sizeof(time_t)=8, sizeof(struct timeval)=16, sizeof(struct __kernel_old_timeval)=8, sizeof(struct __kernel_sock_timeval)=16 (memcpy)
SO_TIMESTAMP_NEW=63: tv=1697787182.1697787182, c_tv=730792.1697787182 0x000b26a8 0x00000000 0x65322d2e 0x00000000
 (000.000000)  can0  18EF8483   [3]  03 FF 00
SO_TIMESTAMP_NEW=63: tv=1697787182.1697787182, c_tv=731497.1697787182 0x000b2969 0x00000000 0x65322d2e 0x00000000
 (000.000000)  can0  18EF8384   [7]  01 11 00 00 00 80 4F
SO_TIMESTAMP_NEW=63: tv=1697787182.1697787182, c_tv=732557.1697787182 0x000b2d8d 0x00000000 0x65322d2e 0x00000000
 (000.000000)  can0  18EF8384   [7]  01 12 00 00 00 80 4F
SO_TIMESTAMP_NEW=63: tv=1697787182.1697787182, c_tv=733613.1697787182 0x000b31ad 0x00000000 0x65322d2e 0x00000000
 (000.000000)  can0  18EF8384   [7]  01 13 00 00 00 80 4F

```

marckleinebudde, 2023-10-20:
yet another update

marckleinebudde, 2023-10-20:
Can you give me access to your yocto?

Kuzma30, 2023-10-20:
https://doc.qt.io/Boot2Qt/b2qt-how-to-create-b2qt-image.html
I use dev.xml. There is update 1 hour ago. I recompute my rootfs now

Kuzma30, 2023-10-20:
```
candump -t delta can0
main: sizeof(time_t)=8, sizeof(struct timeval)=16, sizeof(struct __kernel_old_timeval)=8, sizeof(struct __kernel_sock_timeval)=16 (memcpy/print)
SO_TIMESTAMP_NEW=63: tv=-1092684128.1697796645, c_tv=290993.1697796645 0x000470b1 0x00000000 0x65325225 0x00000000
 (000.000000)  can0  18EF8183   [3]  03 FF 00
SO_TIMESTAMP_NEW=63: tv=-1092684128.1697796645, c_tv=291688.1697796645 0x00047368 0x00000000 0x65325225 0x00000000
 (000.000000)  can0  18EF8381   [7]  01 11 00 00 00 80 4F
SO_TIMESTAMP_NEW=63: tv=-1092684128.1697796645, c_tv=292760.1697796645 0x00047798 0x00000000 0x65325225 0x00000000
 (000.000000)  can0  18EF8381   [7]  01 12 00 00 00 80 4F

```

marckleinebudde, 2023-10-20:
For whatever reason it seems that `usec` and `sec` are swapped. Please try again.

Kuzma30, 2023-10-20:
```
delta can0
main: sizeof(time_t)=8, sizeof(struct timeval)=16, sizeof(struct __kernel_old_timeval)=8, sizeof(struct __kernel_sock_timeval)=16 (memcpy/print/hack)
SO_TIMESTAMP_NEW=63: tv=1697800311.452331, c_tv=1697800311.1697800311 0x0006e6eb 0x00000000 0x65326077 0x00000000
 (000.000000)  can0  18EFEF83   [3]  03 00 00
SO_TIMESTAMP_NEW=63: tv=1697800311.453241, c_tv=1697800311.1697800311 0x0006ea79 0x00000000 0x65326077 0x00000000
 (910.000000)  can0  18EF83EF   [7]  01 35 00 81 BE BF 40
SO_TIMESTAMP_NEW=63: tv=1697800311.454218, c_tv=1697800311.1697800311 0x0006ee4a 0x00000000 0x65326077 0x00000000
 (977.000000)  can0  18EF83EF   [7]  01 36 00 1F EF FF 3F
SO_TIMESTAMP_NEW=63: tv=1697800311.455237, c_tv=1697800311.1697800311 0x0006f245 0x00000000 0x65326077 0x00000000
 (1019.000000)  can0  18EF83EF   [7]  01 37 00 00 00 00 00
```

marckleinebudde, 2023-10-20:
> https://doc.qt.io/Boot2Qt/b2qt-how-to-create-b2qt-image.html I use dev.xml. There is update 1 hour ago. I recompute my rootfs now

Which `MACHINE` are you using?

Kuzma30, 2023-10-20:
MACHINE=raspberrypi4

marckleinebudde, 2023-10-20:
Ok, the website only lists `raspberrypi4-64`, will try `raspberrypi4`.

Kuzma30, 2023-10-20:
You reproduce problem on your system?

marckleinebudde, 2023-10-20:
It failed to compile the whole system, but `meta/conf/distro/defaultsetup.conf` includes `conf/distro/include/time64.inc` which activates 64 bit `time_t`. I'll build a small yocto BSP next week.

Kuzma30, 2023-10-21:
https://sourceware.org/glibc/wiki/Y2038ProofnessDesign mb will be usefull

tisyang, 2023-11-27:
It maybe only a `sprint_timesamp` bug on some platform, like apline linux on armv7l/armhf.

I had meet same problem, `candump -ta`only show secs with sub-secs always be zero. If change `printf` formatter of `struct timeval.tv_sec` and `struct timeval.tv_usec` from `%lu` to `%llu`, 
then program will print correct timestamps.

Alpine linux use musl-libc, and musl use time64, link [musl time64 Release Notes](https://musl.libc.org/time64.html), so `time_t` and `struct timeval` use 64bit, not 32bit on armv7/armhf. So maybe we should change timestamp formatter in can-utils source code.

This only affect 32bit platform, 64bit will all be fine.

tisyang, 2023-11-27:
This screenshot shows compiling warning about `candump.c` on alpine armhf/armv7l, in fact it's in  an alpine armhf lxc container.

[image]

tisyang, 2023-11-27:
We should consider using C99 `long long` types for formatting timestamp:
example origin:
```c
sprintf(ts_buffer, "(%010lu.%06lu) ", tv->tv_sec, tv->tv_usec);
```
example new with C99:
```c
sprintf(ts_buffer, "(%010llu.%06llu) ", (unsigned long long)tv->tv_sec, (unsigned long long)tv->tv_usec);
```

closed 2023-11-30 by Kuzma30