linux-can/can-utils #479: Sending keyframes using can utils will result in multiple repetitions of the received packet

Hsensor, 2023-12-05:
sudo ip link set can0 down
sudo ip link set can0 type can bitrate 500000
sudo ip link set can0 up
sudo cansend can0 00000123#1122334455667788

marckleinebudde, 2023-12-05:
- What's your Hardware/SoC?
- Which Kernel version are you using?
- Where are you seeing multiple frames? On the receiving side?
- Describe your receiving side (Hardware/SoC/Kernel version).
- Describe your CAN Bus setup (CAN stations, bus Termination).

Hsensor, 2023-12-06:
Send the data like this first
sudo cansend ca0 00010123#1122334455667788
The receiving end only receives data once

Hsensor, 2023-12-06:
> Send the data like this first sudo cansend ca0 00010123#1122334455667788 The receiving end only receives data once
So I think the bug may have been generated here within the tool

marckleinebudde, 2023-12-06:
> The receiving end only receives data one time

This is good.

Sorry, I don't understand what your problem is. Can you please rephrase it.

Hsensor, 2023-12-06:
sudo cansend can0 00010123#1122334455667788  ---> Received only once
sudo cansend can0 123#1122334455667788  ---> Received only once
sudo cansend can0 00000123#1122334455667788   ---> Received multiple repeated frame data

marckleinebudde, 2023-12-06:
- What's your sending Hardware/SoC?
- Which Kernel version are you using?
- Describe your receiving side (Hardware/SoC/Kernel version).
- Describe your CAN Bus setup (CAN stations, bus Termination).

Hsensor, 2023-12-06:
> > `sudo cansend can0 00000123#1122334455667788` ---> Received multiple repeated frame data
> 
> How many times do you receive the frame?

yes

Hsensor, 2023-12-06:
> > `sudo cansend can0 00000123#1122334455667788` ---> Received multiple repeated frame data
> 
> How many times do you receive the frame?

Not a fixed number of times

marckleinebudde, 2023-12-06:
> Not a fixed number of times

Does it receive the CAN frame indefinitely? Or does it receive the frame a certain number of times, but that number varies?

Hsensor, 2023-12-06:
https://[fastice-tech.oss-cn-huhehaote.aliyuncs.com/can-utils.mp4](https://fastice-tech.oss-cn-huhehaote.aliyuncs.com/can-utils.mp4)

marckleinebudde, 2023-12-06:
Looking at you host name `orangepi5plus` indicates that you're probably using a [Orange Pi 5 Plus](http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-5-plus-32GB.html), which has a RK3588 SoC.

There is no mainline CAN support for the CAN controller in that chip. Further the CAN controller has some serious hardware problems. You are probably using the downstream driver by rockchip, which has some workarounds for the hardware problems, but the driver is not of good quality.