linux-can/can-utils #483: cangw and multiple vcan

pompushko, 2023-12-11:
Hello.

I have a strange issue with cangw and 2 virtual CANs.

cat /etc/network/interfaces.d/vcan0
```
auto vcan0
iface vcan0 inet manual
        pre-up /sbin/ip link add dev $IFACE type vcan
        pre-up /sbin/ip link set dev $IFACE mtu 16
        up /sbin/ifconfig $IFACE up mtu 16
        down /sbin/ifconfig $IFACE down
        post-up /usr/bin/cangw -A -s vcan0 -d can0 -e -f 98DA00F1:DFFF00FF
        post-up /usr/bin/cangw -A -s can0 -d vcan0 -e -f 98DAF100:DFFFFF00
```

cat /etc/network/interfaces.d/vcan1
```
auto vcan1
iface vcan1 inet manual
        pre-up /sbin/ip link add dev $IFACE type vcan
        pre-up /sbin/ip link set dev $IFACE mtu 16
        up /sbin/ifconfig $IFACE up mtu 16
        down /sbin/ifconfig $IFACE down
        post-up /usr/bin/cangw -A -s vcan1 -d can1 -e -f 98DA00F1:DFFF00FF
        post-up /usr/bin/cangw -A -s can1 -d vcan1 -e -f 98DAF100:DFFFFF00
```

My cangw rules works strange:

```
cangw -A -s can1 -d vcan1 -e -f 98DAF100:DFFFFF00 # 0 handled 0 dropped 0 deleted
cangw -A -s vcan1 -d can1 -e -f 98DA00F1:DFFF00FF # 12 handled 0 dropped 0 deleted           <=
cangw -A -s can0 -d vcan0 -e -f 98DAF100:DFFFFF00 # 18 handled 0 dropped 0 deleted           <=
cangw -A -s vcan0 -d can0 -e -f 98DA00F1:DFFF00FF # 0 handled 0 dropped 0 deleted
```

Some of mine packets go to second virtual interface, but should go to first one.
Very strange...

Or my rules are bad?

hartkopp, 2023-12-11:
Nothing looks wrong here.
You get traffic from can0 that is routed to vcan0 and you get traffic from vcan1 that is routed to can1.
There is no connection between *can0 and *can1  - or is there some userspace application for this?

pompushko, 2023-12-11:
Yep.... no connection. Even different speed :D

hartkopp, 2023-12-11:
So what is your problem then?
There are different filters too.