linux-can/can-utils #484: Help me with data padding

hungcuiga1, 2023-12-11:
Dear Develop team,
I am a newbie in C++ and I want to send data to can. I use ```iostpsend``` and see data is padding like I posted here https://stackoverflow.com/questions/77635005/padding-stdvectoruint8-t-in-embedded-c
I see this code over a week and I still can not understand where you do this future. Can you help me with that question? Thank you so much.

marckleinebudde, 2023-12-11:
The ISO-TP is implemented in the Linux Kernel, the padding is added there.

What is your actual task?

hungcuiga1, 2023-12-11:
Dear @marckleinebudde ,
Thank for your reply, I do not understand how this data is padded, I see option -p but I can not know what is follow code(how to it run). In https://stackoverflow.com/questions/77635005/padding-stdvectoruint8-t-in-embedded-c is my task. Sorry for my english. Thank you!

hartkopp, 2023-12-11:
So you want an example to configure the Linux isotp socket (in your C code) to provide a padding with 0x00 as shown on stackoverflow?

hungcuiga1, 2023-12-17:
Dear @marckleinebudde  @hartkopp ,
I can not send data have length 8192 bytes. It is error
[image]
Why i think it is a vector large

hungcuiga1, 2023-12-17:
Dear @marckleinebudde , @hartkopp 
I send 4096 bytes and I see result:
[image]
only have a line can0 seen. Please help me explain them. Thank you!

hartkopp, 2023-12-17:
Please send the output of `uname -a` and `cat /sys/module/can_isotp/parameters/max_pdu_size`

hungcuiga1, 2023-12-18:
Dear @hartkopp ,
I can not find ```/sys/module/can_isotp/parameters/max_pdu_size``` on ubuntu 20.04, please help me.

hartkopp, 2023-12-18:
Please send the output of `uname -a`

marckleinebudde, 2024-05-28:
Feel free to reopen, if you still have an issue.

closed 2024-05-28 by marckleinebudde