linux-can/can-utils #490: Miss frame when using canfdtest

fxt-7, 2024-01-16:
Dear all,

When I tried to run "./canfdtest -f 2 -g can0" to test the loopback funtion, I met the below problem ,could you please help to tell me how to debug it? When I using cansend and candump, it is normal.
ip -details  link show can0
can0: <NOARP,UP,LOWER_UP,ECHO> mtu 72 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10000
    link/can  promiscuity 0 minmtu 0 maxmtu 0
    can <LOOPBACK,FD> state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 0
          bitrate 500000 sample-point 0.875
          tq 25 prop-seg 34 phase-seg1 35 phase-seg2 10 sjw 1
          mcp25xxfd: tseg1 2..256 tseg2 1..128 sjw 1..128 brp 1..256 brp-inc 1
          dbitrate 8000000 dsample-point 0.600
          dtq 25 dprop-seg 1 dphase-seg1 1 dphase-seg2 2 dsjw 1
          mcp25xxfd: dtseg1 1..32 dtseg2 1..16 dsjw 1..16 dbrp 1..256 dbrp-inc 1
          clock 40000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535
uanme -a
**Linux kernel 5.4.254 #1 SMP PREEMPT Mon Jan 15 02:15:09 UTC 2024 aarch64 aarch64 aarch64 GNU/Linux**
send
[image]
receive
[image]

marckleinebudde, 2024-01-16:
Hey @fxt-7,

first of all, but unrelated to this issue, you should be using a proper mainline kernel with the mainline driver (`mcp251xfd`) for your `mcp2518fd` CAN controller. You're using the `mcp25xxfd`, which is a old version.

I think `canfdtest` doesn't properly work in loopback mode, it's not a problem on your side. Better setup a proper CAN bus and start `canfdtest -v can0` on one side and `canfdtest -vg can0` on the other side. If you have 2 CAN controllers on the same system, you can also use `can0` and `can1`.

fxt-7, 2024-01-17:
Hi @marckleinebudde,

Thank you for your quick and professional answer! ! I will update the driver version to the latest.
So if I want to test the loopback function in one can device,  I should use the other tools, just like cangen? Is there a tool similar to canfdtest for testing the loopback function?

Hope your answer~

marckleinebudde, 2024-01-17:
Consider updating your kernel, not just the CAN driver.

I've just updates the `canfdtest` in the https://github.com/linux-can/can-utils repo. It should now work on CAN devices in loopback mode, use `-xx` to filter out other and locally generated CAN traffic.

In one terminal:

```
canfdtest -xx -v can0
```

In another:

```
canfdtest -xx -vg can0
```

fxt-7, 2024-01-17:
Hi @marckleinebudde 

I can't update this product's kernel version, I must work on this kernel for the other work, so shall I update the driver to mcp251xfd base on the 5.4.254 kernel ?

On the other hand,thanks for your upadtes! ! I had tried the latest canfdtest to test loopback function, but how to determine if there is a frame drop, judge by these dot “.” ?
send
[image]

receive
[image]

Hopes your answer, thanks~