linux-can/can-utils #515: unable to transmit CAN frames with ixxat USB-to-CAN v2

josselinL69, 2024-04-24:
Hello,

**Subject : I try to set up a CAN connection between an STM32U5 and my computer using an USB-to-CAN v2.**

Currently i'm able to receive data from the STM32 in my computer with candump.
But when i try to send data to the STM from my computeur, there is error in communication.

I tried to send data in loopback mod in order to see sended data in candump but nothing is shown.

I have successfully installed the driver for ixxat USB-to-CAN in the kernel and there is no error when i do this command for initialization :

```
sudo ip link set down can0
sudo ip link set can0 type can bitrate 1000000
sudo ip link set can0 type can loopback on
sudo ip link set up can0
```

here the result of  ``  before sending data : 

here the result of  `` after sending data : 

Thanks per advance for any kind of help !

marckleinebudde, 2024-04-24:
Which driver are you using for the USB-to-CAN v2?

josselinL69, 2024-04-24:
I use this driver : 
[IXXAT_SocketCAN_2_0_378_Modified_2023-11-22Beta (copy).zip](https://github.com/linux-can/can-utils/files/15095953/IXXAT_SocketCAN_2_0_378_Modified_2023-11-22Beta.copy.zip)

marckleinebudde, 2024-04-24:
If you configure the driver for loopback mode you cannot send any data. If it's correctly implemented by the driver and USB firmware.

marckleinebudde, 2024-05-28:
Feel free to reopen, if you have any issues.

closed 2024-05-28 by marckleinebudde