linux-can/can-utils #520: CAN FD, data phase bitrate is equal to arbitration phase (RPI4 + MCP2518FD)

JuraszekL, 2024-05-08:
Hello team! 
I am testing the FD driver for the first time, and i am a bit confused about the results. To make the thing simple:
- my hardware config is RPI4 with with [CAN FD shield](https://wiki.seeedstudio.com/2-Channel-CAN-BUS-FD-Shield-for-Raspberry-Pi/). 
```
$ uname -a
Linux rpilite 6.1.21-v7l+ #1642 SMP Mon Apr  3 17:22:30 BST 2023 armv7l GNU/Linux
```
- both interfaces are connected together. 

- I've used the command from the official website:
```
$ sudo ip link set can0 up type can bitrate 500000 sample-point 0.75 dbitrate 4000000 dsample-point 0.8 fd on fd-non-iso off
$ sudo ip link set can1 up type can bitrate 500000 sample-point 0.75 dbitrate 4000000 dsample-point 0.8 fd on fd-non-iso off
```
that gives me this result:
```
$ ip -details link show can0
4: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 72 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can <FD> state ERROR-ACTIVE (berr-counter tx 0 rx 0) restart-ms 0 
          bitrate 500000 sample-point 0.750 
          tq 25 prop-seg 29 phase-seg1 30 phase-seg2 20 sjw 1
          mcp251xfd: tseg1 2..256 tseg2 1..128 sjw 1..128 brp 1..256 brp-inc 1
          dbitrate 4000000 dsample-point 0.800 
          dtq 25 dprop-seg 3 dphase-seg1 4 dphase-seg2 2 dsjw 1
          mcp251xfd: dtseg1 1..32 dtseg2 1..16 dsjw 1..16 dbrp 1..256 dbrp-inc 1
          clock 40000000 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 
```
- i can send and receive CAN FD frames:
```
$ cansend can0 123##0112233445566778899101011111111223344556677899665544343
```
```
$ candump can1
can1  123  [32]  11 22 33 44 55 66 77 88 99 10 10 11 11 11 11 22 33 44 55 66 77 89 96 65 54 43 43 00 00 00 00 00
```

- bitrate of the bus doesn't change during the data phase:

[image]

As You can see, max. bit width is 2us, and that gives 1/0,000002s = 500000b/s, which is fine for arbitration phase, but not for the data phase. Ignore the fact that Saleae doesn't recognize the data fields properly - the CAN FD analyzer is not implemented yet. 

The MCP2518FD CAN controller is compliant with ISO 11898-1:2015.

Am i missing something? Please let me know what i'm doing wrong.

BR.
Łukasz

hartkopp, 2024-05-10:
Everything is fine in your setup.

The BRS (bit rate setting) bit is part of every CAN FD frame.

In your example
`$ cansend can0 123##0112233445566778899101011111111223344556677899665544343`
The BRS bit is not set. The struct canfd_frame.flags (directly behind the two `##`'s) is zero.

Try
`$ cansend can0 123##5112233445566778899101011111111223344556677899665544343`

That should make it.

Btw. `fd-non-iso off` is obsolete, as this setting is off by default.

JuraszekL, 2024-05-10:
There couldn't be a better example of how to use the BRS bit than my issue. Thank You so much!

closed 2024-05-12 by hartkopp