linux-can/can-utils #523: Error when using dbitrate with fd on : RTNETLINK answers : Operation not supported

VarnL0, 2024-05-17:
Hi, 

I am using a can-fd capable usb-to-can adapter, however I cannot manage to use the fd option without having error messages
Normal can configuration like this works fine:

`sudo ip link set can0 type can bitrate 500000`

Though the following command returns an error : 

`sudo ip link set can0 type can bitrate 500000 dbitrate 1000000 fd on
RTNETLINK answers: Operation not supported`

Is there a workaround to this ? 

Thanks

marckleinebudde, 2024-05-17:
What's the output of `ip --details -s link show can0`. (Please paste the text, no screenshot.)

VarnL0, 2024-05-17:
I get the following output :

`$ ip --details -s link show can0
6: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state STOPPED restart-ms 0 
	  bitrate 500000 sample-point 0.875 
	  tq 125 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
	  gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp-inc 1
	  clock 48000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 parentbus usb parentdev 1-6.1:1.0 
    RX:  bytes packets errors dropped  missed   mcast           
             0       0      0       0       0       0 
    TX:  bytes packets errors dropped carrier collsns           
             0       0      0       0       0       0 
`

marckleinebudde, 2024-05-17:
Which kernel are you using? It's too old and doesn't support CAN-FD for the `gs_usb` driver. Update your kernel or you can try this external module: https://github.com/marckleinebudde/gs_usb-external

```
sudo apt install git dkms                                                                                                                                                                     
git clone https://github.com/marckleinebudde/gs_usb-external.gitgs_usb-external.git                                                                                                                                  
sudo dkms add gs_usb-external                                                                                                                                                                 
sudo dkms install -m gs_usb -v 20221019-1                                                                                                                                                     
```

Make sure to unload the original driver and load the new one:

```
sudo modprobe -r gs_usb                                                                                                                                                                       
sudo modprobe gs_usb                      
```

marckleinebudde, 2024-05-17:
which device are you using exactly? what's the output of `lsusb`

marckleinebudde, 2024-05-17:
> I am using kernel version 6.5.0-15-generic so it should be recent enough I guess.

ACK

VarnL0, 2024-05-17:
I tried the two following devices : 
```lsusb
Bus 001 Device 008: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter
Bus 001 Device 005: ID 1d50:606f OpenMoko, Inc. Geschwister Schneider CAN adapter
Bus 001 Device 010: ID 16d0:117e MCS CANable2 b158aa7 github.com/normaldotcom/canable2.git
```
I didn't managed to make any of them work with can fd

marckleinebudde, 2024-05-17:
Please unplug, `sudo dmesg -c`, plug one after the other, and send me the output of `sudo dmesg`.

VarnL0, 2024-05-17:
sudo dmesg gives me the following output : 
```$ sudo dmesg
[101341.003768] usb 1-7: new full-speed USB device number 14 using xhci_hcd
[101341.153527] usb 1-7: New USB device found, idVendor=16d0, idProduct=117e, bcdDevice= 2.00
[101341.153542] usb 1-7: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[101341.153548] usb 1-7: Product: CANable2 b158aa7 github.com/normaldotcom/canable2.git
[101341.153553] usb 1-7: Manufacturer: Openlight Labs
[101341.153557] usb 1-7: SerialNumber: 205739605942
[101341.156141] cdc_acm 1-7:1.0: ttyACM0: USB ACM device
[101396.645000] usb 1-6: new high-speed USB device number 15 using xhci_hcd
[101396.793348] usb 1-6: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.11
[101396.793363] usb 1-6: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[101396.793369] usb 1-6: Product: USB 2.0 Hub
[101396.794934] hub 1-6:1.0: USB hub found
[101396.794978] hub 1-6:1.0: 4 ports detected
[101397.441063] usb 1-6.1: new full-speed USB device number 16 using xhci_hcd
[101397.663101] usb 1-6.1: New USB device found, idVendor=1d50, idProduct=606f, bcdDevice= 0.00
[101397.663115] usb 1-6.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[101397.663122] usb 1-6.1: Product: USB2CAN V3.3
[101397.663126] usb 1-6.1: Manufacturer: InnoMaker
[101397.663130] usb 1-6.1: SerialNumber: 003D00494657531120343133
[101397.669333] gs_usb 1-6.1:1.0: Configuring for 1 interfaces
[101397.869053] usb 1-6.2: new full-speed USB device number 17 using xhci_hcd
[101398.091229] usb 1-6.2: New USB device found, idVendor=1d50, idProduct=606f, bcdDevice= 0.00
[101398.091243] usb 1-6.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[101398.091250] usb 1-6.2: Product: USB2CAN V3.3
[101398.091254] usb 1-6.2: Manufacturer: InnoMaker
[101398.091258] usb 1-6.2: SerialNumber: 003D001E4657531120343133
[101398.097556] gs_usb 1-6.2:1.0: Configuring for 1 interfaces
```
ip --details -s link show gives me the following output : 

```1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00 promiscuity 0 minmtu 0 maxmtu 0 addrgenmode eui64 numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 
    RX:  bytes packets errors dropped  missed   mcast           
     128430827  664647      0       0       0       0 
    TX:  bytes packets errors dropped carrier collsns           
     128430827  664647      0       0       0       0 
2: enp8s0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN mode DEFAULT group default qlen 1000
    link/ether 78:d0:04:2d:bc:48 brd ff:ff:ff:ff:ff:ff promiscuity 0 minmtu 68 maxmtu 9216 addrgenmode none numtxqueues 8 numrxqueues 8 gso_max_size 65536 gso_max_segs 65535 parentbus pci parentdev 0000:08:00.0 
    RX:  bytes packets errors dropped  missed   mcast           
             0       0      0       0       0       0 
    TX:  bytes packets errors dropped carrier collsns           
             0       0      0       0       0       0 
3: enp0s31f6: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc fq_codel state DOWN mode DEFAULT group default qlen 1000
    link/ether 78:d0:04:2d:bc:47 brd ff:ff:ff:ff:ff:ff promiscuity 0 minmtu 68 maxmtu 9000 addrgenmode none numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 parentbus pci parentdev 0000:00:1f.6 
    RX:  bytes packets errors dropped  missed   mcast           
             0       0      0       0       0       0 
    TX:  bytes packets errors dropped carrier collsns           
             0       0      0       0       0       0 
4: enp1s0f0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN mode DEFAULT group default qlen 1000
    link/ether b4:96:91:40:87:d8 brd ff:ff:ff:ff:ff:ff promiscuity 0 minmtu 68 maxmtu 9710 addrgenmode none numtxqueues 64 numrxqueues 64 gso_max_size 65536 gso_max_segs 65535 parentbus pci parentdev 0000:01:00.0 
    RX:  bytes packets errors dropped  missed   mcast           
             0       0      0       0       0       0 
    TX:  bytes packets errors dropped carrier collsns           
             0       0      0       0       0       0 
5: enp1s0f1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq state UP mode DEFAULT group default qlen 1000
    link/ether b4:96:91:40:87:d9 brd ff:ff:ff:ff:ff:ff promiscuity 0 minmtu 68 maxmtu 9710 addrgenmode none numtxqueues 64 numrxqueues 64 gso_max_size 65536 gso_max_segs 65535 parentbus pci parentdev 0000:01:00.1 
    RX:    bytes   packets errors dropped  missed   mcast           
    848920324809 601207995      0  107026       0   40395 
    TX:    bytes   packets errors dropped carrier collsns           
      1425198358   1448935      0       0       0       0 
8: can2: <NOARP,ECHO> mtu 16 qdisc pfifo_fast state DOWN mode DEFAULT group default qlen 10
    link/can  promiscuity 0 minmtu 0 maxmtu 0 
    can state STOPPED (berr-counter tx 0 rx 0) restart-ms 0 
	  bitrate 500000 sample-point 0.875 
	  tq 125 prop-seg 6 phase-seg1 7 phase-seg2 2 sjw 1
	  sja1000: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..64 brp-inc 1
	  clock 8000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
[... 29 lines cut ...]
    TX:  bytes packets errors dropped carrier collsns           
       3371375  481625      0       0       0       0 
18: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 10
[... cut ...]
	  gs_usb: tseg1 1..16 tseg2 1..8 sjw 1..4 brp 1..1024 brp-inc 1
	  clock 48000000 
	  re-started bus-errors arbit-lost error-warn error-pass bus-off
	  0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 parentbus usb parentdev 1-6.2:1.0 
    RX:  bytes packets errors dropped  missed   mcast           
             0       0      0       0       0       0 
    TX:  bytes packets errors dropped carrier collsns           
             0       0      0       0       0       0 
```

marckleinebudde, 2024-05-17:
The firmware on non of these devices supports/exposes CAN-FD support via USB to Linux.

This is how a supported device looks like:
```
187: can0: <NOARP,UP,LOWER_UP,ECHO> mtu 72 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0 allmulti 0 minmtu 0 maxmtu 0 
    can <FD> state ERROR-ACTIVE restart-ms 1000 
	 bitrate 1000000 sample-point 0.750
	 tq 25 prop-seg 14 phase-seg1 15 phase-seg2 10 sjw 5 brp 1
	 gs_usb: tseg1 1..256 tseg2 1..128 sjw 1..128 brp 1..512 brp_inc 1
	 dbitrate 5000000 dsample-point 0.750
	 dtq 25 dprop-seg 2 dphase-seg1 3 dphase-seg2 2 dsjw 1 dbrp 1
	 gs_usb: dtseg1 1..32 dtseg2 1..16 dsjw 1..16 dbrp 1..32 dbrp_inc 1
	 clock 40000000 
	 re-started bus-errors arbit-lost error-warn error-pass bus-off
	 0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 tso_max_size 65536 tso_max_segs 65535 gro_max_size 65536 gso_ipv4_max_size 65536 gro_ipv4_max_size 65536 parentbus usb parentdev 2-1.3.3:1.0 
    RX:  bytes packets errors dropped  missed   mcast           
             0       0      0       0       0       0 
    TX:  bytes packets errors dropped carrier collsns           
             0       0      0       0       0       0 
188: can1: <NOARP,UP,LOWER_UP,ECHO> mtu 72 qdisc pfifo_fast state UP mode DEFAULT group default qlen 10
    link/can  promiscuity 0 allmulti 0 minmtu 0 maxmtu 0 
    can <FD> state ERROR-ACTIVE restart-ms 1000 
	 bitrate 1000000 sample-point 0.750
	 tq 25 prop-seg 14 phase-seg1 15 phase-seg2 10 sjw 5 brp 1
	 gs_usb: tseg1 1..256 tseg2 1..128 sjw 1..128 brp 1..512 brp_inc 1
	 dbitrate 5000000 dsample-point 0.750
	 dtq 25 dprop-seg 2 dphase-seg1 3 dphase-seg2 2 dsjw 1 dbrp 1
	 gs_usb: dtseg1 1..32 dtseg2 1..16 dsjw 1..16 dbrp 1..32 dbrp_inc 1
	 clock 40000000 
	 re-started bus-errors arbit-lost error-warn error-pass bus-off
	 0          0          0          0          0          0         numtxqueues 1 numrxqueues 1 gso_max_size 65536 gso_max_segs 65535 tso_max_size 65536 tso_max_segs 65535 gro_max_size 65536 gso_ipv4_max_size 65536 gro_ipv4_max_size 65536 parentbus usb parentdev 2-1.3.3:1.0 
    RX:  bytes packets errors dropped  missed   mcast           
             0       0      0       0       0       0 
    TX:  bytes packets errors dropped carrier collsns           
             0       0      0       0       0       0 
```

See the `can <FD>`.

```
Mai 17 16:40:35 hardanger kernel: usb 2-1.3.3: new full-speed USB device number 85 using xhci_hcd
Mai 17 16:40:36 hardanger kernel: usb 2-1.3.3: New USB device found, idVendor=1d50, idProduct=606f, bcdDevice= 0.00
Mai 17 16:40:36 hardanger kernel: usb 2-1.3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mai 17 16:40:36 hardanger kernel: usb 2-1.3.3: Product: candleLightFD gs_usb
Mai 17 16:40:36 hardanger kernel: usb 2-1.3.3: Manufacturer: candleLightFD
Mai 17 16:40:36 hardanger kernel: usb 2-1.3.3: SerialNumber: 003200205841500B20373233
Mai 17 16:40:36 hardanger kernel: gs_usb 2-1.3.3:1.0: Configuring for 2 interfaces
```

VarnL0, 2024-05-17:
Oh yes I see, that's surprising... CANable  wrote on their github (https://github.com/normaldotcom/canable2-fw) that their firmware supports can-fd and it is especially for this that I bought their device. I will order an actual USB to can-fd device and hope that it actually supports can fd this time. 
Anyway thanks a lot for your precious help

marckleinebudde, 2024-05-18:
It says:
> This firmware implements non-standard slcan commands to support CANFD messaging alongside a LAWICEL-style command set.
That means it doesn't support the `gs_usb` USB protocol :(

closed 2024-05-18 by marckleinebudde