linux-can/can-utils #525: New tools: canerrsim and canerrdump

zeljkoavramovic, 2024-05-22:
Developing robust CAN software is not easy when you do not have a quick way to generate all kinds of CAN errors. So I created such tool and called it canerrsim. Although candump is able to show CAN errors, I have created canerrdump which is specialized to do just that.

I would like to donate both canerrsim and canerrdump to can-utils project, but before doing that I would like to discuss them here and find out if anything needs to be changed to reach that goal (like different code formatting, or adding features like timestamp to canerrdump).

For now, you can find source code of both tools in the attachments.

[canerrsim.zip](https://github.com/linux-can/can-utils/files/15408116/canerrsim.zip)
[canerrdump.zip](https://github.com/linux-can/can-utils/files/15408118/canerrdump.zip)

marckleinebudde, 2024-05-23:
Hey @zeljkoavramovic,

thanks for your contribution. `canerrsim` looks interesting, but I think the functionality of `canerrdump` is already covered with `candump`. Compare the output of both tools:

```
$ candump any,0:0,#FFFFFFFF -cexdtd
 (590.634662)  vcan0  TX - -  20000025   [8]  00 40 00 00 00 00 00 00   ERRORFRAME
        tx-timeout
        controller-problem{back-to-error-active}
        no-acknowledgement-on-tx
```

```
$ ./canerrdump vcan0
CAN Sockets Error Messages Dumper
Listening CAN bus vcan0 for errors...
0x025 [8] 00 40 00 00 00 00 00 00  ERR=TxTimeout,NoAck,Ctrl(Active)
```

zeljkoavramovic, 2024-05-23:
Hello @marckleinebudde,

I thank you for your comments and I am glad you like canerrsim. Please let me know if there are any changes needed before it's acceptance.

It's fine if canerrdump is not accepted, but please let me try to explain why I still think it should be. There are 2 main use cases for canerrdump: troubleshooting CAN errors, and as a checker tool when you develop CAN application that yow want to react properly on some specific CAN errors. In both cases it is beneficial if you can hide regular messages, show all errors, or ignore some specific annoying recurring errors that are not in your current focus. This is where specialized tool like canerrdump shines. For example, using simple switches like IgnoreTxTimeout, IgnoreController and IgnoreBusError is very descriptive and does the job quickly. While it can be done with candump, you would need to browse through many documents and do some bit calculations before figuring out the proper hex mask. What do you think?

Btw, I have just found one issue while testing:

`./canerrsim vcan0 RxCount=10 TxCount=20 NoAck`

produces this output in canerrdump:
`0x220 [8] 00 00 00 00 00 00 20 10  ERR=NoAck,Count(TX=32,RX=16)`

and this in candump:
`Error class 0x220 is invalid
 (2024-05-23 06:24:59.302032)  vcan0  20000220   [8]  00 00 00 00 00 00 20 10   ERRORFRAME
        20000220   [8]  00 00 00 00 00 00 20 10   ERRORFRAME`

I found that this code snippet from candump.c is responsible for such error message:
`	class = cf->can_id & CAN_EFF_MASK;
	if (class > (1 << ARRAY_SIZE(error_classes))) {
		fprintf(stderr, "Error class %#x is invalid\n", class);
		return 0;
	}`

Do you know if 0x220 is really invalid? If so, then I should probably forbid such a combination in canerrsim output.

marckleinebudde, 2024-05-23:
I've imported the canerrsim and done some cleanups and creates PR #528 for further review.