linux-can/can-utils #539: That was easy to help on empty problems!

striork, 2024-05-28:
That was easy to help on empty problems! <--- If only they all were that easy!

_Originally posted by @hartkopp in https://github.com/linux-can/can-utils/issues/538#issuecomment-2135966121_
            Hey fellas, I'm in desperate need of some assistance. It appears I have Bitten off more than I can chew! I'm attempting a RPi Android Head Unit. Once I put the Android on the RPI 5 I seemingly lose access to anything involving can! (Termux doesn't allow sudo apt-get) I happened to stumble upon the work of you fine gents, but I'm admittedly in over my head! Having never compiled C-plus anything before today, I'm pretty much following what I can find online verbatim!
So, I clone under the 'app folder"
jr@Expiremental:~/AndroidStudioProjects/Can/app$ git clone https://github.com/linux-can/can-utils.git Cloning into 'can-utils'...
Then I pretty much Copy and Paste from your instructions... change the path to the toolchain... add the subdirectory to the makefile.txt via Android Studio
jr@Expiremental:~/AndroidStudioProjects/Can/app/build$ cmake -DCMAKE_TOOLCHAIN_FILE=~/home/jr/Android/Sdk/ndk/27.0.11718014/build/cmake/android.toolchain.cmake -DANDROID_PLATFORM=android-27 -DANDROID_ABI=arm64-v8a .. && make 
`CMake Warning:
Ignoring extra path from command line:

".."

CMake Error: The source directory "/home/jr/AndroidStudioProjects/Can/app" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
`

Clearly I'm missing something(s)

marckleinebudde, 2024-05-28:
Try:

```
cd ~/AndroidStudioProjects/Can/app
cmake -DCMAKE_TOOLCHAIN_FILE=/home/jr/Android/Sdk/ndk/27.0.11718014/build/cmake/android.toolchain.cmake -DANDROID_PLATFORM=android-27 -DANDROID_ABI=arm64-v8a -B build
cmake --build build
```

striork, 2024-05-28:
You probably did that in your sleep!!  lol
Gotta run to work, i'll check it all out after.... it did build though!  thanks!

closed 2024-05-28 by striork

marckleinebudde, 2024-05-28:
Not sleeping yet :)