linux-can/can-utils #544: How do I compile AOSP without the Android.mk file?

osos11-Git, 2024-06-05:
Hi, 
I have an AOSP that keeps track of the master branch. So, Android.mk file has been deleted. 
How can I integrate the new build system into my AOSP without the Android.mk file?

marckleinebudde, 2024-06-05:
https://github.com/linux-can/can-utils#cmake-project-generator
```
cmake -DCMAKE_BUILD_TYPE=Debug -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK/build/cmake/android.toolchain.cmake -B build
cmake --build build
```

osos11-Git, 2024-06-05:
@marckleinebudde 
If I remember correctly, it is not possible to build cmake projects in the AOSP build system.
If I wrong, Where exactly should I add this project and the build codes you wrote?

olerem, 2024-06-05:
https://developer.android.com/ndk/guides/cmake
https://cmake.org/cmake/help/latest/manual/cmake-toolchains.7.html#cross-compiling-for-android

osos11-Git, 2024-06-06:
I dont want to compile any app/lib via NDK for user space or i dont want to compile in Termux via cmake.
I am talking about AOSP, I mean compiling the whole Android Operating System with Can-utils and putting Can-utils in the system/bin or vendor/bin. 
I am writing again, it is not possible to directly build cmake projects in the AOSP build system. AOSP supports GNU Make and Soong.
In this vein, I need Android.mk(Make) or Android.bp(Soong) file to compiling in AOSP. 

Please tell me if you know that any proper way for compiling cmake in the AOSP because i dont know.

Thanks.

olerem, 2024-06-06:
OK, i see. I do not have any expertise in AOSP. This was one of the reasons to clean up and drop some of the build systems -  we can't maintain them all.

osos11-Git, 2024-06-07:
OK. Its understandable... 
I think, I have to create my Android.mk/.bp file myself and walk this way alone. :smile: 
Btw, What are the build systems currently available? Only cmake?

hartkopp, 2024-06-07:
We have cmake and the 'plain old Makefile' in the top of the repo, so that you can still use `make && make install` to build and install can-utils.

closed 2024-06-07 by osos11-Git