linux-can/can-utils #559: using our proprietory can emulator at company and NVIDIA Jetson AGX ORIN 64g devkit I only get 4 CAN signals while using UBUNTU 22.04 laptop, I get all 20 signals, why?

monajalal, 2024-10-18:
When I connect the Jetson device with Ubuntu 22.04 to a proprietary can emulator (cannot share more) I get only 4 signals however when I connect my laptop to same emulator I get all 20 can signals (including those 4).

What is the problem and how could it be fixed?
I am not able to share what those signals are.

I am getting CAN through CAN2 in jetson and through CAN0 in laptop.

I am using PCAN-USB.

olerem, 2024-10-20:
Hi @monajalal,

First of all, we need to clarify the context to properly understand your question.

Based on your description, it seems like you’re referring to CAN frames when you mention "signals."

Let’s break down the test setup as we understand it:

- Jetson AGX ORIN 64GB devkit using mttcan? (Ubuntu 22.04) using CAN2 interface is connected via CAN bus to a proprietary CAN emulator. In this scenario, only 4 CAN frames are received.

- Laptop (Ubuntu 22.04) using PCAN-USB (CAN0 interface) is connected via CAN bus to the same proprietary CAN emulator. In this case, more than 4 CAN frames are received, including the 4 seen with the Jetson device.

So, the main problem in this configuration is the Jetson board, which is using a potentially non-mainline mttcan driver. Correct?

olerem, 2024-10-20:
Ah, there is already Issue #558

marckleinebudde, 2024-10-21:
Can you connect the Jetson board and your laptop at the same time to the CAN bus? Or connect the Jetson board only to your laptop. As @hartkopp said in https://github.com/linux-can/can-utils/issues/170#issuecomment-2359299295 your CAN bus might not be correctly terminated.

monajalal, 2024-10-21:
@marckleinebudde well this is a different issue. previously we couldn't even get connected to CAN2 at all now we connect and get live signals only from four of the available CAN signals not all the 20.

marckleinebudde, 2024-10-21:
I assume that you mean CAN frames by signals. If you mean something else than CAN frames, then please clarify.

Can you connect the Jetson board and your laptop at the same time to the CAN bus and compare the received CAN frames?

monajalal, 2024-10-21:
@marckleinebudde sorry I have very limited CAN knowledge, what do you call the output of this?
`awk -F' ' '{print $2}' test_can2_out`

marckleinebudde, 2024-10-21:
Each line `candump` prints is the dump of a CAN message or CAN frame. Usually we call it CAN frame.

monajalal, 2024-10-21:
> Each line `candump` prints is the dump of a CAN message or CAN frame. Usually we call it CAN frame.

@marckleinebudde thanks for explanation. In this scenario, what do you suggest the path for debugging and fixing the problem?

olerem, 2024-10-21:
Please try ethtool -i can0, the can0 should be replaced to the actual interface name.

marckleinebudde, 2024-10-21:
Can you connect the Jetson board and your laptop at the same time to the CAN bus and compare the `candump` output.

monajalal, 2024-10-21:
> Can you connect the Jetson board and your laptop at the same time to the CAN bus and compare the `candump any,0:0,#FFFFFFFF -cexdtA` output.

I cannot connect them simultaneously to the CAN Emulator. I only have one PCAN-USB cable.

marckleinebudde, 2024-10-21:
you only need to connect 2 wires (CAN-high and CAN-low).

closed 2024-10-21 by monajalal