linux-can/can-utils #560: Potential error in kernel doc regarding J1939 destination address with PDU1 format

xile273, 2024-10-23:
In the kernel documentation for J1939 in the section about the PGN
the following is stated:

"On the other hand, when using PDU1 format, the PS-field contains a so-called Destination Address, which is _not_ part of the PGN. When communicating a PGN from user space to kernel (or vice versa) and **PDU2** format is used, the PS-field of the PGN shall be set to zero. The Destination Address shall be set elsewhere."

Shouldn't it be:
"When communicating a PGN from user space to kernel (or vice versa) and **PDU1** format is used, the PS-field of the PGN shall be set to zero. The Destination Address shall be set elsewhere."?

PDU2 format does not have a destination address, just a group extension.

Or am I misunderstanding something?

See: https://docs.kernel.org/networking/j1939.html

olerem, 2024-10-23:
Hi @xile273, you are right! Thank you. Do you wont to send a patch?

olerem, 2024-10-23:
Addresses:
To:
Robin van der Gracht <robin@protonic.nl>
Oliver Hartkopp <socketcan@hartkopp.net>
Marc Kleine-Budde <mkl@pengutronix.de>
"David S. Miller" <davem@davemloft.net>
Eric Dumazet <edumazet@google.com>
Jakub Kicinski <kuba@kernel.org>
Paolo Abeni <pabeni@redhat.com>
Jonathan Corbet <corbet@lwn.net>

Cc:
kernel@pengutronix.de                   
linux-can@vger.kernel.org               
linux-kernel@vger.kernel.org

xile273, 2024-10-23:
Ok great, thank you! Will do that.

marckleinebudde, 2024-10-23:
FTR: https://lore.kernel.org/all/20241023145257.82709-1-alexander.hoelzl@gmx.net/

closed 2024-11-22 by xile273