linux-can/can-utils #561: sudo ip link set can0 up throws segmentation error fault

Markkolas, 2024-10-30:
Ok so I am not sure if this issue should be posted here but I saw that the previous issue was about something related with the kernel docs, so here we go.

I am doing a setup with a Raspberry Pi 4B, Raspbian 12 (6.6 kernel version) and this 2-CH CAN FD HAT: https://www.waveshare.com/wiki/2-CH_CAN_FD_HAT.

I've followed the link's intructions to do a simple config: SPI interface activated via `raspi-config` and added the relevant SPI config at the end of file _/boot/firmware/config.txt_:

```
dtoverlay=spi1-3cs
dtoverlay=mcp251xfd,spi0-0,interrupt=25
dtoverlay=mcp251xfd,spi1-0,interrupt=24
```

Now ip link shows CAN buses:
```
esautomatix@raspberrymatix:~ $ ip link
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN mode DEFAULT group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
2: eth0: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN mode DEFAULT group default qlen 1000
    link/ether d8:3a:dd:74:98:1c brd ff:ff:ff:ff:ff:ff
3: can0: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/can
4: can1: <NOARP,ECHO> mtu 16 qdisc noop state DOWN mode DEFAULT group default qlen 10
    link/can
5: wlan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP mode DORMANT group default qlen 1000
    link/ether d8:3a:dd:74:98:1d brd ff:ff:ff:ff:ff:ff
```

But when setting them up this occurs:
```
esautomatix@raspberrymatix:~ $ sudo ip link set can0 up type can bitrate 500000 dbitrate 2000000 restart-ms 1000 berr-reporting on fd on
Segmentation fault
```

Sometimes the S.O. even crashes. `journalctl -xe` says:

```
Oct 30 15:31:12 raspberrymatix sudo[969]: esautomatix : TTY=pts/0 ; PWD=/home/esautomatix ; USER=root ; COMMAND=/usr/sbin/ip link set can0 up type can bitrate 500000 dbitrate 2000000 restart-ms 1000 berr-reporting on fd on
Oct 30 15:31:12 raspberrymatix sudo[969]: pam_unix(sudo:session): session opened for user root(uid=0) by esautomatix(uid=1000)
Oct 30 15:31:12 raspberrymatix kernel: Unable to handle kernel NULL pointer dereference at virtual address 0000000000000000
Oct 30 15:31:12 raspberrymatix kernel: Mem abort info:
Oct 30 15:31:12 raspberrymatix kernel:   ESR = 0x0000000096000005
Oct 30 15:31:12 raspberrymatix kernel:   EC = 0x25: DABT (current EL), IL = 32 bits
Oct 30 15:31:12 raspberrymatix kernel:   SET = 0, FnV = 0
Oct 30 15:31:12 raspberrymatix kernel:   EA = 0, S1PTW = 0
Oct 30 15:31:12 raspberrymatix kernel:   FSC = 0x05: level 1 translation fault
Oct 30 15:31:12 raspberrymatix kernel: Data abort info:
Oct 30 15:31:12 raspberrymatix kernel:   ISV = 0, ISS = 0x00000005, ISS2 = 0x00000000
Oct 30 15:31:12 raspberrymatix kernel:   CM = 0, WnR = 0, TnD = 0, TagAccess = 0
Oct 30 15:31:12 raspberrymatix kernel:   GCS = 0, Overlay = 0, DirtyBit = 0, Xs = 0
Oct 30 15:31:12 raspberrymatix kernel: user pgtable: 4k pages, 39-bit VAs, pgdp=0000000103a28000
Oct 30 15:31:12 raspberrymatix kernel: [0000000000000000] pgd=0000000000000000, p4d=0000000000000000, pud=0000000000000000
Oct 30 15:31:12 raspberrymatix kernel: Internal error: Oops: 0000000096000005 [#1] PREEMPT SMP
Oct 30 15:31:12 raspberrymatix kernel: Modules linked in: cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep brcmfmac_wcc vc4 brcmfmac hci_uart snd_soc_hdmi_codec brcmutil btbcm drm_display_helper bluetooth cec cfg80211 binfmt_misc drm_dma_helper mcp251xfd drm_kms_helper can_dev ecdh_generic snd_soc_core ecc bcm2835_v4l2(C) bcm2835_isp(C) v3d bcm2835_codec(C) rpivid_hevc(C) rfkill snd_compress gpu_sched libaes v4l2_mem2mem bcm2835_mmal_vchiq(C) snd_bcm2835(C) snd_pcm_dmaengine drm_shmem_helper videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops raspberrypi_hwmon snd_pcm videobuf2_v4l2 videodev snd_timer videobuf2_common raspberrypi_gpiomem vc_sm_cma(C) snd mc nvmem_rmem uio_pdrv_genirq uio drm fuse dm_mod drm_panel_orientation_quirks backlight ip_tables x_tables ipv6 spidev i2c_brcmstb spi_bcm2835 spi_bcm2835aux
Oct 30 15:31:12 raspberrymatix kernel: CPU: 2 PID: 971 Comm: ip Tainted: G         C         6.6.51+rpt-rpi-v8 #1  Debian 1:6.6.51-1+rpt3
Oct 30 15:31:12 raspberrymatix kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.5 (DT)
Oct 30 15:31:12 raspberrymatix kernel: pstate: 60000005 (nZCv daif -PAN -UAO -TCO -DIT -SSBS BTYPE=--)
Oct 30 15:31:12 raspberrymatix kernel: pc : timecounter_read+0x20/0x80
Oct 30 15:31:12 raspberrymatix kernel: lr : mcp251xfd_ring_init+0x1a0/0x500 [mcp251xfd]
Oct 30 15:31:12 raspberrymatix kernel: sp : ffffffc0816eb420
Oct 30 15:31:12 raspberrymatix kernel: x29: ffffffc0816eb420 x28: ffffff8101bb6000 x27: 0000000000000001
Oct 30 15:31:12 raspberrymatix kernel: x26: 0000000000000000 x25: ffffff8101bb6040 x24: ffffff81027e2940
Oct 30 15:31:12 raspberrymatix kernel: x23: 0000000000000000 x22: 0000000000000430 x21: 0000000000000001
Oct 30 15:31:12 raspberrymatix kernel: x20: ffffff81027e0940 x19: ffffff81027e38b0 x18: 0000000000000000
Oct 30 15:31:12 raspberrymatix kernel: x17: 0000000000000000 x16: ffffffe006553b38 x15: 0000000000000000
Oct 30 15:31:12 raspberrymatix kernel: x14: 0000000000000001 x13: 0000000000000000 x12: 0000000000000000
Oct 30 15:31:12 raspberrymatix kernel: x11: 00000000000001de x10: 0000000000000001 x9 : ffffffdff22a5980
Oct 30 15:31:12 raspberrymatix kernel: x8 : 0000000000000001 x7 : 0000000000000000 x6 : 000000000000000a
Oct 30 15:31:12 raspberrymatix kernel: x5 : 0000000000000001 x4 : ffffffdff22b09c8 x3 : 0000000000000000
Oct 30 15:31:12 raspberrymatix kernel: x2 : 000000000000005c x1 : 0000000000002f70 x0 : 0000000000000000
Oct 30 15:31:12 raspberrymatix kernel: Call trace:
[... 18 lines cut ...]
Oct 30 15:31:12 raspberrymatix kernel:  __sys_sendmsg+0x8c/0xf0
Oct 30 15:31:12 raspberrymatix kernel:  __arm64_sys_sendmsg+0x2c/0x40
Oct 30 15:31:12 raspberrymatix kernel:  invoke_syscall+0x50/0x128
Oct 30 15:31:12 raspberrymatix kernel:  el0_svc_common.constprop.0+0x48/0xf0
Oct 30 15:31:12 raspberrymatix kernel:  do_el0_svc+0x24/0x38
Oct 30 15:31:12 raspberrymatix kernel:  el0_svc+0x40/0xe8
Oct 30 15:31:12 raspberrymatix kernel:  el0t_64_sync_handler+0x100/0x130
Oct 30 15:31:12 raspberrymatix kernel:  el0t_64_sync+0x190/0x198
Oct 30 15:31:12 raspberrymatix kernel: Code: 910003fd f9000bf3 aa0003f3 f9400000 (f9400001)
Oct 30 15:31:12 raspberrymatix kernel: ---[ end trace 0000000000000000 ]---
Oct 30 15:31:12 raspberrymatix sudo[969]: pam_unix(sudo:session): session closed for user root
```

CAN drivers seems to have been loaded correctly, `dmesg | grep spi` gives:
```
[    6.809325] spi_master spi1: will run message pump with realtime priority
[    6.832781] mcp251xfd spi1.0 can0: MCP2518FD rev0.0 (-RX_INT -PLL -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD o:40.00MHz c:40.00MHz m:20.00MHz rs:17.00MHz es:16.66MHz rf:17.00MHz ef:16.66MHz) successfully initialized.
[    6.833475] spi_master spi0: will run message pump with realtime priority
[    6.877258] mcp251xfd spi0.0 can1: MCP2518FD rev0.0 (-RX_INT -PLL -MAB_NO_WARN +CRC_REG +CRC_RX +CRC_TX +ECC -HD o:40.00MHz c:40.00MHz m:20.00MHz rs:17.00MHz es:16.66MHz rf:17.00MHz ef:16.66MHz) successfully initialized.
[ 1019.548103] Modules linked in: cmac algif_hash aes_arm64 aes_generic algif_skcipher af_alg bnep brcmfmac_wcc vc4 brcmfmac hci_uart snd_soc_hdmi_codec brcmutil btbcm drm_display_helper bluetooth cec cfg80211 binfmt_misc drm_dma_helper mcp251xfd drm_kms_helper can_dev ecdh_generic snd_soc_core ecc bcm2835_v4l2(C) bcm2835_isp(C) v3d bcm2835_codec(C) rpivid_hevc(C) rfkill snd_compress gpu_sched libaes v4l2_mem2mem bcm2835_mmal_vchiq(C) snd_bcm2835(C) snd_pcm_dmaengine drm_shmem_helper videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops raspberrypi_hwmon snd_pcm videobuf2_v4l2 videodev snd_timer videobuf2_common raspberrypi_gpiomem vc_sm_cma(C) snd mc nvmem_rmem uio_pdrv_genirq uio drm fuse dm_mod drm_panel_orientation_quirks backlight ip_tables x_tables ipv6 spidev i2c_brcmstb spi_bcm2835 spi_bcm2835aux
```

I am no expert and I may f up with the setup, but the kernel throwing a segmentation fault feels like something that shouldn't happen, even if I messed at some point of the setup. I've noticed that the Rapbian version that I am using is from October 22nd, so I'll try with an older one, but before that I wanted to leave what happened here to see if it is something serious.

Have a nice day.

marckleinebudde, 2024-10-30:
Also have a look at https://github.com/raspberrypi/linux/issues/6406#issuecomment-2431732058

Markkolas, 2024-10-30:
I see, thanks for the fast response :)

closed 2024-11-11 by marckleinebudde