linux-can/can-utils #562: Obtain CAN hardware transmission acknowledgement

xile273, 2024-11-05:
Hello,
I just wanted to know if there is any way to know from userspace, if a sent CAN frame has been transmitted by the hardware.
I know for Ethernet, hardware transmission timestamps could be used for such a purpose but I'm not sure if that would work/be the best
approach for CAN.

I'm asking because for CAN, or any protocol building on CAN like J1939, cyclical transmission of data is very common.
If there is no other active device on the bus or the interface is physically not connected, the SocketCAN stack will send 
all packets at once as soon as a connection is established. If there have been multiple frames queued, all of them will be sent at once
which is not desirable behavior. Instead it would be preferable to delay sending the next message of a cyclical service, until 
the previous message has been sent successfully by the hardware.

Thank you for your help!

marckleinebudde, 2024-11-05:
Are you using `CAN_RAW` or `CAN_J1939`?

xile273, 2024-11-05:
> Are you using `CAN_RAW` or `CAN_J1939`?

Sorry I wasn't clear. I'm using both, but the general situation is the same regardless of the protocol.
I'm using a mcp251xfd controller btw.

marckleinebudde, 2024-11-05:
For `CAN_RAW` you can receive your own messages on the same socket, see https://github.com/torvalds/linux/blob/v6.10/Documentation/networking/can.rst#raw-socket-option-can_raw_recv_own_msgs.

On J1939 sockets you can activaate `setsockopt(SO_J1939_ERRQUEUE)`, see https://github.com/linux-can/can-utils/blob/6526683a10d8700ceea5148e5d2652fadaf3db6d/j1939cat.c#L605

olerem, 2024-11-05:
Here is some documentation which should help:
https://lore.kernel.org/all/20241013181715.3488980-1-o.rempel@pengutronix.de/

rusoku, 2024-11-05:
> For `CAN_RAW` you can receive your own messages on the same socket, see https://github.com/torvalds/linux/blob/v6.10/Documentation/networking/can.rst#raw-socket-option-can_raw_recv_own_msgs.

But IMHO in this case CAN frames are just loop backed and you will get you own message but not ACK'ed by another node on the bus.

marckleinebudde, 2024-11-05:
With proper driver support, you only get the TX'ed message after it has been successfully send to the bus.

xile273, 2024-11-13:
Hello, sorry the reopen this issue, but I'm still having some problems. I think there is a small conceptional problem with regard to the tx-acks in the J1939 module.
When I send a J1939 message and it gets transmitted by the controller I receive a SCM_TSTAMP_ACK from the error-queue.
If it cannot be transmitted after 10 seconds the J1939_SIMPLE_ECHO_TIMEOUT timeout expires and the error queue returns J1939_EE_INFO_TX_ABORT.
Unfortunately, even though the transmission was aborted, the message is queued in the tx-buffer of the CAN-controller and gets sent as soon as the bus is reestablished.

This situation is detected by the the module as it warns about `Received already invalidated message`, but there's no way for the userspace to know about this situation. For my use case this is a bit problematic as I don't see a clear way on how to deal with those kind of no-bus conditions. I can't just wait for SCM_TSTAMP_ACK as after 10 seconds the message will expire and it 
won't be sent. And if I allow sending again after a J1939_EE_INFO_TX_ABORT, then I will still run into the problem of sending
all queued frames after the bus is available again. Albeit the buffer filling at a lower rate of one packet every 10 seconds. 

Is there a possibility to not let the transmission timeout? Like this, the state of the controller and the J1939 stack would not
get out of sync? Alternatively it would also help to be notified about receiving an already invalidated message.
The most perfect solution though would be to remove the queued frame from the controllers tx-buffer but I guess that's not possible due to the limits of CAN-hardware.

Thank you for your help already!

olerem, 2024-11-13:
Yes, this is know limitation. Current kernel CAN stack can't abort queued message - it can be potentially implemented (at least for some HW). The same behavior will affect user space J1939 implementation too.
With other words, it will be great if some one send patches, or if possible, assign some contractor to implement it. One of possible contractors will be pengutronix.

xile273, 2024-11-14:
Ok I see thank you for the information! I'll see what I can do about it.

closed 2024-11-14 by xile273