linux-can/can-utils #564: Cannot send zero length j1939 messages.

xile273, 2024-12-05:
Hello,
sorry to bother you again regarding the J1939 implementation.

It seems it is impossible to send J1939 messages with data length 0.
Unfortunately the J1939 standard requires the transmission of such messages,
for example J1939-73 DM11 (5.7.11.3) or DM55 (5.7.55).

Receiving such messages is possible though.

Thank you for your help!

olerem, 2024-12-05:
@xile273 , good point. Thx!
If you not fear the kernel hacking, it will be great if you'll try to fix it. If not, i'll try to do it later.

xile273, 2024-12-05:
Hello, 
I do not fear it, and I'll gladly take a look but it will take me much longer than you. 
I'll try it though :)

olerem, 2024-12-05:
i'll be very happy if there are more hackers in the stack :)

xile273, 2024-12-06:
I did get it to work, see xile273/linux@3bf3fa038f631436209c8998b400670f93afb18c.
The condition in j1939_session_skb_get_by_offset is now maximally ugly, and I'll simplify it, but it does seem to work.
However I don't have the infrastructure to thoroughly check for regressions in J1939 protocol
behavior. Especially not when it comes to ETP.

olerem, 2024-12-07:
Nice, you can test it at least with vcan and following test scripts:

https://github.com/linux-can/can-tests/tree/master/j1939

For example you can run: ./j1939_ac_1m_local0.sh vcan0 vcan0
in this case sender and receiver will be on same interface.

xile273, 2024-12-09:
They all seem to work on vcan as well as mcp2518fd

olerem, 2024-12-13:
ok, can you please prepare the patches and send them upstream?

marckleinebudde, 2024-12-13:
Should we put stable@k.o on Cc?

olerem, 2024-12-13:
yes, good point

xile273, 2025-01-07:
Sorry, I was very busy the last weeks, finally sent a patch.

marckleinebudde, 2025-01-07:
FTR: https://lore.kernel.org/all/20250107133217.119646-1-alexander.hoelzl@gmx.net/

xile273, 2025-01-07:
> It seems it is impossible to send J1939 messages with data length 0. Unfortunately the J1939 standard requires the transmission of such messages, for example J1939-73 DM11 (5.7.11.3) or DM55 (5.7.55).

Also I just want to add that the examples are not really correct, I misread the standard. DM11 and DM55 are not sent directly just as requests. Thats why their data length is specified is zero.
However the proprietary messages, e.g PROPA defined in J1939-21 are specified with a length of 0 to 1785, so the change is still needed to comply with the standard.