linux-can/can-utils #567: mcp251xfd: CRC problems

marckleinebudde, 2024-12-11:
> Fixed by: [raspberrypi/linux#6407](https://github.com/raspberrypi/linux/issues/6407) [raspberrypi/linux#6466](https://github.com/raspberrypi/linux/pull/6466) https://lore.kernel.org/all/20241025-mcp251xfd-fix-coalesing-v1-1-9d11416de1df@pengutronix.de/

2024-12-11_12:47  RPICANMON  ~   
$ modinfo can
filename:       /lib/modules/6.1.21-v8+/kernel/net/can/can.ko.xz
alias:          net-pf-29
author:         Urs Thuermann <urs.thuermann@volkswagen.de>, Oliver Hartkopp <oliver.hartkopp@volkswagen.de>
license:        Dual BSD/GPL
description:    Controller Area Network PF_CAN core
srcversion:     76F20AE7FC3EED720718288
depends:        
intree:         Y
name:           can
vermagic:       6.1.21-v8+ SMP preempt mod_unload modversions aarch64
parm:           stats_timer:enable timer for statistics (default:on) (int)
2024-12-11_12:47  RPICANMON  ~   

I am seeing the CRC errors as well. The driver seems to work for 20 mins or so and then fails (candump stops producing text). Dmesg fragment below

> 
> 54.135255] mcp251xfd spi0.1 can0: Single ECC Error detected at address 0x0000.
> [   54.135579] mcp251xfd spi0.1 can0: CRC write command format error.
> [   54.136033] mcp251xfd spi0.1 can0: Single ECC Error detected at address 0x0000.
> [   54.136349] mcp251xfd spi0.1 can0: CRC write command format error.
> [   54.136802] mcp251xfd spi0.1 can0: Single ECC Error detected at address 0x0000.
> [   54.137117] mcp251xfd spi0.1 can0: CRC write command format error.
> [   54.137571] mcp251xfd spi0.1 can0: CRC write command format error.
> [   54.311054] mcp251xfd spi0.1 can0: CRC write command format error.
> [   54.311484] mcp251xfd spi0.1 can0: CRC write command format error.
> [   54.317928] mcp251xfd spi0.1 can0: CRC read error at address 0x069c (length=68, data=8c 08 00 00 10 00 00 00 00 00 80 00 18 00 00 00 ed f4 89 17 ed f4 89 17 c2 00 00 00 00 20 3f 00 00 20 3f 00 00 00 00 00 00 00 00 00 00 00 20 30 00 80 80 08 18 00 00 00 13 c1 9b 17 13 c1 9b 17, CRC=0x07f2) retrying.
> [   54.339546] mcp251xfd spi0.1 can0: Single ECC Error detected at address 0x05a8.
> [   54.514064] mcp251xfd spi0.1 can0: CRC write command format error.
> [   54.516563] mcp251xfd spi0.1 can0: CRC write command format error.
> [   54.516977] mcp251xfd spi0.1 can0: CRC write error detected. CRC=0xc34c.
> [   54.517484] mcp251xfd spi0.1 can0: CRC write command format error.
> [   54.517899] mcp251xfd spi0.1 can0: CRC write command format error.

_Originally posted by @robin48gx in https://github.com/linux-can/can-utils/issues/563#issuecomment-2535901591_

marckleinebudde, 2024-12-11:
The communication between your SoC and the mcp251xfd includes a CRC sum. If the SoC writes to the mcp251xfd and the CRC is not correct, the chip raises a `CRC write command format error`. A CRC problem during read is signaled and retried. And if the chip detects a CRC error in it's RAM, you get a `ECC Error detected at address 0x....`.

- Which SoC are you using?
- Is this problem new?
- Is your setup in a electromagnetically noisy environment?
- Do you have a proper power supply? Don't know how the chip reacts to brown outs.

robin48gx, 2024-12-11:
The SoC is a standard raspberry pi 4. 
The problem is new, but we have 20 years of experience with CAN (mainly PIC18, MCP2515 and various STM32 devices).
The electrical environment is pretty clean (No 415a.c. inverters or contactors).
Power supply Supply is as recommended for pi4

It is a prototype PCB see pic
[image]
and the SPI lines wander a little on the PCB to test points, but for the last 30 mins has been running fine when set to 20MHz SPI.

when this error does occur, it seems to kill the can driver. The dmesg is full of crc errors but taking the link up down and then up
fixes it again, for a while. 

Although the driver is the same we have two of these running better now by upgrading the raspberry pi kernel.
The version of the can driver is the same however (modinfor can gives same result after kernel upgrade)
Being an intermittent fault, red herrings are, of course, popping up!

marckleinebudde, 2024-12-11:
> when this error does occur, it seems to kill the can driver. The dmesg is full of crc errors

If you don't stop the interface, does the messages from the driver stop? Can you post the whole output?

robin48gx, 2024-12-12:
> > when this error does occur, it seems to kill the can driver. The dmesg is full of crc errors
> 
> If you don't stop the interface, does the messages from the driver stop? Can you post the whole output?

After updating the kernel (which kept the same version of the can module) this appears to be behaving well. I have two on long term test. If it CRC errors again I;ll send the full dmesg report.

marckleinebudde, 2024-12-12:
What's the `uname -r` before and after the update?

robin48gx, 2024-12-12:
uname -r now: 
6.1.77-v8+
2024-12-12_10:29  RPICANMON  ~/can_logs   

Here are my old kernels:
$ ls -l /boot/k*
-rwxr-xr-x 1 root root 8757403 Dec 11 13:53 /boot/kernel_2712.img
-rwxr-xr-x 1 root root 7013976 Dec 11 13:53 /boot/kernel7.img
-rwxr-xr-x 1 root root 7443352 Dec 11 13:53 /boot/kernel7l.img
-rwxr-xr-x 1 root root 8754568 Dec 11 13:53 /boot/kernel8.img
-rwxr-xr-x 1 root root 6632208 Dec 11 13:53 /boot/kernel.img

marckleinebudde, 2024-12-12:
FTR: The old kernel was `6.1.21-v8+` now you're on `6.1.77-v8+`.

closed 2024-12-12 by marckleinebudde

pompushko, 2025-01-30:
I'm not sure, but I'm facing same issues...

```
filename:       /lib/modules/6.6.62+rpt-rpi-v7/kernel/net/can/can.ko.xz
alias:          net-pf-29
author:         Urs Thuermann <urs.thuermann@volkswagen.de>, Oliver Hartkopp <oliver.hartkopp@volkswagen.de>
license:        Dual BSD/GPL
description:    Controller Area Network PF_CAN core
srcversion:     9FCDE9DC80F2506E3093972
depends:
intree:         Y
name:           can
vermagic:       6.6.62+rpt-rpi-v7 SMP mod_unload modversions ARMv7 p2v8
parm:           stats_timer:enable timer for statistics (default:on) (int)
```

```
Linux device 6.6.62+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.62-1+rpt1 (2024-11-25) armv7l GNU/Linux
```

marckleinebudde, 2025-01-31:
This looks subtle different, can you please open a new issue.